pschatzmann/ESP32-A2DP #244: Where are the `avrc_peer_rn_cap_bits` defined?

clarkbremer, 2022-06-24:
I can get a value back, but I don't know how to interpret the bits.

pschatzmann, 2022-06-24:
I suggest to search for esp_avrc_ct_send_get_rn_capabilities_cmd. This provides esp_avrc_rn_evt_cap_mask_t which is 
bit mask representation of PASSTHROUGH commands: see esp_avrc_pt_cmd_t.
I havn't double checked this however...

pschatzmann, 2022-06-24:
If it is not the esp_avrc_pt_cmd_t values, I would not know....

clarkbremer, 2022-06-24:
Those don't look like "capabilities".  The look like "commands".  Plus, since it's a bitmap in a uint16_t value, there could only be 16 of them.  There are far more of these commands.

pschatzmann, 2022-06-24:
OK, I see the bits are pointless and I need to change the API:
- I need to store s_avrc_peer_rn_cap
- The API should call e.g. esp_avrc_rn_evt_bit_mask_operation(ESP_AVRC_BIT_MASK_OP_TEST, &s_avrc_peer_rn_cap,
                                           ESP_AVRC_RN_TRACK_CHANGE))

pschatzmann, 2022-06-24:
I replaced the method with: bool is_avrc_peer_rn_cap(esp_avrc_rn_event_ids_t cmd)

taseraul, 2022-06-26:
Hello, 
I think something broke in the latest commit 
I am getting this error 
```
BluetoothA2DPSink.cpp:27:9: error: 's_avrc_peer_rn_cap' was not declared in this scope
         s_avrc_peer_rn_cap.bits = 0;
```

pschatzmann, 2022-06-26:
I just committed a correction.