pschatzmann/ESP32-A2DP #246: 1.7.1 don't compile.

Oppa-YA, 2022-06-24:
Hello! I can't compile "Release v1.7.1 - I2S output in separate task with ringbuffer & new functionality"
While compiling i was getting error `no matching function for call to 'A2DPVolumeControl::get_volume_factor(uint8_t&)'`
It seems it don't like code in the **_BluetoothA2DPSink32.h_**
I was using code from "bt_music_receiver_32bit_ext" example. In version 1.7.0 it is compiling without problems, I was using Arduino IDE 1.8.19 + ESP32 v 2.0.3 latest board versions.
Here is the screenshot of Arduino IDE output while compiling latest 1.7.1:

[image]

pschatzmann, 2022-06-24:
I think this example has not been updated for quite some time...
Just change the line to  volumeFactor = volume_control()->get_volume_factor();
The correction has also been committed

Oppa-YA, 2022-06-24:
Thank you for fast response!
Can confirm it is fixed now and working!

closed 2022-06-24 by Oppa-YA