pschatzmann/ESP32-A2DP #248: Sometimes fail to reconnect after reboot

clarkbremer, 2022-06-24:
About 1 time out of 5, when I reboot the ESP32, it fails to automatically reconnect to my (android) phone.  Here are "good" and "bad" log outputs:

Good:
``` 
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4
[   262][I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 222708
[   265][I][BluetoothA2DPSink.cpp:139] start(): Device name will be set to 'TenonTech CDX1'
[   274][I][BluetoothA2DPSink.cpp:187] init_i2s(): init_i2s
[   278][I][BluetoothA2DPSink.cpp:189] init_i2s(): init_i2s is_i2s_output
[   377][I][BluetoothA2DPSink.cpp:266] init_bluetooth(): controller initialized
[   682][I][BluetoothA2DPSink.cpp:275] init_bluetooth(): bluedroid initialized
[   757][I][BluetoothA2DPSink.cpp:284] init_bluetooth(): bluedroid enabled
[   758][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[   763][I][BluetoothA2DPCommon.cpp:241] set_scan_mode_connectable(): set_scan_mode_connectable true
[   771][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[   773][I][BluetoothA2DPSink.cpp:487] app_gap_callback(): event: 10
[   780][I][BluetoothA2DPSink.cpp:564] av_hdl_a2d_evt(): A2DP PROF STATE: Init Compl
[   781][I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 119672
[   794][I][BluetoothA2DPSink.cpp:487] app_gap_callback(): event: 10
[   807][I][BluetoothA2DPSink.cpp:487] app_gap_callback(): event: 10
[   818][I][BluetoothA2DPSink.cpp:487] app_gap_callback(): event: 10
[   827][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[   832][I][BluetoothA2DPSink.cpp:650] handle_connection_state(): partner address: f8:1a:2b:e9:cc:8e
[   840][I][BluetoothA2DPSink.cpp:658] handle_connection_state(): A2DP connection state: Connecting, [f8:1a:2b:e9:cc:8e]
[   850][I][BluetoothA2DPSink.cpp:747] handle_connection_state(): ESP_A2D_CONNECTION_STATE_CONNECTING
[   870][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
[  1467][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1468][I][BluetoothA2DPSink.cpp:582] handle_audio_cfg(): a2dp audio_cfg_cb , codec type 0
[  1474][I][BluetoothA2DPSink.cpp:594] handle_audio_cfg(): a2dp audio_cfg_cb , sample_rate 44100
[  1482][I][BluetoothA2DPSink.cpp:605] handle_audio_cfg(): configure audio player 21-15-2-35
[  1491][I][BluetoothA2DPSink.cpp:611] handle_audio_cfg(): audio player configured, samplerate=44100
[  1500][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1509][I][BluetoothA2DPSink.cpp:650] handle_connection_state(): partner address: f8:1a:2b:e9:cc:8e
[  1518][I][BluetoothA2DPSink.cpp:658] handle_connection_state(): A2DP connection state: Connected, [f8:1a:2b:e9:cc:8e]
[  1528][I][BluetoothA2DPSink.cpp:699] handle_connection_state(): ESP_A2D_CONNECTION_STATE_CONNECTED
[  1537][I][BluetoothA2DPCommon.cpp:241] set_scan_mode_connectable(): set_scan_mode_connectable false
[  1546][I][BluetoothA2DPSink.cpp:1229] bt_i2s_task_start_up(): BtI2STask Started
[  1553][I][BluetoothA2DPSink.cpp:724] handle_connection_state(): i2s_start
[  1560][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1569][I][BluetoothA2DPSink.cpp:789] av_hdl_avrc_evt(): AVRC conn_state evt: state 1, [f8:1a:2b:e9:cc:8e]
[  1577][I][BluetoothA2DPSink.cpp:477] app_gap_callback(): ESP_BT_GAP_READ_REMOTE_NAME_EVT stat:0
[  1579][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1597][I][BluetoothA2DPSink.cpp:1323] av_hdl_avrc_tg_evt(): AVRC conn_state evt: state 1, [f8:1a:2b:e9:cc:8e]
[  1606][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1616][I][BluetoothA2DPSink.cpp:828] av_hdl_avrc_evt(): AVRC remote features 25b
[  1626][I][BluetoothA2DPSink.cpp:479] app_gap_callback(): ESP_BT_GAP_READ_REMOTE_NAME_EVT remote name:Clarks Phone
[  1642][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1643][I][BluetoothA2DPSink.cpp:1339] av_hdl_avrc_tg_evt(): AVRC register event notification: 13, param: 0x0
[  1652][I][BluetoothA2DPSink.cpp:1341] av_hdl_avrc_tg_evt(): AVRC Volume Changes Supported
[  1660][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1670][I][BluetoothA2DPSink.cpp:836] av_hdl_avrc_evt(): remote rn_cap: count 7, bitmask 0x1e26
[  1678][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1687][I][BluetoothA2DPSink.cpp:813] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x1, King
[  1696][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1705][I][BluetoothA2DPSink.cpp:813] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x2, TesseracT
[  1715][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1724][I][BluetoothA2DPSink.cpp:813] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x4, Sonder
[  1733][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1742][I][BluetoothA2DPSink.cpp:813] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x8, 1
[  1751][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1760][I][BluetoothA2DPSink.cpp:813] av_hdl_avrc_evt(): AVRC metadata rsp: attribute id 0x10, 1
[  1769][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1778][I][BluetoothA2DPSink.cpp:1333] av_hdl_avrc_tg_evt(): AVRC set absolute volume: 75%
[... 30 lines cut ...]
[   768][I][BluetoothA2DPSink.cpp:284] init_bluetooth(): bluedroid enabled
[   769][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[   774][I][BluetoothA2DPCommon.cpp:241] set_scan_mode_connectable(): set_scan_mode_connectable true
[   783][I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 115808
[   783][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[   784][I][BluetoothA2DPSink.cpp:487] app_gap_callback(): event: 10
[   798][I][BluetoothA2DPSink.cpp:564] av_hdl_a2d_evt(): A2DP PROF STATE: Init Compl
[   812][I][BluetoothA2DPSink.cpp:487] app_gap_callback(): event: 10
[   823][I][BluetoothA2DPSink.cpp:487] app_gap_callback(): event: 10
[   826][I][BluetoothA2DPSink.cpp:487] app_gap_callback(): event: 10
[   839][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[   847][I][BluetoothA2DPSink.cpp:650] handle_connection_state(): partner address: f8:1a:2b:e9:cc:8e
[   855][I][BluetoothA2DPSink.cpp:658] handle_connection_state(): A2DP connection state: Connecting, [f8:1a:2b:e9:cc:8e]
[   865][I][BluetoothA2DPSink.cpp:747] handle_connection_state(): ESP_A2D_CONNECTION_STATE_CONNECTING
[   885][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
[ 60838][I][BluetoothA2DPSink.cpp:363] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 60839][I][BluetoothA2DPSink.cpp:650] handle_connection_state(): partner address: f8:1a:2b:e9:cc:8e
[ 60846][I][BluetoothA2DPSink.cpp:658] handle_connection_state(): A2DP connection state: Disconnected, [f8:1a:2b:e9:cc:8e]
[ 60856][I][BluetoothA2DPSink.cpp:661] handle_connection_state(): ESP_A2D_CONNECTION_STATE_DISCONNECTED
[ 60866][I][BluetoothA2DPSink.cpp:1243] bt_i2s_task_shut_down(): BtI2STask shutdown
[ 60873][I][BluetoothA2DPSink.cpp:675] handle_connection_state(): i2s_stop
[ 60880][I][BluetoothA2DPCommon.cpp:241] set_scan_mode_connectable(): set_scan_mode_connectable true
```

Based on the timestamps, it looks like something times out after a minute.

pschatzmann, 2022-06-24:
Hmm, can you double check if this also if you use BluetoothA2DPSinkMinRAM instead BluetoothA2DPSink ?

clarkbremer, 2022-06-25:
Much improved!  I may have seen it once in 20 attempts, but not even sure about that one.  Thanks, Phil!

clarkbremer, 2022-06-30:
I'm going to have to retract that premature celebration.  I'm still seeing this fairly often.  Is there something else we can tune to make this more aggressive?  The bluetooth in my wife's Honda *always* reconnects.  The ESP32-based player in my truck still fails to reconnect sometimes.

pschatzmann, 2022-06-30:
I suggest that you play around with the delay in BluetoothA2DPSink.cpp on line 686. If an increased value helps I will make the value configurable with a setter method...

clarkbremer, 2022-06-30:
As far as I can tell, that code is not being executed.  I added several more log statements:

```
 if (!end_in_progress) {
            ESP_LOGI(BT_AV_TAG,"try reconnect1");
            if (is_reconnect(a2d->conn_stat.disc_rsn)) {
                ESP_LOGI(BT_AV_TAG,"try reconnect2");
                if (connection_rety_count < try_reconnect_max_count ){
                    ESP_LOGI(BT_AV_TAG,"try reconnect3");
                    ESP_LOGI(BT_AV_TAG,"Connection try number: %d", connection_rety_count);
                    // make sure that any open connection is timing out on the target
                    delay(1000);
                    connect_to_last_device();
                    // when we lost the connection we do allow any others to connect after 2 trials
                    if (connection_rety_count==2) set_scan_mode_connectable(true);

                } else {
                    ESP_LOGI(BT_AV_TAG, "Reconect retry limit reached");
                    if ( has_last_connection() && a2d->conn_stat.disc_rsn == ESP_A2D_DISC_RSN_NORMAL ){
                        clean_last_connection();
                    }
                    set_scan_mode_connectable(true);
                }
            } else {
                set_scan_mode_connectable(true);   
            }
        }
```

And I don't see any of these messages.

Log:  
```
[   263][W][BigRed2.ino:1228] setup(): Tenon Tech, Model CDX1, booting.  Copyright (c) 2022 Tenon Technology.
[   265][I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 222988
[   269][I][BluetoothA2DPSink.cpp:141] start(): Device name will be set to 'CDX1 Vader Black'
[   279][I][BluetoothA2DPSink.cpp:189] init_i2s(): init_i2s
[   283][I][BluetoothA2DPSink.cpp:191] init_i2s(): init_i2s is_i2s_output
[   377][I][BluetoothA2DPSink.cpp:268] init_bluetooth(): controller initialized
[   704][I][BluetoothA2DPSink.cpp:277] init_bluetooth(): bluedroid initialized
[   778][I][BluetoothA2DPSink.cpp:286] init_bluetooth(): bluedroid enabled
[   778][I][BluetoothA2DPSink.cpp:365] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[   783][I][BluetoothA2DPCommon.cpp:241] set_scan_mode_connectable(): set_scan_mode_connectable true
[   792][I][BluetoothA2DPSink.cpp:365] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[   792][I][BluetoothA2DPSink.cpp:489] app_gap_callback(): event: 10
[   801][I][BluetoothA2DPSink.cpp:566] av_hdl_a2d_evt(): A2DP PROF STATE: Init Compl

[   801][I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 120724
[   815][I][BluetoothA2DPSink.cpp:489] app_gap_callback(): event: 10
[   822][I][BigRed2.ino:591] run(): LCD addr: 0x27
[   828][I][BluetoothA2DPSink.cpp:489] app_gap_callback(): event: 10
[   832][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
[   838][I][BluetoothA2DPSink.cpp:489] app_gap_callback(): event: 10
[   854][I][BluetoothA2DPSink.cpp:365] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[   860][I][BluetoothA2DPSink.cpp:652] handle_connection_state(): partner address: f8:1a:2b:e9:cc:8e
[   869][I][BluetoothA2DPSink.cpp:660] handle_connection_state(): A2DP connection state: Connecting, [f8:1a:2b:e9:cc:8e]
[   879][I][BluetoothA2DPSink.cpp:755] handle_connection_state(): ESP_A2D_CONNECTION_STATE_CONNECTING
[  1987][I][BigRed2.ino:891] run(): Master Initialized
[  2045][I][BigRed2.ino:917] run(): Got 51 (BOOT), moving on
[  8046][W][BigRed2.ino:68] state_timeout(): Master - Timed out in state: 7
[  9046][I][BigRed2.ino:891] run(): Master Initialized
[  9081][I][BigRed2.ino:917] run(): Got 51 (BOOT), moving on
[ 10106][I][BigRed2.ino:949] run(): State WAIT_FOR_PLAY_PAUSE: Got msg_stop, sending msg_04
[ 10106][I][BigRed2.ino:964] run(): State: PAUSE
[ 11107][I][BigRed2.ino:1163] run(): LOST BLUETOOTH CONNECTION
[ 60853][I][BluetoothA2DPSink.cpp:365] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 60853][I][BluetoothA2DPSink.cpp:652] handle_connection_state(): partner address: f8:1a:2b:e9:cc:8e
[ 60860][I][BluetoothA2DPSink.cpp:660] handle_connection_state(): A2DP connection state: Disconnected, [f8:1a:2b:e9:cc:8e]
[ 60871][I][BluetoothA2DPSink.cpp:663] handle_connection_state(): ESP_A2D_CONNECTION_STATE_DISCONNECTED
[ 60880][I][BluetoothA2DPSink.cpp:1252] bt_i2s_task_shut_down(): BtI2STask shutdown
[ 60888][I][BluetoothA2DPSink.cpp:677] handle_connection_state(): i2s_stop
[ 60894][I][BluetoothA2DPSink.cpp:683] handle_connection_state(): try reconnect1
[ 60901][I][BluetoothA2DPSink.h:355] is_reconnect(): is_reconnect: false
[ 60908][I][BluetoothA2DPCommon.cpp:241] set_scan_mode_connectable(): set_scan_mode_connectable true
```

pschatzmann, 2022-06-30:
Can you try to find out what the value of a2d->conn_stat.disc_rsn is ?
is_reconnect() seems to return false instead of true...

clarkbremer, 2022-07-01:
I can do that tomorrow.  You want that in hex?

clarkbremer, 2022-07-01:
Actually, that was already in my logs: 
```
[ 60882][I][BluetoothA2DPSink.cpp:683] handle_connection_state(): try reconnect1.  disc_rsn: 0x0
```

And what about the 60 second delay before it even attempts the retry code?

clarkbremer, 2022-07-01:
Also, I don't want `afterNormalDisconnect = true`.  If the user initiates the disconnect from the phone, it immediately reconnects!  That's not the desired behavior.  This is for the case where the user shuts off the power to the ESP32 device - I want it to try to reconnect when the power comes back.

pschatzmann, 2022-07-01:
It's 1 sec and not 60 seconds and it seems that different partners are reporting different values for disc_rsn for this case: I guess that I need to change the approach. Fortunately I can just extend on the logic that was set in place for the A2DP Source lately...

clarkbremer, 2022-07-01:
It's definitely 60 seconds after boot when it first begins the retry logic.

clarkbremer, 2022-07-01:
What I'm saying is that when this fails, it's about 60 seconds after it boots before we get into the code with:
`if (a2d->conn_stat.state == ESP_A2D_CONNECTION_STATE_DISCONNECTED) {`

Note the timestamps in the logs:
```
[  4113][I][BigRed2.ino:1163] run(): LOST BLUETOOTH CONNECTION
[ 60841][I][BluetoothA2DPSink.cpp:365] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
```

My log message about "LOST" bluetooth might be a little misleading - we may not have had a connection in the first place.

pschatzmann, 2022-07-02:
I managed to get rid of the afterNormalDisconnect logic locally. However I am still struggling that sometimes it takes about a minute to reconnect.  I think I will need some more time to figure out how to address this....

clarkbremer, 2022-07-02:
Thanks, Phil.  It does the one-minute thing as it fails to connect on the first try.  Let me know if there's anything I can try on my end.

belm0, 2022-07-03:
Am I seeing the same issue, or shall I open a new report?

With a macOS peer, this is 100% reproducible:
   * reconnect happens fine when the sink device is started
   * but after a few seconds, there is a disconnect
   * if I then connect again after that, it stays connected

The disconnect looks like this at log INFO:
```
[ 18476][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 18476][I][BluetoothA2DPSink.cpp:621] handle_audio_state(): [BT_AV] A2DP audio state: Suspended
[ 18483][W][BluetoothA2DPSink.cpp:637] handle_audio_state(): [BT_AV] i2s_stop
[ 18490][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 18492][I][BluetoothA2DPSink.cpp:488] app_gap_callback(): [BT_AV] event: 13
[ 18500][I][BluetoothA2DPSink.cpp:793] av_hdl_avrc_evt(): [BT_AV] AVRC conn_state evt: state 0, [00:00:00:00:00:00]
[ 18507][I][BluetoothA2DPSink.cpp:488] app_gap_callback(): [BT_AV] event: 13
[ 18517][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 18534][I][BluetoothA2DPSink.cpp:1328] av_hdl_avrc_tg_evt(): [BT_AV] AVRC conn_state evt: state 0, [00:00:00:00:00:00]
[ 18544][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 18554][I][BluetoothA2DPSink.cpp:651] handle_connection_state(): [BT_AV] partner address: a4:d1:8c:c5:25:87
[ 18564][I][BluetoothA2DPSink.cpp:659] handle_connection_state(): [BT_AV] A2DP connection state: Disconnected, [a4:d1:8c:c5:25:87]
[ 18575][I][BluetoothA2DPSink.cpp:662] handle_connection_state(): [BT_AV] ESP_A2D_CONNECTION_STATE_DISCONNECTED
[ 18585][I][BluetoothA2DPSink.cpp:676] handle_connection_state(): [BT_AV] i2s_stop
[ 18593][I][BluetoothA2DPSink.h:346] is_reconnect(): [BT_AV] is_reconnect: false
[ 18600][I][BluetoothA2DPCommon.cpp:245] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
```

For my use case, I don't want to rely on auto-reconnect for problems like this.  Nothing should interrupt the stream after the initial connect.

pschatzmann, 2022-07-03:
I had a closer look and came to the following conclusions. The reconnect to Android seems to be reliable.
The reconnect to IOS works sometimes fast, sometimes it's slow (around 1 minute). The issue seems to be with IOS since it takes that much time to send out the necessary confirmations and the ESP is just waiting for these messages:
 I tried the following
- send a disconnect just after the startup  to inform the target device that it should be disconnected 
- reprocess reconnects() if the target device does not react within some defined time
None of this was helping...

clarkbremer, 2022-07-03:
FYI, I'm only using Android.

sennt frommy mobile phone with a tinny keybard

pschatzmann, 2022-07-04:
I committed my latest corrections, but I don't think that they will bring any improvements. The only advise I would have is to make sure that you use the latest ESP IDF/Arduino version.

pschatzmann, 2022-07-04:
If you use Arduino you can upgrade the ESP32 version in the board manager.

clarkbremer, 2022-07-05:
Yes, I'm on 2.0.3.  I'm not bothered that the first connection attempt sometimes fails - that can happen.  I just wish it didn't take 60 seconds to recognize that.  It would be better if we were notified of the failure after, say, 10 seconds, so we could initiate our retry sooner.

pschatzmann, 2022-07-05:
That's exactly what I have have implemented and in my case a retry did not help at all! 
It really took the audio source that amount of time to provide a confirmation...

clarkbremer, 2022-07-05:
OK,  Let me experiment a little with your latest changes.

pschatzmann, 2022-07-05:
Have a look at A2DP_EXPERIMENTAL_RECONNECT_AFTER_IDLE and change the portMAX_DELAY to e.g. 10 seconds

pschatzmann, 2022-07-05:
That would be my expected outcome if you don't change portMAX_DELAY to e.g. 10000

clarkbremer, 2022-07-05:
I looked, but couldn't find where that was defined?  `portMAX_DELAY`

pschatzmann, 2022-07-05:
```
void BluetoothA2DPSink::app_task_handler(void *arg)
{
    ESP_LOGD(BT_AV_TAG, "%s", __func__);
    app_msg_t msg;
    while (true) {
        if (!app_task_queue){
            ESP_LOGE(BT_APP_TAG, "%s, app_task_queue is null", __func__);
            delay(100);
        } else if (pdTRUE == xQueueReceive(app_task_queue, &msg, (portTickType)portMAX_DELAY)) {
            ESP_LOGD(BT_APP_TAG, "%s, sig 0x%x, 0x%x", __func__, msg.sig, msg.event);
```
You would replace portMAX_DELAY with your timeout setting....

clarkbremer, 2022-07-05:
OK.  I think I've replicated what you did, with `portMAX_DELAY ` set at 10000 (10 seconds).  It now attempts to reconnect, but until the ` handle_connection_state(): [BT_AV] A2DP connection state: Disconnected, [f8:1a:2b:e9:cc:8e]` connection state change event happens, it's doesn't work.  Once we get that event, the reconnect does work.  So that 60 seconds until we get the connection state change event (disconnected) - is that configured somewhere deep in the espressif code?

pschatzmann, 2022-07-05:
My understanding would be that this is in the target device and has nothing to do with the ESP32

clarkbremer, 2022-07-05:
Maybe.  I found this to be repeatable:

- Connect the device as normal.
- Power-down the ESP32.  It takes about 10 sec for the phone to notice and show the dropped connection.
- If I wait that 10 seconds before I power on the ESP32, it seems to reconnect correctly every time.  (preliminary - more tests needed).
- If I do NOT wait that 10 seconds before I power on the ESP32, the initial connection fails (the phone still thinks it's connected).  
- If I reboot again soon after (LONG before the 60 second timer would have expired), it connects successfully.

So, in general, a simple reboot will always lead to this condition (because we don't wait for the phone to notice).  I think that's new behavior.

clarkbremer, 2022-07-05:
The phone knows it's disconnected, and is willing to accept the reconnect *before* the 60 seconds timeout has elapsed.  Which makes me think that timeout is not in the phone.

clarkbremer, 2022-07-05:
Also, If I disable bluetooth on my phone halfway through the 60-second timeout, We still don't see
`[ 60951][I][BluetoothA2DPSink.cpp:690] handle_connection_state(): [BT_AV] A2DP connection state: Disconnected` 
until 60 seconds have elapsed.

pschatzmann, 2022-07-05:
You can double check if A2DP_EXPERIMENTAL_DISCONNECT_ON_START is helping.
For me it did not...

clarkbremer, 2022-07-05:
Nope.  So I still think the 60 second timeout is somewhere in the ESP code - probably in the espressif code.

pschatzmann, 2022-07-05:
> * If I wait that 10 seconds before I power on the ESP32, it seems to reconnect correctly every time.  (preliminary - more tests needed).

If this is consistent, the best way to deal with this would be to add a 15 second delay after a reconnect restart - before we start the app_work_dispatch queue

clarkbremer, 2022-07-05:
Is that stored somewhere in NV ram?  That when we powered down, we were connected to something?

clarkbremer, 2022-07-05:
If I set `portMAX_DELAY` to 10 seconds, a reboot gets us into the bad state, but we eventually do reconnect after 70 seconds.

pschatzmann, 2022-07-05:
We only store the partner address to which we send the connect request...

pschatzmann, 2022-07-05:
To test I suggest that you add a delay(15000); just before or after

#ifdef A2DP_EXPERIMENTAL_DISCONNECT_ON_START
        ESP_LOGW(BT_AV_TAG,"A2DP_EXPERIMENTAL_DISCONNECT_ON_START");
        disconnect();
#endif

We can manage the delay configurable....

clarkbremer, 2022-07-05:
Also, would be nice to have `a2dp_sink.is_connecting();`, so I could at least give my users feedback about why nothing is happening.

clarkbremer, 2022-07-05:
I'm also pretty certain this is new behavior.  It fails every other reset.  I would have noticed that before.

pschatzmann, 2022-07-05:
There used to be a  hard coded 1 second delay which I removed with one of the last commits. I added it back and  with the last commit there is the set_reconnect_delay() method.

clarkbremer, 2022-07-05:
Same.  But I think I may have a work-around for my situation.  In the car (but not the lab), I do get advanced warning that the power is going to go down.  So I can issue a `disconnect()` just before I die.  That should enable a fast reboot.

clarkbremer, 2022-07-06:
This is working pretty well for me now, with default settings.  More pressing is #256 .

mitchjs, 2022-07-06:
i can jump in and do testing... i havent really started my project with this yet
but... running just basically this   a2dp_sink.start("MyMusic", true);  
and testing with iOS 15.5 on iPhone 11

when i turn off the ESP32, the iPhone picks it up, and of course disconnects and pauses playback
when i apply power to esp32, the iPhone reconnects
i even waited a few mins between power cycle
i have not got any connection failures
im using A2DP code before the july4/5 changes
i can pull the updates, and test too

(not to switch topics here)
any easy way to resume play, if it was playing before? and making sure its from the reconnect
maybe add option to audio resume? or there a simple way?
could we store last play state, like last connection is stored?

clarkbremer, 2022-07-06:
Mitch, have you tried a simple reset of the esp32?  Don't wait for the phone to notice, just off and on.  

Regarding auto-play, in my system, there's another component (the head unit in the car) that keeps track of that.

mitchjs, 2022-07-06:
ah ok, just quick resets...

```
I (1172) BT_AV: partner address: 88:64:40:2e:69:b0
I (1182) BT_AV: A2DP connection state: Connecting, [88:64:40:2e:69:b0]
I (1192) BT_AV: ESP_A2D_CONNECTION_STATE_CONNECTING
```

but iphone isnt reconnected
after some time

```
W (61162) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 2
I (61162) BT_API: app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
I (61162) BT_AV: partner address: 88:64:40:2e:69:b0
I (61172) BT_AV: A2DP connection state: Disconnected, [88:64:40:2e:69:b0]
I (61182) BT_AV: ESP_A2D_CONNECTION_STATE_DISCONNECTED
I (61182) BT_AV: BtI2STask shutdown
I (61192) BT_AV: i2s_stop
I (61192) BT_AV: is_reconnect: false
I (61192) BT_AV: set_scan_mode_connectable true
```

this the issue (im still running code before 7/4 & 7/5 commits)
should i switch to that code for the "improved" reconnect logic?

clarkbremer, 2022-07-06:
It's working much better for me now.  I think it's the presumptive `disconnect()` [here](https://github.com/pschatzmann/ESP32-A2DP/commit/3e9fea4f2bacff648dc037ca5d91b842bd73d128#diff-5127d0fcdbea9a029efde708f2d4610c79bfca6aa591b23de6d7478be559d52eR152) that made the difference.

pschatzmann, 2022-07-14:
I did some tests today and could not find any reconnect fails. You can reopen the issue if you think it is still not OK.

closed 2022-07-14 by pschatzmann