pschatzmann/ESP32-A2DP #250: Tasks, callback functions and Processor Affinity

generic-beat-detector, 2022-07-01:
Hello, I'm assuming the "task context" of the user callback function specified via `set_stream_reader()`, as well as the internal task that writes to the i2s pins have no particular CPU affinity? I see they are created via `xTaskCreate()` in `BluetoothA2DPSink.cpp`? Is this accurate?

I ask because I intend to perform some fast and furious FFT, and was thinking of reserving/pinning, say, CPU 1 for the signal analysis.

pschatzmann, 2022-07-01:
I would expect that the data callback is part of the ESP32 Bluetooth processing which is on core0.
The BtI2STask task is not pinned to any core but I would expect that it is usually allocated to core1, but please double check with the help of xPortGetCoreID()

generic-beat-detector, 2022-07-01:
@pschatzmann "... please double check with the help of xPortGetCoreID()" ... heheh, of course! How embarrassing! Best regards.

closed 2022-07-01 by generic-beat-detector