pschatzmann/ESP32-A2DP #251: compile warnings

belm0, 2022-07-02:
are these worth cleaning up?

```
.../ESP32-A2DP/src/BluetoothA2DPCommon.cpp: In member function 'virtual void BluetoothA2DPCommon::get_last_connection()':
.../ESP32-A2DP/src/BluetoothA2DPCommon.cpp:140:10: warning: unused variable 'bda_str' [-Wunused-variable]
     char bda_str[18];
          ^~~~~~~
Compiling .pio/build/m5stack-core2/lib624/ESP32-A2DP/BluetoothA2DPSource.cpp.o
In file included from .../ESP32-A2DP/src/BluetoothA2DPSink.cpp:16:
.../ESP32-A2DP/src/BluetoothA2DPSink.h: In member function 'virtual bool BluetoothA2DPSink::is_reconnect(esp_a2d_disc_rsn_t)':
.../ESP32-A2DP/src/BluetoothA2DPSink.h:345:110: warning: suggest parentheses around '&&' within '||' [-Wparentheses]
         bool result = reconnect_on_normal_disconnect || type==ESP_A2D_DISC_RSN_ABNORMAL && is_auto_reconnect && has_last_connection();
                                                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
.../ESP32-A2DP/src/BluetoothA2DPSink.cpp: In member function 'virtual void BluetoothA2DPSink::handle_connection_state(uint16_t, void*)':
.../ESP32-A2DP/src/BluetoothA2DPSink.cpp:739:23: warning: unused variable 'esp_err' [-Wunused-variable]
             esp_err_t esp_err = esp_bt_gap_read_remote_name(a2d->conn_stat.remote_bda);
                       ^~~~~~~
Archiving .pio/build/m5stack-core2/libfdb/libtgx.a
.../ESP32-A2DP/src/BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::reset_last_connection()':
.../ESP32-A2DP/src/BluetoothA2DPSource.cpp:236:17: warning: unused variable 'bda_str' [-Wunused-variable]
     const char *bda_str = to_str(last_connection);
                 ^~~~~~~
.../ESP32-A2DP/src/BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::filter_inquiry_scan_result(esp_bt_gap_cb_param_t*)':
.../ESP32-A2DP/src/BluetoothA2DPSource.cpp:379:13: warning: variable 'rssi' set but not used [-Wunused-but-set-variable]
     int32_t rssi = -129; /* invalid value */
             ^~~~
Indexing .pio/build/m5stack-core2/libfdb/libtgx.a
.../ESP32-A2DP/src/BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::bt_app_av_state_unconnected(uint16_t, void*)':
.../ESP32-A2DP/src/BluetoothA2DPSource.cpp:668:26: warning: unused variable 'p' [-Wunused-variable]
                 uint8_t *p = peer_bd_addr;
                          ^
.../ESP32-A2DP/src/BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::bt_av_hdl_avrc_ct_evt(uint16_t, void*)':
.../ESP32-A2DP/src/BluetoothA2DPSource.cpp:889:22: warning: unused variable 'bda' [-Wunused-variable]
             uint8_t *bda = rc->conn_stat.remote_bda;
                      ^~~
```

belm0, 2022-07-02:
Warnings from headers are a little annoying-- if a project happens to use `-Wall` it will get a warning on every compile.

Happy to make a PR!

```
In file included from src/main.cpp:9:
.../ESP32-A2DP/src/BluetoothA2DPSink.h: In member function 'virtual bool BluetoothA2DPSink::is_reconnect(esp_a2d_disc_rsn_t)':
.../ESP32-A2DP/src/BluetoothA2DPSink.h:345:110: warning: suggest parentheses around '&&' within '||' [-Wparentheses]
         bool result = reconnect_on_normal_disconnect || type==ESP_A2D_DISC_RSN_ABNORMAL && is_auto_reconnect && has_last_connection();
```

mitchjs, 2022-07-02:
only "warning" i got was and it was an error error and simple fix

in virtual bool is_reconnect(esp_a2d_disc_rsn_t type)():

` bool result = reconnect_on_normal_disconnect || (type==ESP_A2D_DISC_RSN_ABNORMAL && is_auto_reconnect && has_last_connection());`

had to add set of ( ) around the multiple && group

pschatzmann, 2022-07-02:
Please note that this logic is about to change anyhow in order to address https://github.com/pschatzmann/ESP32-A2DP/issues/248#issuecomment-1172524102

mitchjs, 2022-07-03:
yea i knew that was coming :) but mentioned it if anyone needed to compile 1.7.1

pschatzmann, 2022-07-04:
I tried to addess some of the issues. The critical thing from my point of view is, that we do not get any unnecessary Compile warnings in Arduino. In all other environments that can be controlled by using compile parameters.

If I missed anything feel free to submit a PR...

pschatzmann, 2022-07-05:
No you can't: However I noticed that the unused variable Warning is in some cases just wrong...

pschatzmann, 2022-07-14:
Should be closed with the last commit. I tested with Arduino compiler warnings All.
Please feel free to submit a PR if you find futher issues.

closed 2022-07-14 by pschatzmann