pschatzmann/ESP32-A2DP #253: ringbuffer I2S output doesn't work

belm0, 2022-07-02:
there is no I2S output when using BluetoothA2DPSink as of v1.7.1 (and head), due to the ringbuffer PR #240

using BluetoothA2DPSinkMinRAM works

device:  M5Stack Core2

mitchjs, 2022-07-02:
im running it right now on my esp32 devi board with i2s connected to a PCM5102A board
and a little powered speaker

my only code in setup() is:

```
 i2s_pin_config_t const i2sportConfig  = {
    .bck_io_num = 26,
    .ws_io_num = 22,
    .data_out_num = 25,
    .data_in_num = I2S_PIN_NO_CHANGE
  };

  a2dp_sink.set_pin_config(i2sportConfig);
  a2dp_sink.set_volume(15);
  a2dp_sink.start("MyMusic", true);  
```

try it as a stand alone test
maybe you dont got enough ram, i would expect you see an error from log if thats the case

pschatzmann, 2022-07-02:
I was testing with the bt_music_receiver_simple on a cheap ESP32 Dev board and I did not have any issues. 
If you have any problems, I guess this must be related to the M5Stack. When you compile the sketch do not forget to activate the logging so that you can see any warnings or error messages when the sketch is executed.

belm0, 2022-07-02:
Sorry I haven't provided anything to help with diagnosing yet.  It took me several hours as it was to narrow things down to the ringbuffer PR.

> I guess this must be related to the M5Stack

I wouldn't expect so, since I2S is part of the ESP module, and whatever the M5Stack device has physically connected to I2S works in other cases.

Has the ringbuffer been confirmed to work on devices having PSRAM?  I wonder if it could be about the implementation explicitly declaring the memory attributes of any buffers.

Another difference could be platform software version.

Anyway I'll check logs, etc. when I get a chance.

pschatzmann, 2022-07-03:
My gut feeling is that you might not have enough RAM available to allocate the Ringbuffer. Try to set it to a smaller value then the default (4K): with set_i2s_ringbuffer_size().

If this is a case you should see a xRingbufferCreate error message in your logs.

Currently there is no logic implemented which would use PSRAM.

belm0, 2022-07-03:
> My gut feeling is that you might not have enough RAM available to allocate the Ringbuffer.

it's an otherwise empty sketch, and the device has 520K SRAM

I didn't find anything interesting in the debug/verbose log output.  It's just silently failing.

<details>
<summary>log output</summary>

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
[ ⸮⸮mum⸮⸮⸮⸮2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   846][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
axp: vbus limit off
axp: gpio1 init
axp: gpio2 init
axp: rtc battery charging enabled
axp: esp32 power voltage was set to 3.35v
axp: lcd backlight voltage was set to 2.80v
axp: lcd logic and sdcard voltage preset to 3.3v
axp: vibrator voltage preset to 2v
[  1292][I][BluetoothA2DPSink.cpp:1050] set_volume(): [BT_AV] set_volume 80
[  1293][I][BluetoothA2DPSink.cpp:1294] volume_set_by_local_host(): [BT_AV] Volume is set locally to: 62
[  1297][D][BluetoothA2DPSink.cpp:127] start(): [BT_AV] start
[  1303][I][BluetoothA2DPCommon.cpp:211] log_free_heap(): [BT_AV] Available Heap: 4417135
[  1311][I][BluetoothA2DPSink.cpp:140] start(): [BT_AV] Device name will be set to 'xx'
[  1319][D][BluetoothA2DPSink.cpp:996] init_nvs(): [BT_AV] init_nvs
[  1325][D][BluetoothA2DPCommon.cpp:117] get_last_connection(): [BT_AV] get_last_connection
[  1334][D][BluetoothA2DPCommon.cpp:141] get_last_connection(): [BT_AV] => a4:d1:8c:c5:25:87
[  1341][I][BluetoothA2DPSink.cpp:188] init_i2s(): [BT_AV] init_i2s
[  1347][I][BluetoothA2DPSink.cpp:190] init_i2s(): [BT_AV] init_i2s is_i2s_output
[  1356][D][BluetoothA2DPSink.cpp:262] init_bluetooth(): [BT_AV] init_bluetooth
[  1451][I][BluetoothA2DPSink.cpp:267] init_bluetooth(): [BT_AV] controller initialized
[  1769][I][BluetoothA2DPSink.cpp:276] init_bluetooth(): [BT_AV] bluedroid initialized
[  1881][I][BluetoothA2DPSink.cpp:285] init_bluetooth(): [BT_AV] bluedroid enabled
[  1881][D][BluetoothA2DPSink.cpp:383] app_task_start_up(): [BT_AV] app_task_start_up
[  1885][D][BluetoothA2DPSink.cpp:306] app_work_dispatch(): [BT_API] app_work_dispatch event 0x0, param len 0
[  1885][D][BluetoothA2DPSink.cpp:1115] ccall_app_task_handler(): [BT_AV] ccall_app_task_handler
[  1903][D][BluetoothA2DPSink.cpp:354] app_task_handler(): [BT_AV] app_task_handler
[  1910][D][BluetoothA2DPSink.cpp:338] app_send_msg(): [BT_AV] app_send_msg
[  1917][D][BluetoothA2DPSink.cpp:361] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x0
[  1925][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1935][D][BluetoothA2DPSink.cpp:329] app_work_dispatched(): [BT_AV] app_work_dispatched
[  1943][D][BluetoothA2DPSink.cpp:1145] ccall_av_hdl_stack_evt(): [BT_AV] ccall_av_hdl_stack_evt
[  1952][D][BluetoothA2DPSink.cpp:859] av_hdl_stack_evt(): [BT_AV] av_hdl_stack_evt evt 0
[  1960][D][BluetoothA2DPSink.cpp:864] av_hdl_stack_evt(): [BT_AV] av_hdl_stack_evt av_hdl_stack_evt BT_APP_EVT_STACK_UP
[  1970][D][BluetoothA2DPSink.cpp:873] av_hdl_stack_evt(): [BT_AV] AVRCP controller initialized!
[  1972][D][BluetoothA2DPSink.cpp:1139] ccall_app_gap_callback(): [BT_AV] ccall_app_gap_callback
[  1979][D][BluetoothA2DPSink.cpp:908] av_hdl_stack_evt(): [BT_AV] connect_to_last_device
[  1987][I][BluetoothA2DPSink.cpp:488] app_gap_callback(): [BT_AV] event: 10
[  1995][D][BluetoothA2DPCommon.cpp:181] connect_to_last_device(): [BT_AV] connect_to_last_device
[  2011][D][BluetoothA2DPSink.cpp:913] av_hdl_stack_evt(): [BT_AV] set_scan_mode_connectable(true)
[  2019][I][BluetoothA2DPCommon.cpp:245] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
[  2014][D][BluetoothA2DPSink.cpp:1127] ccall_app_a2d_callback(): [BT_AV] ccall_app_a2d_callback
[  1979][I][BluetoothA2DPCommon.cpp:211] log_free_heap(): [BT_AV] Available Heap: 4328223
[  2037][D][BluetoothA2DPSink.cpp:929] app_a2d_callback(): [BT_AV] app_a2d_callback
[  2053][D][BluetoothA2DPSink.cpp:949] app_a2d_callback(): [BT_AV] app_a2d_callback ESP_A2D_AUDIO_CFG_EVT
[  2062][D][BluetoothA2DPSink.cpp:306] app_work_dispatch(): [BT_API] app_work_dispatch event 0x4, param len 16
[  2072][D][BluetoothA2DPSink.cpp:338] app_send_msg(): [BT_AV] app_send_msg
[  2078][D][BluetoothA2DPSink.cpp:361] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x4
[  2079][D][BluetoothA2DPSink.cpp:1139] ccall_app_gap_callback(): [BT_AV] ccall_app_gap_callback
[  2087][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  2095][I][BluetoothA2DPSink.cpp:488] app_gap_callback(): [BT_AV] event: 10
[... 212 lines cut ...]
[  5326][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  5336][D][BluetoothA2DPSink.cpp:329] app_work_dispatched(): [BT_AV] app_work_dispatched
[  5344][D][BluetoothA2DPSink.cpp:1314] ccall_av_hdl_avrc_tg_evt(): [BT_AV] ccall_av_hdl_avrc_tg_evt
[  5353][D][BluetoothA2DPSink.cpp:1322] av_hdl_avrc_tg_evt(): [BT_AV] av_hdl_avrc_tg_evt evt 4
[  5361][I][BluetoothA2DPSink.cpp:1344] av_hdl_avrc_tg_evt(): [BT_AV] AVRC register event notification: 13, param: 0x0
[  5372][I][BluetoothA2DPSink.cpp:1346] av_hdl_avrc_tg_evt(): [BT_AV] AVRC Volume Changes Supported
[... cut ...]
[  5470][I][BluetoothA2DPSink.cpp:1338] av_hdl_avrc_tg_evt(): [BT_AV] AVRC set absolute volume: 62%
[  5479][I][BluetoothA2DPSink.cpp:1278] volume_set_by_controller(): [BT_AV] Volume is set by remote controller to 62
[  5527][D][BluetoothA2DPSink.cpp:962] audio_data_callback(): [BT_AV] audio_data_callback
[  5534][D][BluetoothA2DPSink.cpp:962] audio_data_callback(): [BT_AV] audio_data_callback
[  5540][D][BluetoothA2DPSink.cpp:962] audio_data_callback(): [BT_AV] audio_data_callback
```

</details>

I tried with "PSRAM disabled" option, no change.

mitchjs, 2022-07-03:
add some loggging in i2s_write_data() to make sure its making it there
right after the i2s_write
something like
```
static uint32_t u32logRate = 0;
if (++u32logRate % 100 == 0) {
    ESP_LOGI(BT_AV_TAG, "i2s_write_data(): i2s_bytes_written=%u", i2s_bytes_written);
}
```
just to be sure the i2s writes are happening

i get: 
```
I (11272) BT_AV: A2DP audio state: Started
I (11282) BT_AV: i2s_start
I (13712) BT_AV: i2s_write_data(): i2s_bytes_written=4096
I (16032) BT_AV: i2s_write_data(): i2s_bytes_written=4096
```

pschatzmann, 2022-07-04:
Today I did some tests related to some other issues and started to see the same issue as well...

mitchjs, 2022-07-08:
so i got my android tablet out...
and connected it up
and log spews "W (1454113) BT_APPL: Pkt dropped"

so something is now broken:

```
I (1451693) BT_AV: BtI2STask Started
I (1451693) BT_AV: i2s_start
W (1452213) BT_APPL: Pkt dropped
..
W (1452213) BT_APPL: Pkt dropped.

then worse... 

W (1524673) BT_HCI: DiscCmpl evt: hdl=129, rsn=0x8
E (1524673) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=7
W (1524673) BT_AVCT: ccb 1 not allocated

and that is that,,, nothing connects to it now... i have to reset
going back to my iphone... all good

```

need to step back, as i know this worked before

OK, so i loaded the Espessif A2DP Sample code (and they updated it with corrected task priorities)
NO ISSUES WHAT SO EVER

SO... something is a mess now
maybe its me... not sure now... going crazy :)

mitchjs, 2022-07-08:
ok, i think i found the issue...
its where thebt_i2s_task_start_up() and bt_i2s_task_shut_down() are called from
they (Espressiff calls them in the ESP_A2D_CONNECTION_STATE_CONNECTED / ESP_A2D_CONNECTION_STATE_DISCONNECTED
event handlers
which is where they were before some of the last commits

so i commented them out  lines 658,668

and put it back in line 713
```
if (is_i2s_output) {
    bt_i2s_task_shut_down();
}
```

and back at line 777
```
if (is_i2s_output) {
    bt_i2s_task_start_up();
}
```

and now working smooth on my android tablet and iphone
the android tablet seems to send out audio at times, when not evening specifically playing it
like its sending its "system" sounds, (click sounds when i open app etc)
yup, just opening youtube app

```
I (17563) BT_AV: i2s_start
E (18043) BT_APPL: bta_dm_pm_btm_status hci_status=12
I (18043) BT_AV: event: 13
I (18043) BT_AV: event: 13
I (18153) BT_AV: i2s_task_handler->2560
I (20173) BT_AV: i2s_task_handler->2560
I (20183) BT_AV: i2s_task_handler->512
I (20193) BT_AV: i2s_task_handler->2048
I (20203) BT_AV: i2s_task_handler->2048
I (20203) BT_AV: i2s_task_handler->512
I (20223) BT_AV: i2s_task_handler->2560
I (20223) BT_AV: i2s_task_handler->1024
I (20233) BT_AV: i2s_task_handler->1536
I (20253) BT_AV: i2s_task_handler->2560
I (20263) BT_AV: i2s_task_handler->2560
I (20273) BT_AV: i2s_task_handler->1536
I (20273) BT_AV: i2s_task_handler->1024
I (20293) BT_AV: i2s_task_handler->2560
I (20293) BT_AV: i2s_task_handler->512
I (20303) BT_AV: i2s_task_handler->2048
I (20323) BT_AV: i2s_task_handler->2048
I (20323) BT_AV: i2s_task_handler->512
I (20333) BT_AV: i2s_task_handler->2560
I (20343) BT_AV: i2s_task_handler->1024
I (20353) BT_AV: i2s_task_handler->1536
I (20363) BT_AV: i2s_task_handler->2560
I (20383) BT_AV: i2s_task_handler->2560
I (20383) BT_API: app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
I (20383) BT_AV: A2DP audio state: Suspended
W (20383) BT_AV: i2s_stop
I (23543) BT_AV: event: 13
```

gonna change that log to LOGD :)

i can make a PR (if needs be), with the changes i did, its just where the task is started/shutdown

pschatzmann, 2022-07-08:
Yes, it would be helpful to get a PR. Unfortunately I was busy this week and had no time to work on this...

pschatzmann, 2022-07-14:
The issue has been corrected with the latest commits:

Please not that with my latest commit
- BluetoothA2DPSink is working w/o separate I2S task 
- BluetoothA2DPSinkQueued is working with a separate queue and a task for i2s output

closed 2022-07-14 by pschatzmann