pschatzmann/ESP32-A2DP #254: MacOS peer always disconnects 15 seconds after initial connection

belm0, 2022-07-03:
With a macOS peer that has previously been connected, this is 100% reproducible:

  * reconnect to last peer happens fine when the ESP32 sink device is started
  * but after exactly 15 seconds, there is a disconnect
  * if I then connect again after that, it stays connected

<details>

<summary>INFO logs (disconnect is at time 21757)</summary>

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
[   845][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
axp: vbus limit off
axp: gpio1 init
axp: gpio2 init
axp: rtc battery charging enabled
axp: esp32 power voltage was set to 3.35v
axp: lcd backlight voltage was set to 2.80v
axp: lcd logic and sdcard voltage preset to 3.3v
axp: vibrator voltage preset to 2v
[  1290][I][BluetoothA2DPSink.cpp:1050] set_volume(): [BT_AV] set_volume 80
[  1291][I][BluetoothA2DPSink.cpp:1294] volume_set_by_local_host(): [BT_AV] Volume is set locally to: 62
[  1295][I][BluetoothA2DPCommon.cpp:211] log_free_heap(): [BT_AV] Available Heap: 4417135
[  1303][I][BluetoothA2DPSink.cpp:140] start(): [BT_AV] Device name will be set to 'xx'
[  1313][I][BluetoothA2DPSink.cpp:188] init_i2s(): [BT_AV] init_i2s
[  1317][I][BluetoothA2DPSink.cpp:190] init_i2s(): [BT_AV] init_i2s is_i2s_output
[  1417][I][BluetoothA2DPSink.cpp:267] init_bluetooth(): [BT_AV] controller initialized
[  1739][I][BluetoothA2DPSink.cpp:276] init_bluetooth(): [BT_AV] bluedroid initialized
[  1848][I][BluetoothA2DPSink.cpp:285] init_bluetooth(): [BT_AV] bluedroid enabled
[  1849][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1855][I][BluetoothA2DPCommon.cpp:245] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
[  1865][I][BluetoothA2DPCommon.cpp:211] log_free_heap(): [BT_AV] Available Heap: 4316263
[  1875][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1877][I][BluetoothA2DPSink.cpp:488] app_gap_callback(): [BT_AV] event: 10
[  1882][I][BluetoothA2DPSink.cpp:565] av_hdl_a2d_evt(): [BT_AV] A2DP PROF STATE: Init Compl

[  1893][I][BluetoothA2DPSink.cpp:488] app_gap_callback(): [BT_AV] event: 10
[  1904][I][BluetoothA2DPSink.cpp:488] app_gap_callback(): [BT_AV] event: 10
[  1911][I][BluetoothA2DPSink.cpp:488] app_gap_callback(): [BT_AV] event: 10
[  1922][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  1928][I][BluetoothA2DPSink.cpp:651] handle_connection_state(): [BT_AV] partner address: a4:d1:8c:c5:25:87
[  1937][I][BluetoothA2DPSink.cpp:659] handle_connection_state(): [BT_AV] A2DP connection state: Connecting, [a4:d1:8c:c5:25:87]
[  1948][I][BluetoothA2DPSink.cpp:751] handle_connection_state(): [BT_AV] ESP_A2D_CONNECTION_STATE_CONNECTING
[  5495][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  5495][I][BluetoothA2DPSink.cpp:583] handle_audio_cfg(): [BT_AV] a2dp audio_cfg_cb , codec type 0
[  5503][I][BluetoothA2DPSink.cpp:595] handle_audio_cfg(): [BT_AV] a2dp audio_cfg_cb , [  5512][I][BluetoothA2DPSink.cpp:606] handle_audio_cfg(): [BT_AV] configure audio player 11-15-2-2a
[  5521][I][BluetoothA2DPSink.cpp:612] handle_audio_cfg(): [BT_AV] audio player configured, samplerate=48000
[  5643][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  5644][I][BluetoothA2DPSink.cpp:651] handle_connection_state(): [BT_AV] partner address: a4:d1:8c:c5:25:87
[  5652][I][BluetoothA2DPSink.cpp:659] handle_connection_state(): [BT_AV] A2DP connection state: Connected, [a4:d1:8c:c5:25:87]
[  5663][I][BluetoothA2DPSink.cpp:703] handle_connection_state(): [BT_AV] ESP_A2D_CONNECTION_STATE_CONNECTED
[  5673][I][BluetoothA2DPCommon.cpp:245] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable false
[  5682][I][BluetoothA2DPSink.cpp:728] handle_connection_state(): [BT_AV] i2s_start
[  5734][I][BluetoothA2DPSink.cpp:478] app_gap_callback(): [BT_AV] ESP_BT_GAP_READ_REMOTE_NAME_EVT stat:0
[  5735][I][BluetoothA2DPSink.cpp:480] app_gap_callback(): [BT_AV] ESP_BT_GAP_READ_REMOTE_NAME_EVT remote name:MacBook (2)
[  5936][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  5936][I][BluetoothA2DPSink.cpp:793] av_hdl_avrc_evt(): [BT_AV] AVRC conn_state evt: state 1, [a4:d1:8c:c5:25:87]
[  5945][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  5955][I][BluetoothA2DPSink.cpp:1328] av_hdl_avrc_tg_evt(): [BT_AV] AVRC conn_state evt: state 1, [a4:d1:8c:c5:25:87]
[  5966][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  5976][I][BluetoothA2DPSink.cpp:832] av_hdl_avrc_evt(): [BT_AV] AVRC remote features 24b
[  5984][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[... 50 lines cut ...]
[ 21874][I][BluetoothA2DPSink.h:346] is_reconnect(): [BT_AV] is_reconnect: false
[ 21881][I][BluetoothA2DPCommon.cpp:245] set_scan_mode_connectable(): [BT_AV] set_scan_mode_connectable true
```

</details>

Here is the interesting part of two separate runs, showing the disconnect is always after 15 seconds:

```
[  6430][I][BluetoothA2DPSink.cpp:817] av_hdl_avrc_evt(): [BT_AV] AVRC metadata rsp: attribute id 0x20, 
[  6458][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  6458][I][BluetoothA2DPSink.cpp:1338] av_hdl_avrc_tg_evt(): [BT_AV] AVRC set absolute volume: 62%
[  6465][I][BluetoothA2DPSink.cpp:1278] volume_set_by_controller(): [BT_AV] Volume is set by remote controller to 62
[ 21757][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 21757][I][BluetoothA2DPSink.cpp:621] handle_audio_state(): [BT_AV] A2DP audio state: Suspended

[  3901][I][BluetoothA2DPSink.cpp:817] av_hdl_avrc_evt(): [BT_AV] AVRC metadata rsp: attribute id 0x20, 
[  3932][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[  3933][I][BluetoothA2DPSink.cpp:1338] av_hdl_avrc_tg_evt(): [BT_AV] AVRC set absolute volume: 62%
[  3940][I][BluetoothA2DPSink.cpp:1278] volume_set_by_controller(): [BT_AV] Volume is set by remote controller to 62
[ 19181][I][BluetoothA2DPSink.cpp:364] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 19182][I][BluetoothA2DPSink.cpp:621] handle_audio_state(): [BT_AV] A2DP audio state: Suspended
```

For my use case, I don't want to rely on auto-reconnect for problems like this. Normally nothing should interrupt the stream after the initial connect.

pschatzmann, 2022-07-04:
I could not reproduce this issue with my latest changes. However there is nothing we can do on our side if the macOS sends us a disconnect notification.
I suggest that you double check with the latest corrections and the latest ESP32 version (which is 2.0.3)

belm0, 2022-07-04:
I'm not sure which side is disconnecting-- is it apparent in the logs?  Even if macOS is disconnecting, it may be due to some incorrectness of ESP32-A2DP sink (such as not sending some expected response).

I'll confirm this doesn't happen with other sink devices.

I am on ESP32 2.0.3.

belm0, 2022-07-10:
I couldn't reproduce the disconnect with macOS and other sink devices.  So it seems there is something that the ESP32 framework or A2DP library could do better.

pschatzmann, 2022-07-14:
Just set the set_reconnect_delay to a higher value before you start a2dp. e.g. 
```
  a2dp_sink.set_reconnect_delay(3000);
```

closed 2022-07-14 by pschatzmann