pschatzmann/ESP32-A2DP #256: Audio does not restart after a pause

clarkbremer, 2022-07-05:
This happens a large percentage of the time, may 50%?  But not 100%.

If I am streaming audio, and I initiate a pause from my code, the audio stops as expected.  And the pause command makes it to my phone successfully.  But when I try to resume, I get no audio.  The play command does reach my phone successfully, but no audio.  If I then attempt to pause again, nothing happens, and the pause command does *not* reach my phone.  Gut feel is i2s ring buffer, but no evidence for that.

```
[ 22946][I][BigRed2.ino:1008] run(): Got msg_pause during play, Pausing
[ 22947][I][BigRed2.ino:964] run(): State: PAUSE
[ 23059][I][BluetoothA2DPSink.cpp:372] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 23060][I][BluetoothA2DPSink.cpp:869] av_hdl_avrc_evt(): [BT_AV] AVRC passthrough rsp: key_code 0x46, key_state 0
[ 23081][I][BluetoothA2DPSink.cpp:372] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 23081][I][BluetoothA2DPSink.cpp:869] av_hdl_avrc_evt(): [BT_AV] AVRC passthrough rsp: key_code 0x46, key_state 1
[ 26194][I][BluetoothA2DPSink.cpp:372] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 26194][I][BluetoothA2DPSink.cpp:645] handle_audio_state(): [BT_AV] A2DP audio state: Suspended
[ 26202][I][BigRed2.ino:480] audio_state_change(): ESP32 state changed to: 0
[ 26208][I][BigRed2.ino:489] audio_state_change(): ESP32:  Audio Paused
[ 26215][W][BluetoothA2DPSink.cpp:663] handle_audio_state(): [BT_AV] i2s_stop
[ 26222][I][BluetoothA2DPSink.cpp:1323] bt_i2s_task_shut_down(): [BT_AV] BtI2STask shutdown
[ 26194][I][BigRed2.ino:649] run(): LCD PLAY -> PAUSE
[ 29379][I][BluetoothA2DPSink.cpp:512] app_gap_callback(): [BT_AV] event: 13
[ 30104][I][BigRed2.ino:978] run(): State PAUSE: Got msg_play, moving to PLAY
[ 30104][I][BigRed2.ino:991] run(): PLAYING!!
[ 30142][I][BluetoothA2DPSink.cpp:372] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 30143][I][BluetoothA2DPSink.cpp:869] av_hdl_avrc_evt(): [BT_AV] AVRC passthrough rsp: key_code 0x44, key_state 0
[ 30194][I][BluetoothA2DPSink.cpp:372] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 30194][I][BluetoothA2DPSink.cpp:645] handle_audio_state(): [BT_AV] A2DP audio state: Started
[ 30201][I][BigRed2.ino:480] audio_state_change(): ESP32 state changed to: 2
[ 30208][I][BigRed2.ino:483] audio_state_change(): ESP32:  Audio Started
[ 30214][I][BluetoothA2DPSink.cpp:512] app_gap_callback(): [BT_AV] event: 13
[ 30215][I][BluetoothA2DPSink.cpp:1307] bt_i2s_task_start_up(): [BT_AV] BtI2STask Started
[ 30221][I][BluetoothA2DPSink.cpp:512] app_gap_callback(): [BT_AV] event: 13
[ 30229][I][BluetoothA2DPSink.cpp:657] handle_audio_state(): [BT_AV] i2s_start
[ 30243][I][BigRed2.ino:636] run(): LCD PAUSE -> PLAY
[ 30253][I][BluetoothA2DPSink.cpp:372] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 30258][I][BluetoothA2DPSink.cpp:869] av_hdl_avrc_evt(): [BT_AV] AVRC passthrough rsp: key_code 0x44, key_state 1
```

clarkbremer, 2022-07-06:
Tracked this down.  Introduced in ([Redesign Reconnect](https://github.com/pschatzmann/ESP32-A2DP/commit/3e9fea4f2bacff648dc037ca5d91b842bd73d128)

clarkbremer, 2022-07-06:
As far as I can tell, the logs look identical when it works vs when it fails.  Log level set to INFO.

pschatzmann, 2022-07-06:
I can reproduce the issue

clarkbremer, 2022-07-06:
Please change log statement on line 1239 to LOGD

pschatzmann, 2022-07-07:
I agree: LOGI is too high. The change has been comitted

pschatzmann, 2022-07-14:
This should be corrected in the latest version: See https://github.com/pschatzmann/ESP32-A2DP/pull/259

closed 2022-07-14 by pschatzmann