pschatzmann/ESP32-A2DP #258: Does not work in platformio when framework = espidf

fedorenchik, 2022-07-08:
After setting in `platformio.ini`

```
framework = espidf
```

it can build & upload to esp32 successfully, and there are no errors in serial monitor.

However, Bluetooth doesn't start (cannot find the device).

If add arduino component, it starts working with the same code.

fedorenchik, 2022-07-08:
Seems I found the issue. If change `btStart()` function to:

```
#ifdef CONFIG_BTDM_CONTROLLER_MODE_BTDM
#define BT_MODE ESP_BT_MODE_BTDM
#elif defined(CONFIG_BTDM_CONTROLLER_MODE_BR_EDR_ONLY)
#define BT_MODE ESP_BT_MODE_CLASSIC_BT
#else
#define BT_MODE ESP_BT_MODE_BLE
#endif
bool btStart(){
    esp_bt_controller_config_t cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED){
        return true;
    }
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_IDLE){
        esp_bt_controller_init(&cfg);
        while(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_IDLE){}
    }
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_INITED){
        if (esp_bt_controller_enable(BT_MODE)) {
            ESP_LOGE(BT_APP_TAG, "BT Enable failed");
            return false;
        }
    }
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED){
        return true;
    }
    ESP_LOGE(BT_APP_TAG, "BT Start failed");
    return false;
}
```

then it works. I use Dual Mode Bluetooth, but original function only supports Only Classic Bluetooth mode.

pschatzmann, 2022-07-09:
Are you sure that you used the correct menuconfig settings as descirbed in https://github.com/pschatzmann/ESP32-A2DP/wiki/Espressif-IDF-as-a-Component

fedorenchik, 2022-07-12:
My setup is a bit different:
1. I don't use Arduino (component or framework)
2. ESP32-A2DP as lib (in lib folder)
3. In `platformio.ini`: `framework = espidf`.

Yes, if following suggested config, there's no issue.

closed 2022-07-12 by fedorenchik

pschatzmann, 2022-07-12:
I did not test it, but I would expect that for this case the https://github.com/pschatzmann/ESP32-A2DP/wiki/Espressif-IDF-as-a-Component would apply and you need to make sure that the sdkconfig.defaults defined correctly.

fedorenchik, 2022-07-16:
Yes, need to set following config values then it's ok:
```
CONFIG_BTDM_CTRL_MODE_BLE_ONLY=n
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=y
CONFIG_BTDM_CTRL_MODE_BTDM=n
```