pschatzmann/ESP32-A2DP #260: Popping sound when using external DAC and stream reader

benjamin32561, 2022-07-12:
When using an external DAC (Adafuit / UDA1334A) and a function to read the streamed data the output sound has popping-like distortions (when not using the reader function the sound is clear).
The reader function doesn't change the audio data but has a delay, is there any way I can send the streamed data to another thread or just not wait for the reader function to finish before passing the audio data to the DAC?
Here is my code:
`static const i2s_config_t i2s_config = {
         .mode = (i2s_mode_t)(I2S_MODE_TX),
         .sample_rate = SAMPLING_FREQ,
         .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
         .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
         .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S),
         .intr_alloc_flags = 0, // default interrupt priority
         .dma_buf_count = 8,
         .dma_buf_len = 64,
         .use_apll = false,
         .tx_desc_auto_clear = true // avoiding noise in case of data unavailability
    };
  a2dp_sink.set_i2s_config(i2s_config);
  
  a2dp_sink.set_stream_reader(read_data);
  a2dp_sink.start("esp32");`

pschatzmann, 2022-07-12:
What exactly are you doing in the read_data ? You should not add any delays and it is critical that the data rate of the provided data is not smaller than the output sample rate.

Here is an example that is working: https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/basic-a2dp-i2s/basic-a2dp-i2s.ino

pschatzmann, 2022-07-12:
I don't think that this is the issue and this is an expected behaviour to synchronize the input rate with the output rate.
I suggest to try to increase the I2S buffer..

benjamin32561, 2022-07-12:
> I don't think that this is the issue - if you only do an i2s write - and this is an expected behaviour to synchronize the input rate with the output rate. I suggest to try to increase the I2S buffer..

F*ck yeah! it worked!!! thanks a lot! you saved me so much trouble...

closed 2022-07-14 by pschatzmann