pschatzmann/ESP32-A2DP #261: bursty sink callback?

belm0, 2022-07-13:
Is it expected that the sink stream callback is very bursty?  For example, at 48kHz stereo, the time intervals of callbacks are a pattern like (5, 5, 5, 5, 100, 5, 5, 5, 5, 100, ... ) in milliseconds.

I don't know if it's normal or if A2DP has any control of it, but disadvantages I can think of to being bursty:
  * the max buffering needed on the device is increased
  * the processing cycles for filtering or analysis of the stream is not smooth (because there are basically 4 callbacks within 20 ms, then a pause for 80 ms)
  * it seems to increase the risk of under-run.  When I observe a deficit of samples, it's always after the long callback interval.

<details>
<summary>callback log</summary>

```
bluetooth sample rate: 48000
[t=     5] (   5) stream rate 512000, excess  2320
[t=    10] (   5) stream rate 384000, excess  3360
[t=    15] (   5) stream rate 341333, excess  4400
[t=    21] (   6) stream rate 304761, excess  5392
[t=    26] (   5) stream rate 295384, excess  6432
[t=   146] ( 120) stream rate  61369, excess  1952
[t=   151] (   5) stream rate  67814, excess  2992
[t=   157] (   6) stream rate  73375, excess  3984
[t=   162] (   5) stream rate  79012, excess  5024
[t=   167] (   5) stream rate  84311, excess  6064
[t=   268] ( 101) stream rate  57313, excess  2496
[t=   274] (   6) stream rate  60729, excess  3488
[t=   279] (   5) stream rate  64229, excess  4528
[t=   284] (   5) stream rate  67605, excess  5568
[t=   290] (   6) stream rate  70620, excess  6560
[t=   398] ( 108) stream rate  54673, excess  2656
[t=   404] (   6) stream rate  57029, excess  3648
[t=   409] (   5) stream rate  59462, excess  4688
[t=   414] (   5) stream rate  61835, excess  5728
[t=   420] (   6) stream rate  64000, excess  6720
[t=   533] ( 113) stream rate  52833, excess  2576
[t=   539] (   6) stream rate  54619, excess  3568
[t=   544] (   5) stream rate  56470, excess  4608
[t=   549] (   5) stream rate  58287, excess  5648
[t=   555] (   6) stream rate  59963, excess  6640
[t=   687] ( 132) stream rate  50305, excess  1584
[t=   693] (   6) stream rate  51717, excess  2576
[t=   698] (   5) stream rate  53180, excess  3616
[t=   703] (   5) stream rate  54623, excess  4656
[t=   709] (   6) stream rate  55966, excess  5648
[t=   802] (  93) stream rate  51072, excess  2464
[t=   808] (   6) stream rate  52277, excess  3456
[t=   813] (   5) stream rate  53530, excess  4496
[t=   818] (   5) stream rate  54767, excess  5536
[t=   823] (   5) stream rate  55990, excess  6576
[t=   931] ( 108) stream rate  50870, excess  2672
[t=   936] (   5) stream rate  51965, excess  3712
[t=   942] (   6) stream rate  52993, excess  4704
[t=   947] (   5) stream rate  54065, excess  5744
[t=   952] (   5) stream rate  55126, excess  6784
[t=  1067] ( 115) stream rate  50384, excess  2544
[t=  1073] (   6) stream rate  51295, excess  3536
[t=  1078] (   5) stream rate  52244, excess  4576
[t=  1083] (   5) stream rate  53185, excess  5616
[t=  1089] (   6) stream rate  54067, excess  6608
[t=  1215] ( 126) stream rate  49514, excess  1840
[t=  1220] (   5) stream rate  50360, excess  2880
[t=  1225] (   5) stream rate  51200, excess  3920
[t=  1231] (   6) stream rate  51990, excess  4912
[t=  1236] (   5) stream rate  52815, excess  5952
[t=  1331] (  95) stream rate  50007, excess  2672
[t=  1336] (   5) stream rate  50778, excess  3712
[t=  1342] (   6) stream rate  51505, excess  4704
[t=  1347] (   5) stream rate  52264, excess  5744
[t=  1352] (   5) stream rate  53017, excess  6784
[t=  1480] ( 128) stream rate  49297, excess  1920
[t=  1485] (   5) stream rate  49993, excess  2960
[t=  1490] (   5) stream rate  50684, excess  4000
[... 449 lines cut ...]
[t= 13454] (   5) stream rate  48425, excess  5728
[t= 13459] (   5) stream rate  48502, excess  6768
[t= 13572] ( 113) stream rate  48193, excess  2624
[t= 13577] (   5) stream rate  48269, excess  3664
[t= 13582] (   5) stream rate  48346, excess  4704
[t= 13587] (   5) stream rate  48422, excess  5744
[t= 13593] (   6) stream rate  48495, excess  6736
[t= 13714] ( 121) stream rate  48161, excess  2208
[t= 13720] (   6) stream rate  48233, excess  3200
[t= 13725] (   5) stream rate  48308, excess  4240
[t= 13730] (   5) stream rate  48384, excess  5280
[t= 13736] (   6) stream rate  48456, excess  6272
[t= 13863] ( 127) stream rate  48105, excess  1456
[t= 13869] (   6) stream rate  48176, excess  2448
[t= 13874] (   5) stream rate  48251, excess  3488
[t= 13879] (   5) stream rate  48326, excess  4528
[t= 13884] (   5) stream rate  48401, excess  5568
[t= 13972] (  88) stream rate  48187, excess  2624
[t= 13977] (   5) stream rate  48262, excess  3664
[t= 13982] (   5) stream rate  48336, excess  4704
[t= 13987] (   5) stream rate  48410, excess  5744
[t= 13993] (   6) stream rate  48481, excess  6736
```

</details>

belm0, 2022-07-13:
similar story for an old Android peer @ 44.1 kHz

<details>
<summary>callback log</summary>

```
bluetooth sample rate: 44100
[t=      3] (   3) stream rate 554666, excess  1532
[t=      7] (   4) stream rate 365714, excess  2252
[t=     11] (   4) stream rate 314181, excess  2971
[t=     15] (   4) stream rate 290133, excess  3691
[t=    100] (  85) stream rate  52480, excess   838
[t=    104] (   4) stream rate  59076, excess  1558
[t=    108] (   4) stream rate  65185, excess  2278
[t=    112] (   4) stream rate  70857, excess  2997
[t=    116] (   4) stream rate  76137, excess  3717
[t=    201] (  85) stream rate  48398, excess   864
[t=    205] (   4) stream rate  51200, excess  1456
[t=    209] (   4) stream rate  54507, excess  2176
[t=    213] (   4) stream rate  57690, excess  2895
[t=    216] (   3) stream rate  61037, excess  3659
[t=    301] (  85) stream rate  46777, excess   806
[t=    305] (   4) stream rate  49101, excess  1526
[t=    309] (   4) stream rate  51365, excess  2246
[t=    313] (   4) stream rate  53571, excess  2965
[t=    317] (   4) stream rate  55722, excess  3685
[t=    402] (  85) stream rate  46169, excess   832
[t=    407] (   5) stream rate  47803, excess  1508
[t=    411] (   4) stream rate  49518, excess  2227
[t=    414] (   3) stream rate  51014, excess  2863
[t=    418] (   4) stream rate  52669, excess  3583
[t=    502] (  84) stream rate  45641, excess   774
[t=    506] (   4) stream rate  47051, excess  1494
[t=    511] (   5) stream rate  48344, excess  2169
[t=    514] (   3) stream rate  49805, excess  2933
[t=    518] (   4) stream rate  51150, excess  3653
[t=    604] (  86) stream rate  45350, excess   756
[t=    608] (   4) stream rate  46526, excess  1476
[t=    612] (   4) stream rate  47686, excess  2195
[t=    615] (   3) stream rate  48910, excess  2959
[t=    619] (   4) stream rate  50042, excess  3679
[t=    702] (  83) stream rate  45219, excess   786
[t=    706] (   4) stream rate  46232, excess  1506
[t=    710] (   4) stream rate  47233, excess  2225
[t=    714] (   4) stream rate  48224, excess  2945
[t=    718] (   4) stream rate  49203, excess  3665
[t=    805] (  87) stream rate  44998, excess   724
[t=    809] (   4) stream rate  45883, excess  1444
[t=    813] (   4) stream rate  46760, excess  2163
[t=    817] (   4) stream rate  47627, excess  2883
[t=    821] (   4) stream rate  48487, excess  3602
[t=    904] (  83) stream rate  45026, excess   838
[t=    907] (   3) stream rate  45724, excess  1474
[t=    911] (   4) stream rate  46507, excess  2193
[t=    915] (   4) stream rate  47283, excess  2913
[t=    919] (   4) stream rate  48052, excess  3633
[t=   1005] (  86) stream rate  44831, excess   736
[t=   1009] (   4) stream rate  45542, excess  1456
[t=   1013] (   4) stream rate  46246, excess  2175
[t=   1017] (   4) stream rate  46945, excess  2895
[t=   1021] (   4) stream rate  47639, excess  3614
[t=   1105] (  84) stream rate  44828, excess   806
[t=   1109] (   4) stream rate  45475, excess  1526
[t=   1113] (   4) stream rate  46001, excess  2117
[t=   1116] (   3) stream rate  46681, excess  2881
[t=   1120] (   4) stream rate  47314, excess  3600
[t=   1205] (  85) stream rate  44720, excess   748
[t=   1209] (   4) stream rate  45313, excess  1468
[t=   1213] (   4) stream rate  45902, excess  2187
[t=   1217] (   4) stream rate  46488, excess  2907
[... 109 lines cut ...]
[t=   3428] (   4) stream rate  44919, excess  2810
[t=   3431] (   3) stream rate  45141, excess  3573
[t=   3516] (  85) stream rate  44304, excess   721
[t=   3520] (   4) stream rate  44509, excess  1440
[t=   3524] (   4) stream rate  44712, excess  2160
[t=   3528] (   4) stream rate  44916, excess  2880
[t=   3532] (   4) stream rate  45118, excess  3599
[t=   3620] (  88) stream rate  44269, excess   614
[t=   3624] (   4) stream rate  44467, excess  1334
[t=   3628] (   4) stream rate  44665, excess  2054
[t=   3632] (   4) stream rate  44863, excess  2773
[t=   3635] (   3) stream rate  45072, excess  3537
[t=   3724] (  89) stream rate  44201, excess   380
[t=   3729] (   5) stream rate  44382, excess  1056
[t=   3733] (   4) stream rate  44575, excess  1775
[t=   3736] (   3) stream rate  44779, excess  2539
[t=   3740] (   4) stream rate  44971, excess  3258
[t=   3816] (  76) stream rate  44310, excess   803
[t=   3820] (   4) stream rate  44498, excess  1522
[t=   3824] (   4) stream rate  44686, excess  2242
[t=   3828] (   4) stream rate  44873, excess  2962
[t=   3832] (   4) stream rate  45060, excess  3681
```

</details>

pschatzmann, 2022-07-14:
I have never looked into this and honestly we don't have any influence on this so this is not the right place for this discussion.

Have you tried with a smaller i2s buffer size but bigger buffer count ?

belm0, 2022-07-14:
I don't know that I could file a coherent report to Espressif without using their API first hand.

> Have you tried with a smaller i2s buffer size but bigger buffer count ?

I have i2s disabled (or at least, the consumed audio does not go directly to the output)-- it's purely about the timing of the data stream callbacks.

pschatzmann, 2022-07-14:
Just use Espressif's A2DP example as a basis.

I don't think that this is a valid use case: You will need at least to throttle the processing to limit it to the valid sample rate.

belm0, 2022-07-14:
That sounds strange-- there is no back-pressure available in the A2DP API callback, am I correct?  My logs show that the target sample rate is maintained nonetheless.  (I'm only raising the bursty nature of it.)