pschatzmann/ESP32-A2DP #262: How to make a bluetooth micphone?

ziv2013, 2022-07-14:
Hi,

I'd like to make a bluetooth micphone which can be used with my laptop.

But I find all the examples only works with a bluetooth speaker.

I want the ESP32 can be found by laptop and pair. Is there any demos for this?

Thanks，

Ziv

pschatzmann, 2022-07-14:
Here is the Arduino Sketch example: https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-basic-api/base-i2s-a2dp

Please note that you need to google how to turn your laptop into an A2DP Sink yourself since this is dependent on the OS.

ziv2013, 2022-07-14:
Thanks a lot.

I have tested streams-i2s-webserver_wav, it works well.

And Is it possible for us to make a device that can act as my bluetooth headset?

I can search and pair my bluetooth headset directly with my laptop.

Ziv

pschatzmann, 2022-07-14:
That would be any of the Sink examples in this project. 
Just read the Readme!

You can try to use the internal DAC - though the quality will not be great and I recommend to use an external DAC. There is plenty of information in the Wiki

ziv2013, 2022-07-14:
Thanks, I would stuby WIKI.

closed 2022-07-14 by ziv2013