pschatzmann/ESP32-A2DP #265: cant compile(LINK) the BluetoothA2DPSinkQueued

mitchjs, 2022-07-14:
undefined reference to `vtable for BluetoothA2DPSinkQueued'

guessint some C++ thing
something to do with "virtual" missing?
me no C++ expert

thanks
mitch

pschatzmann, 2022-07-14:
The example bt_music_receiver_queued is compiling w/o error. I think I need some more information:
- what is your exact sketch
- what is your version displayed in the the ESP32 Arduino board manager
- how can I reproduce your problem ?

mitchjs, 2022-07-15:
i dont use arduino...
its the same simple code i been testing with since day 0

i got:

#include "BluetoothA2DPSinkQueued.h"

and this:
 a2dp_sink.start("MyMusic", true);  

thats it for now
its a linking issue im sure due to c__ inheritance, something with constructors/destructors

```
[5/7] Linking CXX executable esp32-a2dp.elf
FAILED: esp32-a2dp.elf 
cmd.exe /C "cd . && C:\Users\mitchjs\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\esp32-a2dp.elf.rsp -o esp32-a2dp.elf  && cd ."
c:/users/mitchjs/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(esp32-a2dp.cpp.obj):(.literal._Z41__static_initialization_and_destruction_0ii+0x4): undefined reference to `vtable for BluetoothA2DPSinkQueued'       
collect2.exe: error: ld returned 1 exit status
```

pschatzmann, 2022-07-15:
Can you double check if adding a destructor makes a difference:
```
class BluetoothA2DPSinkQueued : public BluetoothA2DPSink {
    public:
        BluetoothA2DPSinkQueued() = default;
       virtual ~BluetoothA2DPSinkQueued() {}

```

mitchjs, 2022-07-15:
Sadly no difference

mitchjs, 2022-07-15:
should this file be in the cmakelist :)

pschatzmann, 2022-07-15:
oops, yes

mitchjs, 2022-07-15:
waiting on it to build :)

 SRCS "src/BluetoothA2DPCommon.cpp" "src/BluetoothA2DPSink.cpp" "src/BluetoothA2DPSource.cpp" "src/BluetoothA2DPSourceQueue.cpp" "src/SoundData.cpp"

mitchjs, 2022-07-15:
all good, of course...

closed 2022-07-15 by mitchjs