pschatzmann/ESP32-A2DP #269: Is it possible to play custom music file while Bluetooth is connected?

fedorenchik, 2022-07-22:
Bluetooth speaker (which is ESP32 + ESP32-A2DP) need to play beeps on certain events, such as after Bluetooth is connected.

Following code plays .wav file and works if Bluetooth is not connected and doesn't work when Bluetooth is connected:
```c++
	a2dp_sink.start(name);

	FILE *wav_fp;
	wav_fp = fopen(("/spiffs/" + filename).c_str(), "rb");
	if (wav_fp == NULL)
	{
		printf("err: no file\n");
		return ESP_ERR_INVALID_ARG;
	}

	wav_header_t header;
	fread((void *)&header, sizeof(wav_header_t), 1, wav_fp);
	printf("Wav format:\n");
	printf("bit_depth: %d\n", header.bit_depth);
	printf("num_channels: %d\n", header.num_channels);
	printf("sample_rate: %d\n", header.sample_rate);

	size_t bytes_written;
	size_t cnt;
	uint8_t buff[256];
	esp_err_t ret;
	while (1) {
		cnt = fread(buff, 1, sizeof(buff), wav_fp);
		printf("%u\n", cnt);
		ret = i2s_write(I2S_NUM_1, (const void *)buff, sizeof(buff), &bytes_written, portMAX_DELAY);
		if (ret != ESP_OK) {
			ESP_LOGE(TAG, "err: %s\n", esp_err_to_name(ret));
			break;
		}
		if (cnt < sizeof(buff)) {
			break;
		}
	}
	fclose(wav_fp);
```

Is it possible to play custom music file (or maybe generate some beeps) while Bluetooth is connected?

fedorenchik, 2022-07-22:
The error I get is NULL pointer exception in `i2s_write()`:
```
I (6492) BT_AV: i2s_start
I (11592) BT_API: app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
I (11592) BT_AV: A2DP audio state: Suspended
W (11592) BT_AV: i2s_stop
I (14722) BT_AV: event: 13
Wav format:
bit_depth: 16
num_channels: 2
sample_rate: 44100
256
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4014df06  PS      : 0x00060130  A0      : 0x800d4661  A1      : 0x3ffe83f0  
A2      : 0x00000001  A3      : 0x3ffe8468  A4      : 0x00000100  A5      : 0x3ffe8464  
A6      : 0xffffffff  A7      : 0x00000000  A8      : 0x3ffc9e7c  A9      : 0x3ffe80c0  
A10     : 0x00000004  A11     : 0xffffffff  A12     : 0x00000001  A13     : 0x3ffe82d0  
A14     : 0x00000004  A15     : 0x3ffe82d0  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000018  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace:0x4014df03:0x3ffe83f00x400d465e:0x3ffe8420 0x400d5c5b:0x3ffe8590 0x400d379e:0x3ffe8890 0x400d8181:0x3ffe88d0 0x400d8197:0x3ffe88f0 0x4009482d:0x3ffe8910 

  #0  0x4014df03:0x3ffe83f0 in i2s_write at /home/leonid/.platformio/packages/framework-espidf/components/driver/i2s.c:2038 (discriminator 2)
  #1  0x400d465e:0x3ffe8420 in Speaker::PlayWavFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at src/hw/Speaker.cpp:33
  #2  0x400d5c5b:0x3ffe8590 in Spiffs::PlayActionFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at src/hw/Spiffs.cpp:50
  #3  0x400d379e:0x3ffe8890 in void std::__invoke_impl<void, void (*&)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&), char const*&>(std::__invoke_other, void (*&)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&), char const*&) at /home/leonid/.platformio/packages/toolchain-xtensa-esp
32/xtensa-esp32-elf/include/c++/8.4.0/bits/invoke.h:60
      (inlined by) std::__invoke_result<void (*&)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&), char const*&>::type std::__invoke<void (*&)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&), char const*&>(void (*&)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocat
or<char> > const&), char const*&) at /home/leonid/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/invoke.h:95
      (inlined by) void std::_Bind<void (*(char const*))(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)>::__call<void, , 0u>(std::tuple<>&&, std::_Index_tuple<0u>) at /home/leonid/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/functional:400
      (inlined by) void std::_Bind<void (*(char const*))(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)>::operator()<, void>() at /home/leonid/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/functional:484
      (inlined by) std::_Function_handler<void (), std::_Bind<void (*(char const*))(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)> >::_M_invoke(std::_Any_data const&) at /home/leonid/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
  #4  0x400d8181:0x3ffe88d0 in std::function<void ()>::operator()() const at /home/leonid/.platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
  #5  0x400d8197:0x3ffe88f0 in Action::TaskFunction(void*) at src/module/Action.cpp:180
  #6  0x4009482d:0x3ffe8910 in vPortTaskWrapper at /home/leonid/.platformio/packages/framework-espidf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 893def114f9d17fc

Rebooting...
```

If I add `i2s_start()` then the same error will be in `i2s_start()`.

closed 2022-07-27 by pschatzmann

fedorenchik, 2022-07-28:
At the end I had to modify ESP32-A2DP directly, as I don't use arduino and arduino-audio-tools is heavily dependent on arduino. It would be nice if ESP32-A2DP had this functionality out-of-the-box.

adnanaga, 2023-01-30:
What was your final solution here @fedorenchik ?

fedorenchik, 2023-01-31:
I've added `bt_i2s_task_start_up()` to `BluetoothA2DPSink::start()` and it seems to work.