pschatzmann/ESP32-A2DP #271: Treble and Bass-Control possible?

gitanova, 2022-07-25:
Hello from Austria!
I apPreciate your work A LOT !!! Thank you so much!!!

a2dp_sink.set_volume(1-255) ... works like a charm - but now I also would like to add a very simple treble and bass control to my bluetooth project. How is this possible?

Thank you for your kind help!

Michael

pschatzmann, 2022-07-25:
I suggest that you have a look at my https://github.com/pschatzmann/arduino-audio-tools library.

There you can try to use the https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_equilizer3_bands.html class.

gitanova, 2022-07-25:
I’m afraid – but somehow I do not know how to access the BT stream correctly by the equalizer classes.
Woould you mind to give me a short example?
If possible, I’d like to give a donation of 25 EUR to you for a working example.
Once again: I appreciate your work very much!

 

By the way: Your are from Switzerland – so do you speak German?

 

Thanks for helping me!

 

All the best to you

Michael.

pschatzmann, 2022-07-25:
Just use https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/basic-a2dp-i2s/basic-a2dp-i2s.ino as starting point.
Instead of writing to I2S you would write to Equilizer3Bands where I2S is defined as final output.

gitanova, 2022-07-25:
Thank you for your fast reply – even though I‘ m stuck – so  I’d like to ask you for a working example…
Could you please provide my with an example – that would help really a lot and I would be very gratful for helping me!
best 

Mchael

gitanova, 2022-07-26:
Dear Phil, 
somehow I'm getting stuck.
I believe that the integration of the 3-band equalizer does not work at all.
A quick example from you would really help me a lot. 
Like I said before, I would pay for that too. I have to get the Bluetooth module up and running this week.
Please help me!

Thank you and best regards 
Michael

pschatzmann, 2022-07-26:
I just discovered that the example already exists:
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/basic-a2dp-eq-audiokit

Just replace the AudioKitStream with your desired output sink class (e.g. I2SStream).

However note that I have never tested this functionality properly...

gitanova, 2022-07-26:
Thank you very much!

 

Following code compiles without error. Hardware wörks with your SompleBT_Example (without EQ) very well but I can hear NOTHING with using this code:

 

#include "AudioTools.h"

#include "AudioLibs/AudioA2DP.h"

 

 

BluetoothA2DPSink a2dp_sink;

I2SStream i2s;

Equilizer3Bands eq(i2s);

ConfigEquilizer3Bands cfg_eq;

 

 

void read_data_stream(const uint8_t *data, uint32_t length) {

    eq.write(data, length);

}

 

void setup() {

  Serial.begin(115200);

  AudioLogger::instance().begin(Serial, AudioLogger::Info);

 

  // setup output

  auto cfg = i2s.defaultConfig(TX_MODE);

  i2s.begin(cfg);

 

  // setup equilizer

  cfg_eq = eq.defaultConfig();

  cfg_eq.setAudioInfo(cfg); // use channels, bits_per_sample and sample_rate from i2s

  cfg_eq.gain_low = 1; 

  cfg_eq.gain_medium = 0.5;

  cfg_eq.gain_high = 1.0;

  eq.begin(cfg_eq);

 

  // register callback

  a2dp_sink.set_stream_reader(read_data_stream, true);

  a2dp_sink.start("BT");

}

 

void loop() {

 

}

 

https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/basic-a2dp-eq-audiokit

gitanova, 2022-07-26:
Al last I VE GOT IT !!!
My error was that I did not notice that in your default pins for I2S differed completely from mine … therefor I could not get any sound out of PCM5102…

By the wa: Do you know another alternative for PCM510x chips? They seem to be very rare on the market in our days…

Anyway: Best thanks!
Plese give me your IBAN for a small donation for your work!

 

Best regards Michael

 

https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/basic-a2dp-eq-audiokit

closed 2022-07-27 by pschatzmann

harryberlin, 2024-06-11:
Tried to implement EQ3Bands to bt_music_receiver_simple, but get lot of crackle sound.
Some Links in this discussion are dead.

```
#include "AudioTools.h"
#include "BluetoothA2DPSink.h"

I2SStream out;
BluetoothA2DPSink a2dp_sink(out);

Equilizer3Bands eq(out);

ConfigEquilizer3Bands cfg_eq;

void read_data_stream(const uint8_t *data, uint32_t length) {
  eq.write(data, length);
}

void setup() {
  auto cfg = out.defaultConfig();
  cfg.pin_bck = 32;
  cfg.pin_ws = 27;
  cfg.pin_data = 25;
  out.begin(cfg);

  // setup equilizer
  cfg_eq = eq.defaultConfig();
  cfg_eq.setAudioInfo(cfg); // use channels, bits_per_sample and sample_rate from i2s
  cfg_eq.gain_low = 1.0;
  cfg_eq.gain_medium = 0.5;
  cfg_eq.gain_high = 1.0;
  eq.begin(cfg_eq);

  // register callback
  a2dp_sink.set_stream_reader(read_data_stream, true);

  a2dp_sink.start("MyMusic");
}

void loop() {
  delay(1000); // do nothing
}
```

pschatzmann, 2024-06-11:
I don't think your output chain makes a lot of sense. It should be
```
I2SStream out;
Equilizer3Bands eq(out);
BluetoothA2DPSink a2dp_sink(eq);
```

harryberlin, 2024-06-12:
Thanks for feedback.
The error was the existing `a2dp_sink.set_stream_reader(read_data_stream, true);` and the output chain.

Now, that's the result:
```
#include "AudioTools.h"
#include "BluetoothA2DPSink.h"

I2SStream out;
Equilizer3Bands eq(out);
BluetoothA2DPSink a2dp_sink(eq);
ConfigEquilizer3Bands cfg_eq;

void print_eq_state() {
  Serial.printf("LOW:%.1f\r\n", cfg_eq.gain_low);
  Serial.printf("MID:%.1f\r\n", cfg_eq.gain_medium);
  Serial.printf("HIGH:%.1f\r\n", cfg_eq.gain_high);
}

void setup() {
  Serial.begin(115200);

  auto cfg = out.defaultConfig();
  cfg.pin_bck = 32;
  cfg.pin_ws = 27;
  cfg.pin_data = 25;
  out.begin(cfg);

  // setup equilizer
  cfg_eq = eq.defaultConfig();
  cfg_eq.setAudioInfo(cfg); // use channels, bits_per_sample and sample_rate from i2s
  cfg_eq.gain_low = 1.0;
  cfg_eq.gain_medium = 1.0;
  cfg_eq.gain_high = 1.0;
  eq.begin(cfg_eq);

  a2dp_sink.start("MyMusic");
}

void loop() {
  if (Serial.available()) {
    switch (Serial.read()) {
      case 'U':
        cfg_eq.gain_low += 0.1;
        cfg_eq.gain_medium += 0.1;
        cfg_eq.gain_high += 0.1;
        print_eq_state();
        break;
      case 'D':
        cfg_eq.gain_low += -0.1;
        cfg_eq.gain_medium += -0.1;
        cfg_eq.gain_high += -0.1;
        print_eq_state();
        break;
      case 'B':
        if (Serial.read() == '+') cfg_eq.gain_low += 0.1;
        else cfg_eq.gain_low += -0.1;        
        print_eq_state();
        break;
      case 'T':
        if (Serial.read() == '+') cfg_eq.gain_high += 0.1;
        else cfg_eq.gain_high += -0.1;
        print_eq_state();
        break;
      case 'M':
        if (Serial.read() == '+') cfg_eq.gain_medium += 0.1;
        else cfg_eq.gain_medium += -0.1;       
        print_eq_state();
        break;
      case 'N':
        cfg_eq.gain_low = 1.0;
        cfg_eq.gain_medium = 1.0;
        cfg_eq.gain_high = 1.0;
        print_eq_state();
        break;
      case 'Q':
        print_eq_state();
        break;

    }
  }
}

void print_eq_state() {
  Serial.printf("LOW:%.1f\r\n", cfg_eq.gain_low);
  Serial.printf("MID:%.1f\r\n", cfg_eq.gain_medium);
  Serial.printf("HIGH:%.1f\r\n", cfg_eq.gain_high);
}

```

aqw42, 2024-10-24:
Does dynamically changing the EQ config without using its begin() methods really changes the output ?
If so that's really interesting

pschatzmann, 2024-10-24:
Yes, the ConfigEquilizer3Bands is passed as reference, so changes will automatically reflected