pschatzmann/ESP32-A2DP #273: Streaming of raw files from SD card using A2DP not working

RobotSquad, 2022-08-03:
I was using this:
https://www.pschatzmann.ch/home/2021/04/29/bluetooth-a2dp-streaming-from-files-on-a-sd-card/
i had to edit the code a little to get it to compile but then.. when i uploaded and ran the code and then connected my headphones.. no sound is being played. please help
This is the code i have been using.. what am i doing wrong?
`#include "BluetoothA2DPSource.h"
#include <SPI.h>
#include <SD.h>
#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"
BluetoothA2DPSource a2dp_source;
File sound_file;
const char* file_name = "/audio.raw";
const int sd_ss_pin = 13;
const int frame_size_bytes = sizeof(int16_t) * 2;
// callback used by A2DP to provide the sound data
int32_t get_sound_data(Frame *frame, int32_t len) {
  // the data in the file must be in int16 with 2 channels 
  size_t result_len_bytes = sound_file.read((uint8_t*)frame, len * frame_size_bytes );
  // result is in number of frames
  int32_t result_len = result_len_bytes / frame_size_bytes;
  ESP_LOGD("get_sound_data", "%d -> %d",len );
  return result_len;
}

// Arduino Setup
void setup(void) {
    WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);
  Serial.begin(115200);
   SPI.begin(14, 2, 15, 13);
  // Setup SD and open file
  if(SD.begin(13)){
    Serial.println("SD OPEN");
     sound_file = SD.open(file_name, FILE_READ);

  // start the bluetooth
  Serial.println("starting A2DP...");
  a2dp_source.start("Infinity Glide 510", get_sound_data);  
  }
}

// Arduino loop - repeated processing 
void loop() {
`}

RobotSquad, 2022-08-04:
now these errors are also showing up 
[image]

pschatzmann, 2022-08-04:
You will need to analyze the stacktrace: https://github.com/me-no-dev/EspExceptionDecoder

RobotSquad, 2022-08-13:
well  turns out the problem was the brownout detector statement. and the power.. after properly powering the esp and removing the brownout detector statement the code started working..

closed 2022-08-13 by RobotSquad