pschatzmann/ESP32-A2DP #279: I2S input from second ESP32 (Karadio32) to send audio to a BT Speaker

JocB1, 2022-08-08:
Hi, 

I have built several Karadio32 Internet radios which use the ESP32. Partly also with I2S DAC or I2S Amp (MAX98xx). 

Now I need a way to stream the I2S output signal of the ESP32 (Karadio32) via Bluetooth to my BT speaker at the Tereasse. I think this project would fit if I donate a second ESP32 that only takes care of the BT audio transmission. 

Is this possible and can someone give me some guidance on how to configure this?

Best regards, 

Jochen

JocB1, 2022-08-08:
I mean a example for BT source with I2S input signals, not BT sink with I2S output.

pschatzmann, 2022-08-08:
Here is an example which is part of the arduino-audio-tools: https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-basic-api/base-i2s-a2dp

However I think the challenge might be to get a stable I2S between 2 ESP32. I have never tried this myself, but if you look in the project there are some discussions and issues of failed attempts. My gut feeling is that it might be easier to use a simple Serial connection and maybe transmit the encoded data.