pschatzmann/ESP32-A2DP #285: Data format

JoaoPedroAssis, 2022-08-20:
Hi! This can be a silly question, but how the data is formatted in the `read_data_stream` callback function? I read the ESP I2S docs and didnt understand much of it.

When i get the `uint8_t * data` parameter, in stereo mode, how can I separate both channels or determine how many actual samples there is (data.length / 2 ?) and stuff like that. If you could point me to some articles or materials to better understand this, it would be of great help!

Thanks in advance

pschatzmann, 2022-08-20:
Just look at the Readme: There you see 
```
void read_data_stream(const uint8_t *data, uint32_t length)
{
  int16_t *samples = (int16_t*) data;
  uint32_t sample_count = length/2;
  // Do something with the data packet
}
```
samples[0] is the first same for the left channel
samples[1] is the first same for the right channel
...

JoaoPedroAssis, 2022-08-20:
I said it would probably be silly 😆 . Jokes aside, thanks for the quickness, I've read this thing god knows how many times and missed this sort of thing. I've been asking a lot of silly questions in this repo, so thanks for the patience as well!!

closed 2022-08-21 by JoaoPedroAssis