pschatzmann/ESP32-A2DP #287: A2DP Sink multiple connection.

int-0x03, 2022-08-26:
Hello there! Author, good library thanks. I have question. How I can create multiple a2dp connections? (as far as I know, BT in esp32 support 7 connections (not BLE) at the same time.) I want receive more than one a2dp connections (player, phone, etc) and manage them, plus, a want mix channels, sometimes. Is it possible with this library?

pschatzmann, 2022-08-26:
Where did you get this information from ?
As far as I know you can have just one A2DP Sink or Source, or BLE or Bluetooth Serial at a time.

Insomniac80, 2022-08-28:
https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-faq/en/latest/software-framework/ble-bt.html#how-many-bluetooth-clients-can-be-connected-to-esp32

"
How many Bluetooth® clients can be connected to ESP32?

The Bluetooth LE server supports up to nine client connections, please check the configuration of parameter ble_max_conn for applications. For stable connection, three clients should be good.
"

pschatzmann, 2022-08-28:
That's only for BLE!

Insomniac80, 2022-08-28:
I´m sorry, haven´t seen BLE Audio is only supported in BT 5.2+, while ESPs only provide 4.x.

Then maybe the issue creater has a different source then https://github.com/espressif/esp-idf/issues/2060 which idicates it is not possible.

Great lib btw :)

closed 2022-08-30 by pschatzmann

int-0x03, 2022-08-30:
@Insonmniac80 does it mean headset must support BLE 5.2 also? I read on reddit, esp32 can connect 7 classic bt connections and 8 BLE connections at the same time. Link: https://www.reddit.com/r/esp32/comments/c0mv8s/how_many_bluetooth_devices_can_connect/

int-0x03, 2022-08-30:
I will try to make it on low level esp32 API from espressif. I will text you if it is possible.

rrwakc, 2024-01-08:
Any updates regarding that.
Also would it be possible to create pass-through between 2 headsets.
Eg. have a communication hub for 2 people on a motorcycle + mp3 player