pschatzmann/ESP32-A2DP #288: A2DP sender example: Sender does not reconnect to paired A2DP sink after power cycle

saeugetier, 2022-08-30:
I have tried the example with latest PlatformIO ESP32 Arduino Framework:
https://github.com/pschatzmann/ESP32-A2DP/tree/main/examples/bt_music_sender_write

My expectation was, that I can once pair a Bluetooth headphone with the ESP32 A2DP sender and it will reconnect after every power cycle without new pairing. 

The behavior I noticed from logs: 
1. Switch on for the first time: The ESP32 is searching for the device with the hard coded device name
2. I press the pairing button on my headphone and the pairing was successful. ESP32 and headphone has connected.
3. Piano plays
4. Power cycle of ESP32
5. The log says, that the Bluetooth device is connected after a short time it is disconnected. And it repeats in an endless loop. No piano plays.

pschatzmann, 2022-08-30:
I think this is an issue with your headphone. I think the following information should help 
https://github.com/pschatzmann/ESP32-A2DP/wiki/Auto-Reconnect

saeugetier, 2022-08-30:
I will experiment with the API: https://pschatzmann.github.io/ESP32-A2DP/html/class_bluetooth_a2_d_p_source.html

I learned about this library through this video. The guy faced the same problem, I have. After power cycle the connection to the bluetooth box cannot be established again. The part with that issue starts about 11:30.
https://youtu.be/a936wNgtcRA?t=690

pschatzmann, 2022-08-30:
My current understanding is that this is most likely related to the fact that the receiving device did not notice that the sender got disconnected and does not allow further connections.

Can you try to disconnect properly before starting a new connection ? 
Can you reproduce this issue in this case as well?

saeugetier, 2022-08-31:
You are right. When pressing the reset button for a short time, the headset doesn't notice that the A2DP source has disconnected. After the headset noticed a disconnection, I hear the voice "device disconnected". So the ESP32 tries to reconnect too fast.

closed 2022-08-31 by saeugetier