pschatzmann/ESP32-A2DP #294: Volume control and bt_music_receiver_32bit_ext example

zongone, 2022-09-06:
Hey there
is it possible to enable volume control while using bt_music_receiver_32bit_ext subclass example?

pschatzmann, 2022-09-07:
I don't see any reason why this should not work. 
ps. why don't you just use the volume control on the phone ?

zongone, 2022-09-07:
Volume control on the phone doesn't work that's the point.
I think there is a problem in this line of code:
`if (is_volume_used) {`
is_volume_used seems to be always false.

pschatzmann, 2022-09-07:
This value is false if the A2DP never got any volume information from your phone. 
There are phones which do not send volume information but change the signal directly...

zongone, 2022-09-09:
The thing is that the volume control works flawlessly if I use the original class 'BluetoothA2DPSink.h' - both on my phone and laptop.

If instead of 'BluetoothA2DPSink.h' I use 'BluetoothA2DPSink32.h' the volume control stops working properly.

Maybe it's because the 'is_volume_used' from the if condition in 'BluetoothA2DPSink32.h' comes from the 'BluetoothA2DPCommon' namespace instead of the 'A2DPVolumeControl' namespace like in the original version of the 'BluetoothA2DPSink.h' class?

pschatzmann, 2022-09-09:
The [bt_music_receiver_32bit](https://github.com/pschatzmann/ESP32-A2DP/tree/main/examples/bt_music_receiver_32bit) should work properly.

The volume handling has changed and it seems that the BluetoothA2DPSink32 has not been adapted because it is not part of the maintained codebase - it's just an example.

I tried to make a correction. Can you double check if it is working properly now...

zongone, 2022-09-10:
Seems that after your modification volume controls started to work but the output is very low and I almost don't hear any sound...

pschatzmann, 2022-09-10:
I was not sure about the righ factor. Maybe you can figure it out!

zongone, 2022-09-11:
volumeFactor =0x1000 seems good. Thanks a lot!

closed 2022-09-17 by pschatzmann