pschatzmann/ESP32-A2DP #295: Turn off A2DP

podaen, 2022-09-13:
How can I turn off the A2DP to connect to the WiFi?

```
void A2DPOn(bool BlnOn) {
	if (BlnOn == true) {
		IniA2DP();
	}
	else
	{
		a2dp_sink.end();
		Serial.println("Stopping A2DP");
	}
}
```

pschatzmann, 2022-09-13:
a2dp_sink.end(); should work. However I would recommend to reboot if you want to have the full memory available again!

podaen, 2022-09-13:
The memory is not the issue. It doesn't seem that the a2dp_sink.end(); is enough to connect to my broker.

pschatzmann, 2022-09-13:
Did you just try to start Wifi first with your own code?

podaen, 2022-09-13:
That's what I do with WiFi ini. I didn't connect to the WiFi first.

pschatzmann, 2022-09-13:
Try to activate the ESP32 logging. Maybe you get some more info there...

podaen, 2022-09-13:
Seems that after enabling the A2DP and disabling, I can not load the certificate correctly.

> [  7730][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():276]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed

pschatzmann, 2022-09-13:
I would guess that there is not enough memory....

podaen, 2022-09-14:
> You can try to release additional memory with a2dp_sink.end(true); with the effect that a2dp can not be started any more

I just did and it is connected now. sadly I can not start the a2dp anymore without restarting it. So my idea is nothing worth anymore. I use PSRAM and I know the PSRAM is unacessible for DMA so encryption functions (like those, used in SSL) are not working. I think I am stuck on this...

podaen, 2022-09-14:
Maybe disable PSRAM on code for a2dp... Do you know how to do this?

pschatzmann, 2022-09-14:
No, I never tried. Anyhow I don't think that this is an error in my library: A2DP has nothing to do with Mqtt.
Did you try to search for issues in the ESP32 Arduino or I suggest to open an issue with the MQTT library provider

podaen, 2022-09-15:
I did not put the question on the Arduino side. I know they are working an new designs that are more compatible with encryption. What I did is test my esp32 further to see how it is reacting on certain settings.

> Arduino IDE 1.8.19
> Pins 21,22
> SDK: ESP Arduino 2.0.4
> Board ESP32 DevKit Wrover IE
> I2C device PCF8574
> 
> ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 3, 4MB external flash
> Chip id: D4BD9710
> Model number: ESP32-D0WDQ5

PSRAM OFF -> boot on A2DP -> a2dp_sink.end();  -> WiFi : WiFi can not connect
PSRAM ON -> boot on A2DP -> a2dp_sink.end();  -> WiFi on -> MQTT can not connect

Conclusion: With PSRAM I was able to connect to the WiFi without clearing all the memory of A2DP (a2dp_sink.end();) , but I couldn't connect to the secure MQTT. I believe it is possible if the MQTT is not HTTPS based website. So for HTTP it could equally work with PSRAM.

Because I have in my case a secure server it is easier for me to restart the esp. If there is no device connected to the A2DP I wait for 1 minute and turnoff the a2dp_sink.end(true); initialize the WiFi connect the MQTT and see if there are any messages and the restart.

pschatzmann, 2022-09-15:
I agree: restarting is the cleanest solution and you don't have to worry that certain memory does not get released...

closed 2022-09-17 by pschatzmann