pschatzmann/ESP32-A2DP #296: BluetoothA2DPSource: unhandled evt 4

tueddy, 2022-09-14:
Hi PSchatzmann,

first many thank's for your great library!
Using A2DPSource, PlatformIO with latest Arduino 2.0.4 i get this log/error message:

`[  2037][E][BluetoothA2DPSource.cpp:723] bt_app_av_state_connecting(): [BT_AV] bt_app_av_state_connecting unhandled evt 4
`

Seems that the event ESP_A2D_PROF_STATE_EVT is missing, maybe only available in latest ESP-SDK?
Don't know if it's harmless or do i miss something?

Best regards
tueddy

pschatzmann, 2022-09-16:
I don't think that this is important. After all in the official [example from Espressif](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_source) it is not handled either.

tueddy, 2022-09-16:
OK, thank's!

closed 2022-09-16 by tueddy