pschatzmann/ESP32-A2DP #297: write_data not playing anything

monstermichl, 2022-09-15:
Hi pschatzmann,

firstly, thanks for providing this awesome library to us. Secondly, I'm trying to use the write_data function of the BluetoothA2DPSource class. Even though no data callback was provided on the instantiation, nothing is played back. I thought this might be a problem with the core the code is running on, so I tried with both cores but nothing changed. The task which uses write_data is pinnned to core 1. I checked the data in the frames array which is also filled. Here's the code for the task where the data shall be written to the source. Is there anything special that must be considered?

```cpp
static void sendAudioTask(void*)
{
    while (1)
    {
        if (frameBufferFlush)
        {
            Frame frame;

            for (int32_t i = 0; i < FRAME_BUFFER_LENGTH; ++i)
            {
                xQueueReceive(queueHandle, &frame, 20);

                frameBuffer[i].channel1 = frame.channel1;
                frameBuffer[i].channel2 = frame.channel2;
            }
            soundDataPtr->setData(frameBuffer, FRAME_BUFFER_LENGTH);
            bluetoothSource.write_data(soundDataPtr);
            frameBufferFlush = false;
        }
        vTaskDelay(10 / portTICK_PERIOD_MS);
    }
}
```

These are the global variables used.

```cpp
Frame frameBuffer[FRAME_BUFFER_LENGTH];
bool frameBufferFlush;
TwoChannelSoundData *soundDataPtr; /* Instantiated in setup function. */
```

Thanks

pschatzmann, 2022-09-15:
Forget about write data if you want to constantly feed audio data. This is just a "nice method" to get started. You need to provide the data via callback! as demonstrated in [this example](https://github.com/pschatzmann/ESP32-A2DP/blob/main/examples/bt_music_sender/bt_music_sender.ino)

You also find plenty of examples in the [Arduino-Audio-Tools](https://github.com/pschatzmann/arduino-audio-tools) project

monstermichl, 2022-09-16:
Well, that's not the answer I was looking for :( However, thanks for the fast response. At least, now I know I need to find a workaround

closed 2022-09-16 by monstermichl

pschatzmann, 2022-09-16:
If you want to stick with the write logic you can use the https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_a2_d_p_stream.html class from the audio-tools

monstermichl, 2022-09-16:
Thanks for pointing me into this direction. Tried to set something up which, as I understood, should work. However, the program crashes on decoder.begin(). Am I missing something here?

```cpp
URLStream inStream(SSID, PASSWORD);
A2DPStream outStream = A2DPStream::instance();
MP3DecoderHelix mp3Decoder;
EncodedAudioStream decoder(&outStream, &mp3Decoder);
StreamCopy copier(decoder, inStream);

void setup()
{
    A2DPConfig config;

    config.mode = TX_MODE;
    config.name = EARPHONE_NAME;
    config.auto_reconnect = true;

    outStream.begin(config);
    decoder.begin(); /* This fails for some reason. */
    inStream.begin(STREAM, "audio/mp3");
}

void loop()
{
    copier.copy();
}
```

pschatzmann, 2022-09-16:
https://github.com/pschatzmann/ESP32-A2DP/issues/19