pschatzmann/ESP32-A2DP #301: Possible to make A2DP to support aptX, aptX-HD, aptX-LL, LDAC codecs

Sergionn, 2022-09-27:
Is that possible to make A2DP to support aptX, aptX-HD, aptX-LL, LDAC codecs,
like done here https://github.com/cfint/esp32-a2dp-sink
It woks fine, but not for arduino (only idf), and has some modifications in Bluedroid strack based on April release of IDF.

pschatzmann, 2022-09-27:
No, you would really need push to have this integrated into the regular IDF that is part of Arduino!

closed 2022-09-27 by pschatzmann

generic-beat-detector, 2022-11-26:
Hello,
Forgive my ignorance. I'm still trying to wrap my head around the underlying concepts/mechanisms of the  ArduinoIDE, Platformio, and ESP-IDF build environments. 

I would like to test the performance of this low-latency fork (@ https://github.com/cfint/esp32-a2dp-sink). For this, I suspect that I'd have to use a platformio build environment in order to pull and build against the corresponding ESP-IDF fork (@ https://github.com/cfint/esp-idf)? If so, how would I go about configuring platformio.ini?

Regards,
Gbd