pschatzmann/ESP32-A2DP #310: Compile error ESP32-WROOM-DA Module.

ENOTSERVICE, 2022-10-21:
I/Users/enotik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.5/tools/sdk/esp32/include/esp-sr/esp-tts/esp_tts_chinese/include -I/Users/enotik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.5/tools/sdk/esp32/include/esp-sr/include/esp32 -I/Users/enotik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.5/tools/sdk/esp32/include/esp32-camera/driver/include -I/Users/enotik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.5/tools/sdk/esp32/include/esp32-camera/conversions/include -I/Users/enotik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.5/tools/sdk/esp32/include/fb_gfx/include -I/Users/enotik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.5/tools/sdk/esp32/qio_qspi/include -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu++11 -fexceptions -fno-rtti -MMD -c -Wall -Werror=all -Wextra -DF_CPU=240000000L -DARDUINO=10819 -DARDUINO_ESP32_WROOM_DA -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_WROOM_DA\"" "-DARDUINO_VARIANT=\"esp32da\"" -DARDUINO_PARTITION_default -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @/var/folders/vg/d_v99drx0vz92bdypj7fcd480000gn/T/arduino_build_417225/build_opt.h -I/Users/enotik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.5/cores/esp32 -I/Users/enotik/Library/Arduino15/packages/esp32/hardware/esp32/2.0.5/variants/esp32da -I/Users/enotik/Documents/Arduino/libraries/ESP32-A2DP/src /Users/enotik/Documents/Arduino/libraries/ESP32-A2DP/src/SoundData.cpp -o /var/folders/vg/d_v99drx0vz92bdypj7fcd480000gn/T/arduino_build_417225/libraries/ESP32-A2DP/SoundData.cpp.o
/Users/enotik/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::reset_last_connection()':
/Users/enotik/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPSource.cpp:236:17: error: unused variable 'bda_str' [-Werror=unused-variable]
     const char *bda_str = to_str(last_connection);
                 ^~~~~~~
/Users/enotik/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::filter_inquiry_scan_result(esp_bt_gap_cb_param_t*)':
/Users/enotik/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPSource.cpp:381:13: error: variable 'rssi' set but not used [-Werror=unused-but-set-variable]
     int32_t rssi = -129; /* invalid value */
             ^~~~
/Users/enotik/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::bt_app_av_state_unconnected(uint16_t, void*)':
/Users/enotik/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPSource.cpp:677:26: error: unused variable 'p' [-Werror=unused-variable]
                 uint8_t *p = peer_bd_addr;
                          ^
/Users/enotik/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::bt_av_hdl_avrc_ct_evt(uint16_t, void*)':
/Users/enotik/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPSource.cpp:898:22: error: unused variable 'bda' [-Werror=unused-variable]
             uint8_t *bda = rc->conn_stat.remote_bda;
                      ^~~
cc1plus: some warnings being treated as errors
Используем библиотеку ESP32-A2DP версии 1.7.1 из папки: /Users/enotik/Documents/Arduino/libraries/ESP32-A2DP 
exit status 1
Ошибка компиляции для платы ESP32-WROOM-DA Module.

pschatzmann, 2022-10-21:
Just set the compiler warnings back to normal!

closed 2022-10-21 by pschatzmann

ENOTSERVICE, 2022-10-21:
> Just set the compiler warnings back to normal

The error still remained