pschatzmann/ESP32-A2DP #314: enhancement, DSP capabilities

B4stl3r, 2022-11-02:
Hi Phil,

1st of all, thx for providing this great library !!!

As you already implemented some Volume enhancements (currently still trying to get it running), I would like to ask, what you think about providing DSP capabilities?

As the cheap amplifiers do not have such possibilities, do you think the esp has enough power left to do some processing?
Had also just a quick look, what I can find in regards of DSP:
https://youtu.be/_nCOluMHfGI?t=257
https://github.com/YetAnotherElectronicsChannel/ESP32_Bluetooth_DSP_Speaker/tree/master/code/main

Will give it a try in the next days, but maybe you can let me know your thinking on that.

pschatzmann, 2022-11-02:
What exactly do you think is missing ?

B4stl3r, 2022-11-02:
filtering / enhancing of audio? like an equializer?

Or is that already possible? did I miss out something? :O

pschatzmann, 2022-11-02:
Yes, it is all there. I suggest that you consult the Wiki https://github.com/pschatzmann/arduino-audio-tools/wiki
and class documentation

B4stl3r, 2022-11-02:
allright, thank you. didn't have a look on your other repo's..

closed 2022-11-02 by B4stl3r