pschatzmann/ESP32-A2DP #316: BluetoothA2DPSource not working 🥺🥺

mkhan9047, 2022-11-11:
[I][BluetoothA2DPSource.cpp:387] filter_inquiry_scan_result(): --Class of Device: 0x5a020c
[I][BluetoothA2DPSource.cpp:391] filter_inquiry_scan_result(): --RSSI: -32
[I][BluetoothA2DPSource.cpp:442] filter_inquiry_scan_result(): --Compatiblity: Scanned Device not Compatible.
[I][BluetoothA2DPSource.cpp:381] filter_inquiry_scan_result(): Scanned device: 38:38:4b:9f:5a:e7
[I][BluetoothA2DPSource.cpp:387] filter_inquiry_scan_result(): --Class of Device: 0x5a020c
[I][BluetoothA2DPSource.cpp:391] filter_inquiry_scan_result(): --RSSI: -30

I am a noob in embedded programming, I need urgent help here @pschatzmann .  I just need to send audio data from I2S microphone to my phone. But it's not connecting, showing the above logs. Hint: I am not using Sink. 

Advance thanks! And Awesome works.

pschatzmann, 2022-11-11:
You can only send audio data to a device which implements a A2DP Sink: These are usually Bluetooth Speakers.
Mobile phones do not provide this functionality.
I suggest that you try out my Audio Tools library and maybe have a look at the folloiwng sketch: https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-webserver/streams-i2s-webserver_wav

closed 2022-11-11 by pschatzmann

mkhan9047, 2022-11-11:
@pschatzmann Thanks for your quick answer. My goal is to create a bluetooth headset type device by ESP32, which can stream music from phone to a speaker and also a microphone will be sending data to phone via ESP32. Hope it make more sense to you now. Any suggestions will be greatly helpful 🥰🥰

mkhan9047, 2022-11-11:
@pschatzmann Maybe I just confused you, sorry 😢 . I am just trying to build a Bluetooth headset, think about the Airpod by ESP32, which has a speaker and also a microphone. And when connected to a phone the phone uses the microphone from ESP32 and also the connected speaker. 

Your suggestion will drive me to do the right path for this MVP. Advanced thanks ❤️❤️❤️❤️

pschatzmann, 2022-11-11:
As already explained, your scenario is not supported by A2DP!