pschatzmann/ESP32-A2DP #339: Error compiling libraries

podaen, 2022-12-19:
For 'ESP32 Dev Module(espressif_esp32) I am not able to compile anymore after updating this lib... Could you please look. All my warnings are out.

> BluetoothA2DPSink.cpp: In member function virtual void BluetoothA2DPSink::av_hdl_avrc_evt(uint16_t, void*)
> BluetoothA2DPCommon.cpp: In member function virtual void BluetoothA2DPCommon::set_scan_mode_connectable(bool)
>  
>  
> BluetoothA2DPCommon.cpp: 293:38: error: 'ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE' was not declared in this scope
>    if (esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE)!=ESP_OK){
>    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> BluetoothA2DPSink.cpp: 884:86: error: no matching function for call to 'BluetoothA2DPSink::av_notify_evt_handler(uint8_t&, esp_avrc_rn_param_t&)
> BluetoothA2DPCommon.cpp:293: note  suggested alternative  ESP_BT_GENERAL_DISCOVERABLE
>    if (esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE)!=ESP_OK){
>    av_notify_evt_handler(rc->change_ntf.event_id, rc->change_ntf.event_parameter)
>    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> BluetoothA2DPSink.cpp:828: note  candidate  virtual void BluetoothA2DPSink  av_notify_evt_handler(uint8_t, uint32_t)
>    ESP_BT_GENERAL_DISCOVERABLE
>  void BluetoothA2DPSink*: av_notify_evt_handler(uint8_t event_id, uint32_t event_parameter)
>    ^~~~~~~~~~~~~~~~~
>  
> BluetoothA2DPSink.cpp:828: note    no known conversion for argument 2 from esp_avrc_rn_param_t to uint32_t {aka unsigned int}
> BluetoothA2DPCommon.cpp: 297:38: error: 'ESP_BT_SCAN_MODE_NONE' was not declared in this scope
>    if (esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_NONE)!=ESP_OK){
>    ^~~~~~~~~~~~~~~~~~~~~
> BluetoothA2DPCommon.cpp:297: note  suggested alternative  ESP_BT_COD_SRVC_NONE
>    if (esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_NONE)!=ESP_OK){
>    ^~~~~~~~~~~~~~~~~~~~~
>    ESP_BT_COD_SRVC_NONE

Before I had it working on ESP32 cores 2.0.4. Now I am on the latest 2.0.5 dev.

podaen, 2022-12-19:
Sorry, it was because I imported evryting in my shetch. Now the error is gone.

```
#include <A2DPVolumeControl.h>
#include <BluetoothA2DP.h>
//#include <BluetoothA2DPCommon.h>
//#include <BluetoothA2DPSink.h>
//#include <BluetoothA2DPSinkQueued.h>
//#include <BluetoothA2DPSource.h>
//#include <SoundData.h>
```

pschatzmann, 2022-12-19:
Actually, your right: multiple import should not give any errors.
I committed the related correction...

closed 2022-12-19 by pschatzmann

podaen, 2022-12-19:
Phill, if you add includes=BluetoothA2DP.h to the library.properties file, than only that header is imported in to the code, when working with an Arduino IDE.

pschatzmann, 2022-12-19:
I am aware of this setting, but I think it would be a bad choice to punish everyone who installs this library by making the compilation of all sketches slower. 

It is far better to give the user the choice to import the project when it is really needed.

podaen, 2022-12-19:
Didn't know that was slower. Can the compiling time not be reduced with?

```
#ifndef _x_h //beginning of header file
#define _x_h

#endif// end of header file
```

pschatzmann, 2022-12-19:
If you add this definition Arduino needs to include this file and all related files in all sketches. 
Your proposed guard does not help here. 

ps. this is identical with #pragma once