pschatzmann/ESP32-A2DP #348: Voltage subtelty in PCM5102 DAC wiki docs

markchadwick, 2022-12-29:
On the (excellent!) wiki page for the [PCM5102 DAC](https://github.com/pschatzmann/ESP32-A2DP/wiki/External-DAC#pcm5102-dac), it specifies 3.3v for VIN. As per the [datasheet](https://www.ti.com/lit/ds/symlink/pcm5102a.pdf?ts=1672323619556&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FPCM5102A), this is the correct voltage for the PCM5102A itself.

Any listing I've found for the board pictured advertises a 5V input. Experimentally, the board will indeed sometimes "make sounds" on 3.3V, just not the ones you might expect. Switching to 5V resolves the issue

Not sure how to best make that distinction in the docs, but thanks for a really wonderful library!

clarkbremer, 2022-12-30:
I agree.  The board like [this one]( https://www.aliexpress.us/item/3256802711963831.html) has the 3.3v regulator built in.  BTW, I can't believe these boards are that cheap.  I can't even find the bare PCM5102 chip for less than that, and look at all the little discrete components you'd have to solder yourself if you did.  I have a custom PCB for my design, and I solder a bare ESP32 on it, but I use the DAC module like a daughter card.

pschatzmann, 2022-12-30:
I updated the documentation to use 5V

closed 2022-12-30 by pschatzmann