pschatzmann/ESP32-A2DP #354: ESP32 A2DP + UDA1334A Bluetooth Receiver - Bad audio quality, stutter and out of sync

vischio96, 2023-01-19:
Hi,

First of all, i apologize if this might be a stupid or obviuos question.

I am using an ESP32 board with external DAC (UDA1334A) flashed with your firmware (Bluetooth receiver with metadata), compiled with the custom I2S config that you wrote for that specific DAC:

_"It seems that most DACs from Adafruit have a different byte order then what the ESP32 expects and therefore you need to indicate the communication format as I2S_COMM_FORMAT_I2S_LSB"_

Unfortunately, the audio quality is awful, full of stutters and out of sync.

I uploaded a video on youtube to let you listen. 

[https://youtu.be/02Km2VGLkGU](url)

I tried to lower the debug level but nothing changed. 

Do you have any advice for that?

Thank you!

pschatzmann, 2023-01-19:
My first thought would be some shaky contacts or problems with the power supply. 
Did you try to connect some headphones instead of your amplifier ?

vischio96, 2023-01-19:
Hi and thank you for your super fast answer!

I tried several power supplies (laptop USB ports, powerbank and phone charger) and two amplifier boards, the result is the same :(
I made the PCB with soldered connections just to be sure about contacts and noise. 

Could it be a faulty DAC ?

vischio96, 2023-01-19:
No i didn't try with the headphones yet.

I'll even try with internal DAC and i'll let you know here, thank you a lot by now!

vischio96, 2023-01-20:
Hi,

i finally found the issue and seems to be related to ESP32 Bluetooth antenna \ power, the music starts to sounds good when i place the phone in some specific position and orientation. I tried the RSSI sketch but it stays at zero. 

Can the power be adjusted somehow?

Thank you again.

pschatzmann, 2023-01-20:
Strange, I never had this type of issue. 
Maybe the antenna of your board is damaged...

vischio96, 2023-01-21:
Thank you for your answer, i'll try in the next days and i'll let you know here.

Have a nice weekend ;)

pschatzmann, 2023-02-11:
Closed due to inactivity...

closed 2023-02-11 by pschatzmann