pschatzmann/ESP32-A2DP #367: Add to the Arduino Library Manager

marsfan, 2023-02-18:
Just stumbled across this repository, and made a project I was working on so much simpler. It would have been easier for me to find the library if it was available through the built in Arduino Library manager. 
The submission guide can be found at https://github.com/arduino/library-registry. I ran the compliance tester and the only complaint that it had was that there is not a header file named `ESP32-A2DP.h`, but that was only a warning, not an error.

pschatzmann, 2023-02-18:
Peaple are already filing too many issues!
So I am reluctant to publish this via the library manager in fear that the number of issues will explode.
I just have not the time to deal with this!

closed 2023-02-18 by pschatzmann

marsfan, 2023-02-19:
@pschatzmann That's a pity. Makes keeping things up to date easier when I can update through the manager. I do understand the issue with people not reading documentation though.

pschatzmann, 2023-02-19:
I suggest that you have a look at using Git.
That's the way I keep my installations updated on my different machines.
It has the advantage that you can switch between different versions easily.