pschatzmann/ESP32-A2DP #372: Implementing a screen in the project

leonardokucarz, 2023-02-24:
Hello, my name is Leonardo Kucarz, I'm literally stuck because I don't know how to solve my problem, I'm trying to implement your library with a screen, I'm using the dev kit ESP32-SC01. Separately, the programs work, but when I put the two parts together, the moment I issue the lcd.init command, playback stops, and the screen works normally, I'll pass the script below for playback.

```
#include "BluetoothA2DPSink.h"

#define LGFX_AUTODETECT     // Autodetect board
#define LGFX_USE_V1         // set to use new version of library
#include <LovyanGFX.hpp>    // main library

static LGFX lcd;            // declare display variable

char lcd_buf[40];  // Check the size you need
char lcd_buf1[40];  // Check the size you need
bool flagstate = false;

BluetoothA2DPSink a2dp_sink;

void avrc_metadata_callback(uint8_t id, const uint8_t *text) {

  flagstate = true;  // set flag within metadata function

  if (id == 0x1)
    snprintf(lcd_buf, 40, "%s                                ", text);

  if (id == 0x2)
    snprintf(lcd_buf1, 40, "%s                                ", text);

}

void setup() {
  
  i2s_pin_config_t my_pin_config = {
    .bck_io_num = 26,
    .ws_io_num = 25,
    .data_out_num = 22,    
    .data_in_num = I2S_PIN_NO_CHANGE
  };

  a2dp_sink.set_pin_config(my_pin_config);
  a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);
  a2dp_sink.start("MyPlayer"); //bluetooth name

  //Remove this comand to make a test
  lcd.init(); 
}

void loop()
{
    if (flagstate == true)
    {
       
      lcd.setCursor(10,10 );
      lcd.printf(lcd_buf);

      lcd.setCursor(10, 20);
      lcd.printf(lcd_buf1);

      flagstate = false ;         // reset flag till next metadata callback
    }

  delay(10);
} 
```

[image]

If you remove the command lcd.init the sound return but the screen doesn't work.

If you can help me in any way I would be very grateful.

pschatzmann, 2023-02-24:
I never did anything with screens but here is what you should look for:
- make sure that there is no pin conflict between I2S and your screen
- both are using a lot of memory so try to figure out if you have enough RAM to run both!
- Last you might have an issue with task priorities and maybe your library is not allowing any A2DP events to process
- Did you try to use another LCD library ?

closed 2023-02-24 by pschatzmann

troisiemetype, 2023-03-11:
Hello,

I'm experiencing he same problem, with a different library : https://github.com/Bodmer/TFT_eSPI
I've tried to declare A2DP before or after TFT, move the init function, TFT takes precedence on A2DP.
I've tried to move the TFT calls to a second core, but it doesn't seem to change anything.

By trying to log some informations to Serial, I found that if I use the basic Arduino `println()` function, I have an output on the Serial console. But If I use esp32 stack `prinf()` function instead, nothing is sent, not even the startup info from the ESP.

That makes me think that could be a memory issue, but the difference in the sketch size is minimal,and there is no memory allocation made during the `tft.init()` function, just SPI communication.

I hope you will be able to help.

pschatzmann, 2023-03-11:
Did you set the Core Debug Level in Arduino ?
A2DP is logging the available RAM...

troisiemetype, 2023-03-11:
Hello, and thank you for your answer.

I Haven't set the core debug on this project, as I know that it can increase the size quite a lot. But I will to see if I can obtain information on memory used.

Regarding memory, I've printed the heap size for different configurations : 
Before initialization (at the very beginning of `setup()`), free heap is 243824
After starting a2dp only : 143384
After starting TFT only : 144556
After starting both : 129884
In empty loop, with a2dp running, without display : around 130000 (it varies).

I don't know for other TFT libraries, but the one I linked doesn't create a buffer for the screen (unless you explicitly use sprites and set one or several), it uses the partial refresh as needed by the functions used.

pschatzmann, 2023-03-11:
You are crazy: set the core debug level to verbose, so that you can see what's going on.  Only this way you can see after a connection has been established and you send audio data, if bluetooth is processing it.

And how do you analyse the stacktraces if the system is crashing on you, if you don't see it?

Your numbers also seem to be off. After having a connection I am getting
17:04:56.295 -> [  6412][I][BluetoothA2DPCommon.cpp:243] log_free_heap(): [BT_AV] Available Heap: 88284

troisiemetype, 2023-03-11:
I usually set the debug level to Debug, and use it extensively when coding ! But as your example code worked straight away, I didn't fell the need to until now.

I've set it to Verbose, and saw a few thing. The values for the Heap are not the same from `[debug info]` and from `Serial.println(ESP.getFreeHeap());` I believe it can be explained by the precise moment one or the other is called, and what is the state at this precise moment. However, there are some case where the two prints are very close together, like :
```
[ 13176][I][BluetoothA2DPSink.cpp:514] app_gap_callback(): [BT_AV] ESP_BT_GAP_MODE_CHG_EVT
[ 13181][I][BluetoothA2DPCommon.cpp:243] log_free_heap(): [BT_AV] Available Heap: 100912
124496
```
The second one is a call from the main loop, every two seconds. The first one comes after an `I2S stop`,when the music is paused on the sending device. That may explain the difference, but the `Serial` one is always higher by at least 10k.

The heap is almost the same with or without the TFT, around 120000 when playing. In both case A2DP receives music, and the audio_data_callback is called. Here is what happens when music is played, then stopped (connection being and staying established) :
```
[ 26677][D][BluetoothA2DPSink.cpp:1199] ccall_app_gap_callback(): [BT_AV] ccall_app_gap_callback
[ 26677][I][BluetoothA2DPSink.cpp:514] app_gap_callback(): [BT_AV] ESP_BT_GAP_MODE_CHG_EVT
[ 26682][I][BluetoothA2DPCommon.cpp:243] log_free_heap(): [BT_AV] Available Heap: 100720
[ 26690][D][BluetoothA2DPSink.cpp:1187] ccall_app_a2d_callback(): [BT_AV] ccall_app_a2d_callback
[ 26699][D][BluetoothA2DPSink.cpp:989] app_a2d_callback(): [BT_AV] app_a2d_callback
[ 26706][D][BluetoothA2DPSink.cpp:997] app_a2d_callback(): [BT_AV] app_a2d_callback ESP_A2D_AUDIO_STATE_EVT
[ 26716][D][BluetoothA2DPSink.cpp:318] app_work_dispatch(): [BT_API] app_work_dispatch event 0x1, param len 16
[ 26725][D][BluetoothA2DPSink.cpp:350] app_send_msg(): [BT_AV] app_send_msg
[ 26732][D][BluetoothA2DPSink.cpp:373] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x1
[ 26735][D][BluetoothA2DPSink.cpp:1022] audio_data_callback(): [BT_AV] audio_data_callback
[ 26741][I][BluetoothA2DPSink.cpp:376] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 26759][D][BluetoothA2DPSink.cpp:341] app_work_dispatched(): [BT_AV] app_work_dispatched
[ 26767][D][BluetoothA2DPSink.cpp:1219] ccall_av_hdl_a2d_evt(): [BT_AV] ccall_av_hdl_a2d_evt
[ 26775][D][BluetoothA2DPSink.cpp:574] av_hdl_a2d_evt(): [BT_AV] av_hdl_a2d_evt evt 1
[ 26782][D][BluetoothA2DPSink.cpp:583] av_hdl_a2d_evt(): [BT_AV] av_hdl_a2d_evt ESP_A2D_AUDIO_STATE_EVT
[ 26791][D][BluetoothA2DPSink.cpp:652] handle_audio_state(): [BT_AV] handle_audio_state evt 1
[ 26800][I][BluetoothA2DPSink.cpp:654] handle_audio_state(): [BT_AV] A2DP audio state: Started
[ 26808][I][BluetoothA2DPSink.cpp:665] handle_audio_state(): [BT_AV] i2s_start
[ 26829][D][BluetoothA2DPSink.cpp:1022] audio_data_callback(): [BT_AV] audio_data_callback
[ 26850][D][BluetoothA2DPSink.cpp:1022] audio_data_callback(): [BT_AV] audio_data_callback
...
[ 37026][D][BluetoothA2DPSink.cpp:1022] audio_data_callback(): [BT_AV] audio_data_callback
[ 37047][D][BluetoothA2DPSink.cpp:1022] audio_data_callback(): [BT_AV] audio_data_callback
[ 37066][D][BluetoothA2DPSink.cpp:1187] ccall_app_a2d_callback(): [BT_AV] ccall_app_a2d_callback
[ 37066][D][BluetoothA2DPSink.cpp:989] app_a2d_callback(): [BT_AV] app_a2d_callback
[ 37071][D][BluetoothA2DPSink.cpp:997] app_a2d_callback(): [BT_AV] app_a2d_callback ESP_A2D_AUDIO_STATE_EVT
[ 37081][D][BluetoothA2DPSink.cpp:318] app_work_dispatch(): [BT_API] app_work_dispatch event 0x1, param len 16
[ 37090][D][BluetoothA2DPSink.cpp:350] app_send_msg(): [BT_AV] app_send_msg
[ 37097][D][BluetoothA2DPSink.cpp:373] app_task_handler(): [BT_API] app_task_handler, sig 0x1, 0x1
[ 37106][I][BluetoothA2DPSink.cpp:376] app_task_handler(): [BT_API] app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
[ 37116][D][BluetoothA2DPSink.cpp:341] app_work_dispatched(): [BT_AV] app_work_dispatched
[ 37124][D][BluetoothA2DPSink.cpp:1219] ccall_av_hdl_a2d_evt(): [BT_AV] ccall_av_hdl_a2d_evt
[ 37132][D][BluetoothA2DPSink.cpp:574] av_hdl_a2d_evt(): [BT_AV] av_hdl_a2d_evt evt 1
[ 37139][D][BluetoothA2DPSink.cpp:583] av_hdl_a2d_evt(): [BT_AV] av_hdl_a2d_evt ESP_A2D_AUDIO_STATE_EVT
[ 37148][D][BluetoothA2DPSink.cpp:652] handle_audio_state(): [BT_AV] handle_audio_state evt 1
[ 37157][I][BluetoothA2DPSink.cpp:654] handle_audio_state(): [BT_AV] A2DP audio state: Suspended
[ 37165][W][BluetoothA2DPSink.cpp:671] handle_audio_state(): [BT_AV] i2s_stop
```

The output log is exactly the same in both case, except for the heap which varies a bit (100919 vs 100720 with TFT).  Note that the TFT as been init in the setup, but it's not written to at any moment.
It gives the impression that calling the `tft.init()` function has an impact on I2S, which I don't understand at the moment, as I have another project which is a portable audio player, where this TFT library is used. I'm gonna try to look more closely at both the TFT code and yours.

I thought of PSRAM, if I can I will give it a try. However the current project is on a custom PCB designed for the WROOM module

pschatzmann, 2023-03-11:
If this is really your log output when the TFT is active, then it is definitely not the memory. 
Can it be that the TFT library is using I2S functionality ? You could try to change the i2s port from 0 to 1.
But the most likely culprit would be some pin conflicts.
Did you double check that none of the I2S pins are used by the TFT library ?

troisiemetype, 2023-03-11:
I don't think th TFT library is using I2S. As I said, I use it on another project, where it's used with I2S, and with mp3 decoding.

I've doubled checked the pins, I see no conflicts. This is the config I use for I2S, defined in the setup before starting a2dp : 
```
	const i2s_pin_config_t pin_config = {
		.bck_io_num = 16,   					// Serial Clock (SCK)
		.ws_io_num = 5,    						// Word Select (WS)
		.data_out_num = 17,				 	
		.data_in_num = I2S_PIN_NO_CHANGE,		// Serial Data (SD)
	};
```

And these are the pins used by the TFT library, which are all defined in a dedicated header file : 
```
#define TFT_BL           33    // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define TFT_MOSI 21
#define TFT_SCLK 22
#define TFT_CS   25  // Chip select control pin
#define TFT_DC   26  // Data Command control pin
#define TFT_RST  27  // Reset pin (could connect to RST pin)
```
There are no other pins used by the library, to my knowledge.

I didn't know the ESP32 had two I2S channels, I will try it.

pschatzmann, 2023-03-11:
You could try a simple I2S test output while using your TFT just to confirm that the issue is with the interaction between TFT library and I2s output: https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-generator-i2s/streams-generator-i2s.ino

pschatzmann, 2023-03-11:
Did you try to comment out the PIN_BTN_BOOT functionality. I would definitly loose my audio on my AudioKit if I would mess with pin 0...

troisiemetype, 2023-03-11:
The example you gave helped me ! Not because I could have the TFT working with, but because I didn't have any sound. First because I obviously forgot to setup my pinout, and then for a reason I didn't understand at first.
And the advice to look at the strap pin BOOT helped to, by showing me something. It appears it isn't the cause, I believe it can be used for a user program, as long as it's accessible at startup.

Bref. The cause is the soft mute function of the PCM5102A DAC. For a reason I don't understand yet, when the TFT is init, this pin (pin 19) is driven low, despite `digitalWrite(PIN_DAC_SOFT_MUTE, 1);` at the end of `setup()`. I've tried to run the program with the pin as an input, and the display. If I then short the pin to VCC, I have the sound back.

I'm gonna dig in the TFT library a bit more. :)

pschatzmann, 2023-03-11:
Yes, pin 0 is mainly relevant for devices that need a master clock input (like the ES8388 in the AudioKit).
It would be interesting to know where the assignment to pin 19 is coming from...

troisiemetype, 2023-03-11:
I've found out : GPIO 19 is the default pin for SPI MISO on the ESP32. As I don't need feedback for the display in my program, I didn't set it. In fact there are several setup files for different displays in the library that don't set it. It then defaults to the default value, and is set as an input.
I had a doubt when touching the I2S + mute tracks on my PCB, the capacitive potential of my fingers was enough to make an awful noise.

I cannot be sure, but there are chances that @leonardokucarz was facing the same problem, even with another library, when he opened this issue.

Thank you for your help ! This afternoon I was ready to unmount my display and forget about it, I'm happy I will be able to make something with it ! :)

leonardokucarz, 2023-03-13:
Good morning, I ended up not answering anymore because I managed to find the problem thanks to your help, what was happening was a pin conflict, in my case it was pin 22, before that I measured the ram memory but this entire application did not use even half from memory. here is my code.

`// LGFX
#define LGFX_AUTODETECT     // Autodetcção de Placa
#define LGFX_USE_V1         // Usa a nova versão da biblioteca
#include <LovyanGFX.hpp>    // Livraria principal

// LVGL
#include <C:\Users\Leonardo\Documents\Arduino\libraries\lvgl-8.3.3\lvgl.h>
#include "ui.h"
#include "ui_events.c"

//BLUETOOTH
#include "BluetoothA2DPSink.h"
BluetoothA2DPSink a2dp_sink;

static LGFX lcd;            // declara a variavel do display

char lcd_buf[40];           // Tamanho da variavel nome da Musica
char lcd_buf1[40];          // Tamanho da variavel nome do Artista
bool flagstate = false;

/*Change to your screen resolution*/
static const uint32_t screenWidth = 480;
static const uint32_t screenHeight = 320;
static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf[screenWidth * 10];

void ProximaMusica(lv_event_t * e)
{
	lv_event_code_t event = lv_event_get_code(e);
  lv_obj_t *obj = lv_event_get_target(e);
  if (event == LV_EVENT_CLICKED) {
	a2dp_sink.next();
  }
}

void MusicaAnterior(lv_event_t * e)
{
	lv_event_code_t event = lv_event_get_code(e);
  lv_obj_t *obj = lv_event_get_target(e);
  if (event == LV_EVENT_CLICKED) {
	a2dp_sink.previous();
  }
}

void PlayPause(lv_event_t * e)
{
	lv_event_code_t event = lv_event_get_code(e);
  lv_obj_t *obj = lv_event_get_target(e);
  if (event == LV_EVENT_CLICKED) {
	a2dp_sink.play();
  }
}

void setup(void)
{
  initDisplay();
  lv_init();
  lv_disp_draw_buf_init(&draw_buf, buf, NULL, screenWidth * 10);

  //INICIA O DISPLAY LV
  static lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);

  //MUDA A RESOLUÇÃO DO DISPLAY
  disp_drv.hor_res = screenWidth;
  disp_drv.ver_res = screenHeight;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.draw_buf = &draw_buf;
  lv_disp_drv_register(&disp_drv);

  //Initialize the (dummy) input device driver
  static lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read_cb = my_touchpad_read;
  lv_indev_drv_register(&indev_drv);

  ui_init();

  i2s_pin_config_t my_pin_config = {
    .bck_io_num = 26,
    .ws_io_num = 25,
    .data_out_num = 2,    
    .data_in_num = I2S_PIN_NO_CHANGE
  };

  a2dp_sink.set_pin_config(my_pin_config);
  a2dp_sink.set_bits_per_sample(16);  
  a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);
  a2dp_sink.start("LK_W-Series"); 

}
  

void loop()
{ 
  lv_timer_handler();
  delay(1); // do nothing
}

//INICIA O DISPLAY
void initDisplay() {
  lcd.init();
  lcd.setRotation(1);
  lcd.setBrightness(255);
  lcd.fillScreen(TFT_BLACK);
}

//CHAMA AS INFORMAÇÕES
void avrc_metadata_callback(uint8_t id, const uint8_t *text) {

  flagstate = true;  // set flag within metadata function
  if (id == 0x1)
    snprintf(lcd_buf, 40, "%s                                ", text);
    lv_label_set_text(ui_Nome_Musica, lcd_buf);
     //lv_label_set_text(ui_SpeedLabel, "0");
  if (id == 0x2)
    snprintf(lcd_buf1, 40, "%s                                ", text);
    lv_label_set_text(ui_Nome_da_banda, lcd_buf1);
}

//CONFIG Display flushing
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p) {
  uint32_t w = (area->x2 - area->x1 + 1);
  uint32_t h = (area->y2 - area->y1 + 1);

  lcd.startWrite();
  lcd.setAddrWindow(area->x1, area->y1, w, h);
  lcd.writePixels((lgfx::rgb565_t *)&color_p->full, w * h);
  lcd.endWrite();

  lv_disp_flush_ready(disp);
}

//CONFIG Read the touchpad
void my_touchpad_read(lv_indev_drv_t *indev_driver, lv_indev_data_t *data) {

  uint16_t x, y;
  if (lcd.getTouch(&x, &y)) {
    data->state = LV_INDEV_STATE_PR;
    data->point.x = x;
    data->point.y = y;

  } else {
    data->state = LV_INDEV_STATE_REL;
  }
}`

to run this you need others libraries because I'm using Squareline to made the front-end interface. 
[image]