pschatzmann/ESP32-A2DP #373: Audio is not played if user disconnects and connects phone

garageeks, 2023-02-24:
Dear Phil,
I'm stumbing about a problem:
While running this example
`https://github.com/pschatzmann/ESP32-A2DP/blob/main/examples/bt_music_receiver_simple/bt_music_receiver_simple.ino`
an Android phone is able to connect and play music flawlessly.
However if I disconnect phone and then connect it again, phone get connected but no audio is coming out of the speaker.
I'm using Arduino ESP32 SDK 2.0.4 and the only thing I added before starting the sink is a custom I2S pinout. I tried 2.0.6 and no change.

```
i2s_pin_config_t my_pin_config = {
				.bck_io_num = 4,
				.ws_io_num = 23,
				.data_out_num = 12,
				.data_in_num = I2S_PIN_NO_CHANGE
		};
    a2dp_sink.set_pin_config(my_pin_config);
```

I have tried a workaround that once the callback "Disconnected" is caught, it runs `a2dp_sink.end(false);` and then reinitializes the sink immediately after, but it crashes...

Thank you for your feedback
Nick

garageeks, 2023-02-25:
Disabling i2s_stop call appears to solve the issue, I tried disconnecting and reconnecting manually 10 times in a row without breaking it.
But there must be a reason why you were stopping and starting i2s interface in the first place...

```
void BluetoothA2DPSink::handle_audio_state(uint16_t event, void *p_param){
    ESP_LOGD(BT_AV_TAG, "%s evt %d", __func__, event);
    esp_a2d_cb_param_t* a2d = (esp_a2d_cb_param_t *)(p_param);
    ESP_LOGI(BT_AV_TAG, "A2DP audio state: %s", to_str(a2d->audio_stat.state));
    
    // callback on state change
    audio_state = a2d->audio_stat.state;
    if (audio_state_callback!=nullptr){
        audio_state_callback(a2d->audio_stat.state, audio_state_obj);
    }

    if (is_i2s_output){
        if (ESP_A2D_AUDIO_STATE_STARTED == a2d->audio_stat.state) { 
            m_pkt_cnt = 0; 
            ESP_LOGI(BT_AV_TAG,"i2s_start");
            if (i2s_start(i2s_port)!=ESP_OK){
                ESP_LOGE(BT_AV_TAG, "i2s_start");
            }

        } else if ( ESP_A2D_AUDIO_STATE_REMOTE_SUSPEND == a2d->audio_stat.state || ESP_A2D_AUDIO_STATE_STOPPED == a2d->audio_stat.state ) { 
            ESP_LOGW(BT_AV_TAG,"i2s_stop");
            //i2s_stop(i2s_port);
            //i2s_zero_dma_buffer(i2s_port);
        }

        if (audio_state_callback_post!=nullptr){
            audio_state_callback_post(a2d->audio_stat.state, audio_state_obj_post);
        }    
    }
}
```

pschatzmann, 2023-02-25:
I am sorry, I cant reproduce your issue!
If you check in the board manager: what ESP32 version are you using. I just tested with 2.0.6 which is the latest released version.

You can activate the logging in the ESP32 tools menu. Right before you start to play some audio you should see the following line:
11:40:58.595 -> [743935][I][BluetoothA2DPSink.cpp:668] handle_audio_state(): [BT_AV] A2DP audio state: Started
This will trigger the activation of I2S

closed 2023-02-25 by pschatzmann

garageeks, 2023-02-25:
Thank you for testing it out. Turns out somehow the library got corrupted.
I did a "Clean all" in PlatformIO and started behaving like expected, both on 2.0.4 and 2.0.6.
However I consistently found 2.0.6 is using a lot more heap memory compared to 2.0.4.
With 2.0.4 I got this.
Playing -> heap 264676, free 124556, max 94196
Stop -> heap 264868, free 129080, max 100340
Disconnect -> heap 266516, free 136240, max 100340

With 2.0.6:
Playing -> heap 257304, free 112716, max 81908
Stop -> heap 257484, free 117996, max 90100
Disconnect -> heap 259164, free 125176, max 90100

Almost 12000 bytes gone, which, given the enormous size of the BT stack, can become a serious problem...
I'll open an issue to the esp32-arduino maintainers.