pschatzmann/ESP32-A2DP #374: disable Bluetooth, then re-enable it

garageeks, 2023-02-25:
Hi Phil,
Sorry to bother, I've been looking for a method to switch off Bluetooth but failed.
I used `a2dp_sink.end(false);` I paid attention not to set the flag to true, as it will permanently disable the sink untill it is rebooted.
This ends the sink, but it's still possible to connect to the ESP32 via Bluetooth, and if then audio stream is started, ESP crashes.

I added a `btStop();` call after `a2dp_sink.end(false);` , this prevents connecting to the ESP32 thereafter, as expected, however when the sink is re-enabled, the Bluetooth connection is not established anymore.
I had a look at how the bluetooth stack is initialized in your library. I then tried `esp_bluedroid_disable();` instead of `btStop();` to avoid killing completely the Bluetooth controller but just the stack, still no connection once it is re-enabled.

Do you have any suggestions?

This looks similar to this issue:
`https://github.com/pschatzmann/ESP32-A2DP/issues/37`

pschatzmann, 2023-02-25:
Did you analyze the stacktrace for your crash ? I haven't re-tested the restart after a2dp_sink.end(false);  for quite some time.

If you check the official ESP32 documentation, it is documented that you can't restart bluetooth after you brought it down, so I usually rely on restarting the ESP32 as the cleanest way...

pschatzmann, 2023-02-26:
I was testing the stop/start with an AudioKit and an Android Phone and it was working perfectly w/o any issues. Here is my test sketch:
```

#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"
#include "AudioLibs/AudioA2DP.h"

BluetoothA2DPSink a2dp_sink;
AudioKitStream kit;

void start(bool, int, void*){
  a2dp_sink.start("AudioKit");
}

void stop(bool, int, void*){
  Serial.println("-->End");
  a2dp_sink.end();
}

// Write data to AudioKit in callback
void read_data_stream(const uint8_t *data, uint32_t length) {
    kit.write(data, length);
}

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup output
  auto cfg = kit.defaultConfig(TX_MODE);
  cfg.sd_active = false;
  kit.begin(cfg);
  
  kit.addAction(PIN_KEY5, start);
  kit.addAction(PIN_KEY6, stop);

  // register callback
  a2dp_sink.set_stream_reader(read_data_stream, false);
  a2dp_sink.start("AudioKit");
}

void loop() {
  kit.processActions();
}
```

garageeks, 2023-02-27:
Thank you for taking time to test this. I know most issues are not due to your library, but the fragile BT stack behind it.
I'll conduct more tests and let you know.

garageeks, 2023-02-27:
Hi Phil, I'm not sure which tests did you perform. The commands `a2dp_sink.start("AudioKit");` and `a2dp_sink.end();` are working fine by themselves.
The problem is that the BT stack is still available for connections once you call `a2dp_sink.end(); `
I assumed it would disable BT in addition to the sink, as `a2dp_sink.start("AudioKit"); `starts the BT stack itself.
But looking at the source code, it seems it is only uninstalling the I2S driver, but not closing the stack.
Did you try to connect to the ESP32 after calling `a2dp_sink.end();` ?
If you do it, you should see these error popping out
`[326344][E][BluetoothA2DPSink.cpp:366] app_send_msg(): [BT_API] app_send_msg app_send_msg failed`
If you try to play audio at this stage, it will crash. However if you start the sink again while the host is connected but not streaming music, the sink will happily work.

If this is the expected behaviour, I will ask on the esp32-arduino team how to properly disable and re-enable the BT stack.
As the BT stack consumes too much heap memory, I'm unable to reliably send data on Wi-Fi while BT is enabled, therefore I'm attempting to send data via Wi-Fi periodically while there isn't a host connected to the BT sink. That's why I'm looking to completely close the sink and re-enable it afterwards.

pschatzmann, 2023-02-27:
I was seeing this error, but it did not have any impact on the functionality. As you can see in my sketch above I used separate buttons to stop() and to start() and I repeated the excercise several times.

If you check in the Board Manager, what ESP32 version are you using ?

garageeks, 2023-02-27:
I'm working with PlatformIO, and selected esp32dev as board.
I tried switching between 2.0.4 and 2.0.6 SDK but the behaviour is the same: the BT stack still accept connection once the sink is ended, but then playing something results in a crash.

I'm sorry, but was unable to find references in the official ESP32 documentation about the inability to restart the bluetooth stack; but the documentation is huge and maybe I hadn't looked hard enough. Could you please provide a link?

pschatzmann, 2023-02-27:
Found it: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/controller_vhci.html
Search for esp_bt_controller_mem_release

garageeks, 2023-02-27:
Indeed if you call esp_bt_controller_mem_release it is an irreversible operation until it's rebooted.
However I'm not calling it; I'm just running the four de-initializing routines.
In fact, in a comment in these file:
https://github.com/espressif/esp-idf/blob/master/components/bt/include/esp32/include/esp_bt.h

They state to fully remove the BT stack it's five function calls:
 _*      esp_bluedroid_disable();
 *      esp_bluedroid_deinit();
 *      esp_bt_controller_disable();
 *      esp_bt_controller_deinit();
 *      esp_bt_mem_release(ESP_BT_MODE_BTDM);_

Technically if the memory release function is not called, it should allow re-initializing it again.

The ESP crashes only when playing audio is attempted after ending the A2DP sink, otherwise it doesn't crash, it is just not possible to connect to it.

The stacktrace makes sense, as the I2S driver has been uninitialized. Maybe it could be useful to check if I2S is initialized before writing to it.

```
PC: 0x400f5294: i2s_write at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/i2s.c line 2128
EXCVADDR: 0x00000018

Decoding stack results
0x400f5291: i2s_write at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/i2s.c line 2128
0x400e69ad: BluetoothA2DPSink::i2s_write_data(unsigned char const*, unsigned int) at .pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSink.cpp line 1265
0x400e6a95: BluetoothA2DPSink::write_audio(unsigned char const*, unsigned int) at .pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSink.h line 377
0x40215b91: BluetoothA2DPSink::audio_data_callback(unsigned char const*, unsigned int) at .pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSink.cpp line 1065
0x400e528f: ccall_audio_data_callback(uint8_t const*, uint32_t) at .pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSink.cpp line 1203
0x4014a4ed: btc_a2dp_sink_data_ready at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/btc/profile/std/a2dp/btc_a2dp_sink.c line 159
0x4014369d: osi_thread_generic_event_handler at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/osi/thread.c line 425
0x40143633: osi_thread_run at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/osi/thread.c line 165
```

pschatzmann, 2023-02-27:
Quite a long time ago I spent quite some time on this, but I never got it to work reliably. That was many ESP32 releases ago!
Feel free to improve it, because I am also not happy with the current state...

garageeks, 2023-02-27:
Thank you for your time, your work is amazing and I don't see what could be improved if the underlying libraries are shaky...
I'll wait for an answer from the Espressif guys, otherwise I think I will reboot the ESP when I want to shut the sink off.

If I have new findings I'll post here, in the meantime I close the issue.

closed 2023-02-27 by garageeks

kiliw, 2023-03-01:
Hey, I just wanted to jump in here quickly. Thanks for the nice library. So far everything worked very smooth. I'm building a device where I want a Bluetooth speaker mode in some situations. My idea was to enable it via 
```
a2dp_sink.start("box");
```
as soon as a button is pressed for about 2 seconds. 
Is the same button pressed again, I want to disable the BT service. In addition, it would be great to use Wi-Fi once the BT mode is disabled.

As far as I followed the discussion, this seems currently not possibles. Unfortunately I'm not very deep into microcontrollers, but this seems rather to be a library problem on the side of Espressif, correct?

All the best, looking forward to finding a good solution.

pschatzmann, 2023-03-01:
WIFI and Bluetooth can not be used together. https://github.com/pschatzmann/ESP32-A2DP/wiki/WIFI-and-A2DP-Coexistence

As described above the best way is to restart the ESP32 when you are done with Bluetooth, unless you can live with the fact that Bluetooth is using a lot of memory even if it is stopped...

kiliw, 2023-03-01:
> WIFI and Bluetooth can not be used together. https://github.com/pschatzmann/ESP32-A2DP/wiki/WIFI-and-A2DP-Coexistence
>
> As described above the best way is to restart the ESP32 when you are done with [...]

Thanks for the fast answer. Ok, I will look for a solution to restart the esp.

Regarding Wi-Fi and BT, I came across the WIKI and had a quick discussion with a friend. Is this due to the fact that there is only one antenna which can be used by either BT or Wi-Fi? Or can they just not be used in the same file? 
My idea was to have two working modes. First one uses BT and if this is done, I turn it off and enable Wi-Fi, and vice versa. So only one connection method is used, but the code is in one file.

garageeks, 2023-03-03:
In my tests, Wi-Fi transmission was working while streaming A2DP audio. Wi-Fi and BT co-existence is enabled by default. It is the lack of free heap memory that made the whole thing unstable, i.e. when disconnecting a phone and connecting it again.
It is also possible to reclaim memory when BT is not necessary anymore, and keep the code running, but it is not possible to re-initialize BT without a reboot. It is a limitation of the Espressif BT stack.

pschatzmann, 2023-03-03:
What hardware did you use and how did you prevent to run out of memory ?
Do you have a test sketch where we can reproduce your result ?