pschatzmann/ESP32-A2DP #378: Compilation issues on ESP32S3

0x0fe, 2023-03-07:
When trying to compile for ESP32-S3 apparently there is no conditional statements to handle the diferent ESP32 variants (for example, the ESP32S does not have DAC).

Paths redacted for clarity

```
BluetoothA2DPSink.cpp:248:53: error: 'FUNC_GPIO0_CLK_OUT1' was not declared in this scope
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
                                                     ^~~~~~~~~~~~~~~~~~~
2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/soc.h:136:45: note: in definition of macro 'REG_WRITE'
             (*(volatile uint32_t *)(_r)) = (_v);                                                                       \
                                             ^~
2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/io_mux_reg.h:93:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
src\BluetoothA2DPSink.cpp:248:13: note: in expansion of macro 'PIN_FUNC_SELECT'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
             ^~~~~~~~~~~~~~~
src\BluetoothA2DPSink.cpp:248:53: note: suggested alternative: 'FUNC_GPIO20_CLK_OUT1'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
                                                     ^~~~~~~~~~~~~~~~~~~
2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/soc.h:136:45: note: in definition of macro 'REG_WRITE'
             (*(volatile uint32_t *)(_r)) = (_v);                                                                       \
                                             ^~
2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/io_mux_reg.h:93:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
src\BluetoothA2DPSink.cpp:248:13: note: in expansion of macro 'PIN_FUNC_SELECT'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
             ^~~~~~~~~~~~~~~
src\BluetoothA2DPSink.cpp:252:53: error: 'FUNC_U0TXD_CLK_OUT3' was not declared in this scope
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
                                                     ^~~~~~~~~~~~~~~~~~~
2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/soc.h:136:45: note: in definition of macro 'REG_WRITE'
             (*(volatile uint32_t *)(_r)) = (_v);                                                                       \
                                             ^~
2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/io_mux_reg.h:93:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
src\BluetoothA2DPSink.cpp:252:13: note: in expansion of macro 'PIN_FUNC_SELECT'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
             ^~~~~~~~~~~~~~~
src\BluetoothA2DPSink.cpp:252:53: note: suggested alternative: 'FUNC_U0TXD_CLK_OUT1'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
                                                     ^~~~~~~~~~~~~~~~~~~
2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/soc.h:136:45: note: in definition of macro 'REG_WRITE'
             (*(volatile uint32_t *)(_r)) = (_v);                                                                       \
                                             ^~
2.0.6/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/io_mux_reg.h:93:46: note: in expansion of macro 'REG_SET_FIELD'
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
                                              ^~~~~~~~~~~~~
src\BluetoothA2DPSink.cpp:252:13: note: in expansion of macro 'PIN_FUNC_SELECT'
             PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
             ^~~~~~~~~~~~~~~
src\BluetoothA2DPSink.cpp: In member function 'size_t BluetoothA2DPSink::i2s_write_data(const uint8_t*, size_t)':
src\BluetoothA2DPSink.cpp:1255:33: error: 'I2S_MODE_DAC_BUILT_IN' was not declared in this scope
     if (this->i2s_config.mode & I2S_MODE_DAC_BUILT_IN) {
                                 ^~~~~~~~~~~~~~~~~~~~~
src\BluetoothA2DPSink.cpp:1255:33: note: suggested alternative: 'I2S_MODE_MASTER'
     if (this->i2s_config.mode & I2S_MODE_DAC_BUILT_IN) {
                                 ^~~~~~~~~~~~~~~~~~~~~
                                 I2S_MODE_MASTER
```

pschatzmann, 2023-03-07:
Because I don't want to maintain the differences in I2S for the different ESP32 versions in multiple projects, I have deactivated the custom I2S output in flite!

You can still output audio to I2S via the Stream API.

pschatzmann, 2023-03-07:
https://github.com/pschatzmann/ESP32-A2DP/wiki/Support-for-esp32-c3%3F

closed 2023-03-07 by pschatzmann

0x0fe, 2023-03-07:
i see, it makes sense. What is Flite?

pschatzmann, 2023-03-07:
Forget that answer. I wanted to answer some other issue that I was just received in regard of compilation errors for the flite project with the c3...

0x0fe, 2023-03-07:
https://github.com/pschatzmann/ESP32-A2DP/wiki/Support-for-esp32-c3%3F

Ah, i see, no support for bluetooth classic on S3, I knew C3 did not have but i did not even realized S3 doesnt have either, quite a stupid move from epressif.

0x0fe, 2023-03-07:
it could also have Bleutooth LE Audio, but they did not update the stack to 5.2, and its an extension, i guess they have to pay for