pschatzmann/ESP32-A2DP #390: Help for Builing a useful GUI and get Data from A2DP

Bl4ckHol3ee, 2023-03-20:
So, i really dont get further.

I got a few problems.... And i tried everything.... Even GPT didn't help.

I got a 3.5" TFT Display, on which i want to show every information i got:

1. The date and the current time
2. The paired-devices name
3. The SongTitle
4. The artistName
5. A Graph, which shows the current state of playtime (f.e.: (00:00 sec) --------I----------------------------- (03:35 sec))
                                                                                                                            ^: The current state (f.e. 1 min.)
6. three buttons: "Last Song" (with a arrow backwards), "Pause" ("II") and "Next song" (with a attow forward)
7. a Graph upwards on which i can adjust the current volume. (I think everybody knows what i mean....)

[image]

This is how far i have been coming since now.

My biggest problem is, how to get the SongTitle/InterpretName in a variable, which i can use for sharing with the tft display.

[Please be kind. My understanding of GUIs and the components in this special project is limited. I really would apprechiate some useful comments and help of people which are into the material. Thank you a lot!!!! <3 ]

pschatzmann, 2023-03-20:
You need to store some copies of the metadata in some global variables. I would expect that somethings like this should work:
```C++
BluetoothA2DPSink a2dp_sink;
String title="";
String artist="";

void avrc_metadata_callback(uint8_t id, const uint8_t *text) {
  switch(id){
     case ESP_AVRC_MD_ATTR_TITLE:
        title = text;
        break;
     case ESP_AVRC_MD_ATTR_ARTIST:
        artist = text;
        break;
  }
}

void setup() {
  Serial.begin(115200);
  a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);
  a2dp_sink.start("MyMusic");  
}
```

Bl4ckHol3ee, 2023-03-20:
Yes, that's one thing i already tried (thank you anyway!!) 

But i always receive a error like this:

D:\OpenEnd\HomE\003__Arduino\mp3_Player_Display\mp3_Player_Display.ino: In function 'void avrc_metadata_callback(uint8_t, const uint8_t*)':
D:\OpenEnd\HomE\003__Arduino\mp3_Player_Display\mp3_Player_Display.ino:63:17: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'const char*' [-fpermissive]
         titlexy = text;
                 ^
In file included from C:\Users\nbybl\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:146:0,
                 from C:\Users\nbybl\AppData\Local\Temp\arduino\sketches\AC091DBEB7404FB048FC78D6D7FDCFD4\sketch\mp3_Player_Display.ino.cpp:1:
C:\Users\nbybl\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/WString.h:97:18: note:   initializing argument 1 of 'String& String::operator=(const char*)'
         String & operator =(const char *cstr);
                  ^
D:\OpenEnd\HomE\003__Arduino\mp3_Player_Display\mp3_Player_Display.ino:66:18: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'const char*' [-fpermissive]
         artistxy = text;
                  ^
In file included from C:\Users\nbybl\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:146:0,
                 from C:\Users\nbybl\AppData\Local\Temp\arduino\sketches\AC091DBEB7404FB048FC78D6D7FDCFD4\sketch\mp3_Player_Display.ino.cpp:1:
C:\Users\nbybl\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/WString.h:97:18: note:   initializing argument 1 of 'String& String::operator=(const char*)'
         String & operator =(const char *cstr);
                  ^

Bibliothek ESP32-A2DP in Version 1.7.1 im Ordner: D:\OpenEnd\HomE\003__Arduino\libraries\ESP32-A2DP  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: C:\Users\nbybl\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SPI  wird verwendet
Bibliothek TFT_eSPI in Version 2.5.23 im Ordner: D:\OpenEnd\HomE\003__Arduino\libraries\TFT_eSPI  wird verwendet
Bibliothek FS in Version 1.0 im Ordner: C:\Users\nbybl\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS  wird verwendet
Bibliothek SPIFFS in Version 1.0 im Ordner: C:\Users\nbybl\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SPIFFS  wird verwendet
exit status 1

Compilation error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'const char*' [-fpermissive]

Bl4ckHol3ee, 2023-03-20:
and if i write it like:
void avrc_metadata_callback(uint8_t id, const uint8_t *text) {
  switch(id){
     case ESP_AVRC_MD_ATTR_TITLE:
        titlexy = const uint8_t *(text);
        break;
     case ESP_AVRC_MD_ATTR_ARTIST:
        artistxy = const uint8_t *(text);
        break;
  }
}
The error changes to:

Compilation error: expected primary-expression before 'const'

pschatzmann, 2023-03-20:
The data types of this method are indeed not helpfull. You need to cast: I updated my example...

Bl4ckHol3ee, 2023-03-20:
If it is helpful for you

This one is working (but only in the Serial Monitor):

void avrc_metadata_callback(uint8_t id, const uint8_t *text) {
  Serial.printf("==> AVRC metadata rsp: attribute id 0x%x, %s/n", id, text);
  }
}

But here i got also the problem, if i change it to:

void avrc_metadata_callback(uint8_t id, const uint8_t *text) {
  Serial.printf("==> AVRC metadata rsp: attribute id 0x%x, %s/n", id, text);

  if (id == 0x02) { // ID for song title
    songTitle = (char*) text;
  }
  else if (id == 0x03) { // ID for artist name
    artistName = (char*) text;
  }
}

Then there is a Guru Meditation Error

Bl4ckHol3ee, 2023-03-20:
> The data types of this method are indeed not helpfull. You need to cast: I updated my example...

Sorry, which example did you update? I'm confused. :D 

Pschatzmann, thank you for your help! <3

Bl4ckHol3ee, 2023-03-20:
It is working a few seconds/minutes, but then the Core paniced...
I won't figure it out, why...

13:07:50.880 -> Voices (feat. Oliver Tree)
13:07:50.925 -> KSI, Oliver Tree
13:08:00.854 -> E0ISR
13:08:00.886 -> ASSERT_ERR(co_list_is_empty(&ld_fm_env.fifo) || (timestamp != ((struct ld_fm_frm_elt*)ld_fm_env.fifo.last)->timestamp)), in ld_fm.c at line 936
13:08:01.185 -> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
13:08:01.250 -> Core 0 register dump:
13:08:01.282 -> PC      : 0x400869dc  PS      : 0x00060634  A0      : 0x80088676  A1      : 0x3ffbf2d0  
13:08:01.378 -> A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x3ffbdd3c  
13:08:01.441 -> A6      : 0x3ffbdd3c  A7      : 0x0000119c  A8      : 0x800869dc  A9      : 0x3ffbf2b0  
13:08:01.538 -> A10     : 0x00000090  A11     : 0x00000090  A12     : 0x00000010  A13     : 0xffffffff

pschatzmann, 2023-03-20:
You will need to analyze the stacktrace

Bl4ckHol3ee, 2023-03-20:
The error message you posted indicates that a process called "ld_fm" is crashing on a device that may be running with a CPU. It looks like a watchdog timer interrupt (wdt timeout) occurred and the process didn't respond to it in time, resulting in a panic state.

However, it's hard to say what exactly caused the problem, as the error message doesn't provide specific details about what "ld_fm" is doing or what type of device is affected. It is possible that there is a firmware or hardware failure or that a memory overflow has occurred.

It is best to consult an expert who is familiar with the device in question for more thorough troubleshooting.

This is what ChatGPT is saying about the error.

So the error always comes up after ~1 minute.

I will go deeper into that when i am home again.

pschatzmann, 2023-03-20:
Yes, most likely you are constantly updating the screen and not only when it has changed...

Bl4ckHol3ee, 2023-03-24:
So i was busy the last few days, but i got further!

I created this following code since the last message: And i am very happy, that it is working so far. 

But now i got a new problem, which is stopping me from coming further..... 
I need the version 4.0 (or newer) on my ESP32s-NodeMCU or my ESP-WROOM-32-DevKit-V4, so that i can ask the Controller what's the name of the paired Device....

I am searching since 2 days, how to fix this issue. I tried over 10 libraries. 
But either i am such a dumb fool or it is really so confusing to put the newest Soft-/Firmware onto the ESPs(???)
**Could anyone help me out?**
_For me it would be the easiest way, to connect the ESP via USB onto the PC and just start a Programm to update it._
(BUT there is, so i believe, NO WAY!)

I will update, when i am finished. (Could be that it helps somebody in the futute.) So please don't close the issue.

`#include <BluetoothA2DPSink.h>
#include <SPI.h>
#include <TFT_eSPI.h>
#include "Free_Fonts.h"  // Include the header file attached to this sketch
#include "A2DPVolumeControl.h"

const int P_lauter =  0;
const int P_leiser = 15;

#define BLACK 0x0000
#define WHITE 0xFFFF
#define Grau 0x8888

TFT_eSPI tft = TFT_eSPI();  // Invoke custom library

int16_t h;
int16_t w;

char ende1[4] = "...";

int inc = -2;

float xx, xy, xz;
float yx, yy, yz;
float zx, zy, zz;
float fact;
int Xan, Yan;
int Xoff;
int Yoff;
int Zoff;
struct Point3d {
  int x;
  int y;
  int z;
};
struct Point2d {
  int x;
  int y;
};
int LinestoRender;     // lines to render.
int OldLinestoRender;  // lines to render just in case it changes. this makes sure the old lines all get erased.
struct Line3d {
  Point3d p0;
  Point3d p1;
};
struct Line2d {
  Point2d p0;
  Point2d p1;
};
Line3d Lines[20];
Line2d Render[20];
Line2d ORender[20];

const char *geraetename = "Esp32s-Wroom-BT";
const char *version = "23.03.2023";
[... 411 lines cut ...]
  Lines[8].p1.z = -50;
  Lines[9].p0.x = 50;
  Lines[9].p0.y = -50;
  Lines[9].p0.z = 50;
  Lines[9].p1.x = 50;
  Lines[9].p1.y = -50;
  Lines[9].p1.z = -50;
  Lines[10].p0.x = -50;
  Lines[10].p0.y = 50;
  Lines[10].p0.z = 50;
  Lines[10].p1.x = -50;
  Lines[10].p1.y = 50;
  Lines[10].p1.z = -50;
  Lines[11].p0.x = 50;
  Lines[11].p0.y = 50;
  Lines[11].p0.z = 50;
  Lines[11].p1.x = 50;
  Lines[11].p1.y = 50;
  Lines[11].p1.z = -50;

  LinestoRender = 12;
  OldLinestoRender = LinestoRender;
}`

Thank you anyway Pschatzmann! Very nice library you got there.

pschatzmann, 2023-03-24:
You need version 4 of what ? Please note the overall architecture of things.
The latest Arduino ESP version (if you check in the board manager) is ESP32 Arduino 2.0.7. This comes with ESP-IDF 4.4.4.

If you work with ESP-IDF directly the newest version that you can choose is ESP-IDF 5.0.1, but looking at your code you seem to be using Arduino.

Bl4ckHol3ee, 2023-03-24:
Oh, sorry. I thought there is just ESP-IDF for ESPs. I got to be honest. My knowledge about those things is limited. 

But i can tell you some information:
>
ESP32 Systeminformationen:
  ESP-IDF-Version: v3.3.5-1-g85c43024c
  Chip-ID: 798945480
  Chip-Bezeichnung: ESP32-D0WDQ5
  CPU-Takt: 240 MHz
<

I want to use the code from your library, this one:
`    #ifdef ESP_IDF_4
    /// Get the name of the connected source device
    virtual const char* get_connected_source_name();
    /// Provides the result of the last result for the esp_avrc_tg_get_rn_evt_cap() callback (Available from ESP_IDF_4)
    bool is_avrc_peer_rn_cap(esp_avrc_rn_event_ids_t cmd) {
        return esp_avrc_rn_evt_bit_mask_operation(ESP_AVRC_BIT_MASK_OP_TEST, &s_avrc_peer_rn_cap, cmd);
    }
    /// Returns true if the is_avrc_peer_rn_cap() method can be called
    bool is_avrc_peer_rn_cap_available() {
        return s_avrc_peer_rn_cap.bits != 0;
    }
    #endif   `

But it doesn't work out. 

Can you tell me, what i got to do?
(Step by Step would be awesome.....!)

If it works, i buy you a coffee <3

pschatzmann, 2023-03-24:
You still forgot to tell me with what sofware you are developping: Is it Arduino, PlatformIO or with IDF directly ?

Bl4ckHol3ee, 2023-03-24:
Yeah, like you already said. Thought you knew. 

Arduino IDE. 

Would you say, that VisualStudioCode with Platform IO would be better? 
Or can i stay at Arduino IDE? 

So i looked up what you wrote before. I downloaded the Files from Esp32-2.0.7.
What should i do in the next step?

HAHA, i zipped the folder "Cores/Esp32" and added it via Arduino IDE to the libraries. 
Is this really like i should do it?? :D

pschatzmann, 2023-03-24:
Just go to the Boards Manager and make sure that you have a recent version installed.
As stated above the latest version is ESP32 Arduino 2.0.7

Bl4ckHol3ee, 2023-03-26:
So.... I'm back again.
My Code is near the target.

I've got the most things done right now.

The only thing which is not included yet, is to receive and visualize the following information:

- The duration of the current song
- The current playtime of the song
- The number of songs in the album
- The number of the current song

I don't know how to implement it. I tried it for a few hours today.
If you could give me a hint it would be pretty awesome. 

Greetings, Simon!

#TFT_eSPI.h 
#BluetoothA2DPSink.h

#include <BluetoothA2DPSink.h>
#include <SPI.h>
#include <TFT_eSPI.h>
#include "Free_Fonts.h"  // Include the header file attached to this sketch
#include "A2DPVolumeControl.h"
#include "esp_arduino_version.h"
#include "C:\Users\nbybl\Documents\Arduino\libraries\arduino\Arduino.h"
#include "BluetoothA2DPSource.h"
#include "BluetoothA2DPSink.h"
#include "BluetoothA2DP.h"
#include <BluetoothSerial.h>
#include <BluetoothA2DPCommon.h>

#define BLACK 0x0000
#define WHITE 0xFFFF

typedef struct {
  uint8_t attr_mask;   /*!< Bit mask indicating the attribute which fields are present in this struct. */
  char *title;         /*!< UTF-8 encoded title string. */
  char *artist;        /*!< UTF-8 encoded artist name string. */
  char *album;         /*!< UTF-8 encoded album name string. */
  char *genre;         /*!< UTF-8 encoded genre string. */
  uint32_t track_num;  /*!< Track number. */
  uint32_t num_tracks; /*!< Total number of tracks. */
  uint32_t duration;   /*!< Total album playing time in milliseconds. */
} esp_avrc_metadata_t;

BluetoothA2DPSink audio;
TFT_eSPI tft = TFT_eSPI();  // Invoke custom library
const char *geraetename = "Esp32s-Wroom-BT";
const char *version = "23.03.2023";

bool is_active = false;
#define P_last 19
#define P_PP 18
#define P_next 17
#define IncButton 15
#define DecButton 32

const int INCREASE_INTERVAL = 350;  // in Millisekunden
const int INCREASE_AMOUNT = 1;      // pro Intervall
volatile bool IncButton_pressed = false;
volatile bool DecButton_pressed = false;
int currentvolume;
int istVol;
int VolumeOnehundret;

bool drawed = false;
int timerclear = 4500;
unsigned long past;
[... 1312 lines cut ...]
  }

  xv23 = (x23 * xx3) + (y23 * xy3) + (z23 * xz3);
  yv23 = (x23 * yx3) + (y23 * yy3) + (z23 * yz3);
  zv23 = (x23 * zx3) + (y23 * zy3) + (z23 * zz3);

  zvt23 = zv23 - Zoff3;

  if (zvt23 < -5) {
    rx23 = 256 * (xv23 / zvt23) + Xoff3;
    ry23 = 256 * (yv23 / zvt23) + Yoff3;
  } else {
    Ok3 = 0;
  }

  if (Ok3 == 1) {

    ret3->p03.x3 = rx13;
    ret3->p03.y3 = ry13;

    ret3->p13.x3 = rx23;
    ret3->p13.y3 = ry23;
  }
  // The ifs here are checks for out of bounds. needs a bit more code here to "safe" lines that will be way out of whack, so they don't get drawn and cause screen garbage.
}

pschatzmann, 2023-03-26:
The README describes the method: set_avrc_metadata_attribute_mask() 
Just google the for ESP32 esp_avrc_md_attr_mask_t to determine the supported values

closed 2023-03-30 by pschatzmann