pschatzmann/ESP32-A2DP #392: ESP32 Dev V1 crash loop

zhanex3, 2023-03-23:
Long shot, but when I added the A2DP dependency the esp32 ends up in a crash loop.  Here are the logs i'm able to get out of it.

This disconnected line is when the device crashes and restarts in a loop.

--- Terminal on /dev/cu.usbserial-0001 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[     2][D][BluetoothA2DPSource.cpp:111] Bluetooth�xAM��ɍ��): [BT_API] BluetoothA2DPSource, 
[     7][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 =[    17][D][BluetoothA2DPSource.cpp:153] start(): [BT_API] start, 
[    18][D][BluetoothA2DPSource.cpp:171] start_raw(): [BT_API] start_raw, 
[    19][D][BluetoothA2DPCommon.cpp:149] get_last_connection(): [BT_AV] get_last_connection
[    28][I][BluetoothA2DPCommon.cpp:163] get_last_connection(): [BT_AV] nvs_blob does not exist
[    35][D][BluetoothA2DPCommon.cpp:172] get_last_connection(): [BT_AV] => 00:00:00:00:00:00
Disconnected (read failed: [Errno 6] Device not configured)
Reconnecting to /dev/cu.usbserial-0001 Error: Traceback (most recent call last):

zhanex3, 2023-03-27:
I think i may just have fault hardware. I managed to track it down to `if (!btStart()) {
            ESP_LOGE(BT_AV_TAG,"Failed to initialize controller");
            return;
        }` in AD2PSource. I couldn't get any logs from the btStart() method and i tried changing the def, but no luck, any modifications crashes it on the previous line `ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));`

pschatzmann, 2023-03-27:
I would agree that this might be a hardware problem, since you are the only one with this problem.

zhanex3, 2023-03-28:
Yup, it was a hardware problem. Got a new board today and the library is working fine. Thanks!

closed 2023-03-28 by zhanex3