pschatzmann/ESP32-A2DP #412: Interrupted audio when wifi is connected to the host computer

gingercookies, 2023-04-23:
As the title mention, i feel like my esp32 when connecting to my laptop (with 2.4ghz wifi on) cause the audio quality to go down and can be cut off completely. Change the wifi channel to channel 1 make it better but still the audio quality is bad.
Connect the esp32 to my phone not run into any kind of this issue, just the wifi speed is slow (expected). Any way to fix this?
My laptop model is HP Zbook 15 G3 and this is my code:
> #include "BluetoothA2DPSink.h"
BluetoothA2DPSink a2dp_sink;
void setup() {
    //I2S pin configs:
    i2s_pin_config_t pin_config = {
      .bck_io_num = 26,
      .ws_io_num = 25,
      .data_out_num = 27,
      .data_in_num = I2S_PIN_NO_CHANGE
    };
    //I2S configs:
    static i2s_config_t i2s_config = {
        .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
        .sample_rate = 48000, // updated automatically by A2DP
        .bits_per_sample = (i2s_bits_per_sample_t)32,
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
        .communication_format = (i2s_comm_format_t) (I2S_COMM_FORMAT_STAND_I2S),
        .intr_alloc_flags = 0, // default interrupt priority
        .dma_buf_count = 48,
        .dma_buf_len = 256,
        .use_apll = true,
        .tx_desc_auto_clear = true // avoiding noise in case of data unavailability
    };
    //Setting and start up a2dp sink
      a2dp_sink.set_i2s_config(i2s_config);
      a2dp_sink.set_pin_config(pin_config);
      a2dp_sink.set_auto_reconnect(false);
      a2dp_sink.start("MyESP32");
}
void loop() {
}

pschatzmann, 2023-04-23:
I never had any issues using my phone with Android, my old IPAD. Linux and OS/X is also woring w/o any issues.
I assume you use Windows: I haven't worked with Windows for more than 20 years, so I guess you will need to ask some Windows support

gingercookies, 2023-04-23:
Yeah, i'm using windows.
The audio quality is sometimes bad, sometimes good.
I also tested on linux (kubuntu),
the audio is better there and only sound choppy when i do some kind of heavy download/upload on the wifi. Reconnecting the esp32 will make it come back like normal.

closed 2023-05-02 by pschatzmann