pschatzmann/ESP32-A2DP #419: SPdif example used randomly crash on playback

mmar22, 2023-05-12:
ASSERT_ERR(co_list_is_empty(&ld_fm_env.fifo) || (timestamp != ((struct ld_fm_frm_elt*)ld_fm_env.fifo.last)->timestamp)), in ld_fm.c at line 936
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400905d0: 00f01d00 00004136 f01d0000
Core  0 register dump:
PC      : 0x400905d7  PS      : 0x00060134  A0      : 0x80086521  A1      : 0x3ffc2158
A2      : 0x3f4be164  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f4be15c  
A6      : 0x000003a8  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc20c8
A10     : 0x00000000  A11     : 0x3ffc20eb  A12     : 0x3ffc2097  A13     : 0x00000036  
A14     : 0x00000000  A15     : 0x3ffc209c  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x40086495  LEND    : 0x4008649d  LCOUNT  : 0x00000000  

Backtrace: 0x400905d4:0x3ffc2158 |<-CORRUPTED

  #0  0x400905d4:0x3ffc2158 in r_assert at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/controller/esp32/bt.c:1867

Pls can you help ?

Builded for ESP32 on Platformio

pschatzmann, 2023-05-12:
https://github.com/pschatzmann/ESP32-A2DP/wiki/Support-for-esp32-c3%3F

closed 2023-05-12 by pschatzmann

mmar22, 2023-05-12:
Are you sure this isnt C3!

mmar22, 2023-05-12:
Reopen !

From: Phil Schatzmann 
Sent: Friday, May 12, 2023 9:27 AM
To: pschatzmann/ESP32-A2DP 
Cc: mmar22 ; Author 
Subject: Re: [pschatzmann/ESP32-A2DP] SPdif example used randomly crash on playback (Issue #419)

https://github.com/pschatzmann/ESP32-A2DP/wiki/Support-for-esp32-c3%3F

pschatzmann, 2023-05-12:
I thought I have seen an entry for the S2 in one of your log lines.  
Anyhow did you try to reproduce your issue in Arduino ? Maybe this is a PlatformIO issue ?

mmar22, 2023-05-12:
Ok more info:
[env:esp32dev]
platform = espressif32
;board = esp-wrover-ca100
board = esp32dev
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder, default, log2file
board_build.partitions = default_8MB.csv
extra_scripts = 
    pre:extra_script.py
lib_deps = 
    ***@***.***
    moononournation/GFX Library for Arduino@^1.3.1
    crankyoldgit/IRremoteESP8266@^2.8.4

build log 

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)    
PACKAGES:
- framework-arduinoespressif32 @ 3.20007.0 (2.0.7)
- tool-esptoolpy @ 1.40500.0 (4.5.0)
- toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- lvgl @ 8.3.4
|-- GFX Library for Arduino @ 1.3.4
|-- IRremoteESP8266 @ 2.8.4
|-- Wire @ 2.0.0
|-- audio-tools @ 0.9.5
|-- ESP32-A2DP @ 1.7.1
Building in release mode
...
Linking .pio\build\esp32dev\firmware.elf
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  30.4% (used 99712 bytes from 327680 bytes)
Flash: [======    ]  61.7% (used 2061681 bytes from 3342336 bytes)
Building .pio\build\esp32dev\firmware.bin
esptool.py v4.5
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
================================= [SUCCESS] Took 133.43 seconds

No error, app run , phone pair and connect without issues. Plays music random time and crash info on first post.

pschatzmann, 2023-05-12:
As this is not an issue in my code I would suggest to try different "framework-arduinoespressif32" versions and as already suggested check in Arduino to exclude that the issue comes from PlatformIO. 

The latest version is 2.0.9

mmar22, 2023-05-13:
Copied sources and build in Arduino with latest 2.0.9 result to 

ASSERT_ERR(co_list_is_empty(&ld_fm_env.fifo) || (timestamp != ((struct ld_fm_frm_elt*)ld_fm_env.fifo.last)->timestamp)), in ld_fm.c at line 936
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40090634: 00f01d00 00004136 f01d0000
Core  0 register dump:
PC      : 0x4009063b  PS      : 0x00060034  A0      : 0x80086585  A1      : 0x3ffc2160
A2      : 0x3f4c1ea0  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3f4c1e98  
A6      : 0x000003a8  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc20d0
A10     : 0x00000000  A11     : 0x3ffc20f3  A12     : 0x3ffc209f  A13     : 0x00000036  
A14     : 0x00000000  A15     : 0x3ffc20a4  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400864f9  LEND    : 0x40086501  LCOUNT  : 0x00000000  

Backtrace: 0x40090638:0x3ffc2160 0x40086582:0x3ffc2180 0x4008837f:0x3ffc21a0 0x4008808d:0x3ffc21d0 0x40088595:0x3ffc21f0 0x4008c7fa:0x3ffc2210 0x4008d19f:0x3ffc2230 0x40086016:0x3ffc2250 0x40086335:0x3ffc2270 0x40086317:0x00000000 |<-CORRUPTED

pschatzmann, 2023-05-13:
I was testing [my example sketch](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-basic-api/basic-a2dp-spdif) and played it for serveral minutes w/o issues!

Are you sure that you used the same buffer settings (from the [AudioConfigLocal.h](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/basic-a2dp-spdif/AudioConfigLocal.h)) and the delay in the loop ?

mmar22, 2023-05-13:
No buffer isnt same not available memory , i write it in first message.
Can be buffers in PSRAM ? If yes how to setup.

pschatzmann, 2023-05-13:
I doubt: the issue seems to be that i2s is not returning quick enough, so slow PSRAM will hurt more than it helps.
I would suggest that you keep the buffer size at 384 and that you try if you find a buffer count that works for you...

see https://github.com/espressif/esp-idf/issues/1996

mmar22, 2023-05-13:
Thanks today last question. Isnt source not in iram callback ?
`// Write data to SPDIF in callback
void read_data_stream(const uint8_t *data, uint32_t length) {
    spdif.write(data, length);
}
`

pschatzmann, 2023-05-13:
Nop, that's not relevant! You can double check with the official IDF A2DP example: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_sink

mmar22, 2023-05-15:
I enable PSRAM and all works like charm, no code changed..??. Buff set to 8/768. (without PSRAM crash) But still I mean your code example isnt clean. In compare with reference link you send, your code omit handle buf levels, use threads and semaphore
latest idf
#define RINGBUF_HIGHEST_WATER_LEVEL    (32 * 1024)
#define RINGBUF_PREFETCH_WATER_LEVEL   (20 * 1024)
....

From: Phil Schatzmann 
Sent: Saturday, May 13, 2023 8:41 PM
To: pschatzmann/ESP32-A2DP 
Cc: mmar22 ; Author 
Subject: Re: [pschatzmann/ESP32-A2DP] SPdif example used randomly crash on playback (Issue #419)

I doubt: the issue seems to be that i2s is not returning quick enough, so slow PSRAM will hurt more than it helps.
I would suggest that you keep the buffer size at 384 and that you try if you find a buffer count that works for you...

see espressif/esp-idf#1996

pschatzmann, 2023-05-15:
In the initial version the demo code did not contain this logic. It was added later to address the issue with handling volume changes in IOS. I fist included it as default as well, but it was adding too mauch memory overhead overhead for most users.

So in the end I left it to the choice of the end user to select the proper Sink subclass implementation ! 

Feel free to submit your improvements if you think this can be made better!