pschatzmann/ESP32-A2DP #422: A2DP- source mode

tueddy, 2023-05-21:
H,

A2DP-source works fine [here](https://github.com/biologist79/ESPuino/blob/3bd57edbe195fd2d5ba7ef8d44af1c6f7cb80c0f/src/Bluetooth.cpp#L165), but with Ardiuno 2.0.9 (PlatformIO package 6.3.0) i get this crash:

`[ 21097 ]  Bluetooth source => connection state: Connecting
[ 26220 ]  Bluetooth source => connection state: Disconnected

Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: **Stack canary watchpoint triggered (BtAppT)**
Core  1 register dump:
PC      : 0x40087310  PS      : 0x00060d36  A0      : 0x3fff1ad0  A1      : 0x3fff1a10
A2      : 0x3fff1eb0  A3      : 0x00000018  A4      : 0x00000001  A5      : 0x0000000c
A6      : 0x00000000  A7      : 0x00000008  A8      : 0x00000001  A9      : 0x00000024  
A10     : 0x0000005a  A11     : 0x0000002a  A12     : 0x00000075  A13     : 0x3f466ba0  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x4009449d  LEND    : 0x400944ad  LCOUNT  : 0xffffffff

Backtrace: 0x4008730d:0x3fff1a10 0x3fff1acd:0x3fff1de0 |<-CORRUPTED

  #0  0x4008730d:0x3fff1a10 in _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/xtensa_vectors.S:1078
  #1  0x3fff1acd:0x3fff1de0 in ?? ??:0
`

Any idea how to solve this? For A2DPSink i can set a larger tasksize with `set_event_stack_size();` but for A2DPSource it's not possible?
Should the task size increased for latest Arduino version?

Thanls & best
Dirk

pschatzmann, 2023-05-21:
I increased the stack size and made the set_event_stack_size() available to the source as well

tueddy, 2023-05-22:
pschatzman, that was lightning fast!
I can confirm that A2DP-source do not longer crashes.

Thank's & best regards
Dirk

closed 2023-05-22 by tueddy