pschatzmann/ESP32-A2DP #428: AVRC without a2dp?

Guillemete, 2023-06-07:
is it possible to use your library as a stand alone AVRC controller?

Without hijacking the audio feed from the player?

pschatzmann, 2023-06-07:
What's wrong with you people: Why are you asking questions that have been answered  several times and that are documented in the Readme!: stand alone AVRCP without A2DP are not supported!

closed 2023-06-07 by pschatzmann

pschatzmann, 2023-06-07:
I think the most promising approch would be if you use the ESP32 API directly: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_avrc.html

Guillemete, 2023-06-07:
I'll look into it! thanks :)

kk-kk-k, 2025-06-01:
Hello, it is technically possible. This code should pause and play music:

```
#include "freertos/FreeRTOS.h"
#include "esp_err.h"
#include "esp_log.h"
#include "nvs.h"
#include "nvs_flash.h"

#include "esp_bt.h"
#include "esp_bt_main.h"
#include "esp_avrc_api.h"
#include "esp_gap_bt_api.h"
#include "esp_bt_device.h"
#include "esp_a2dp_api.h"

static void vMainNvsInit(void);
static void vMainBluedroidInit(void);

SemaphoreHandle_t xSemaphore = NULL;

void app_main(void)
{
    xSemaphore = xSemaphoreCreateBinary();

    // initialize nvs - needed for bluetooth
    vMainNvsInit();

    // initialize bluedroid
    vMainBluedroidInit();

    if (xSemaphoreTake(xSemaphore, portMAX_DELAY))
        while (1)
        {
            esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PAUSE, ESP_AVRC_PT_CMD_STATE_PRESSED);
            vTaskDelay(100 / portTICK_PERIOD_MS);
            esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PAUSE, ESP_AVRC_PT_CMD_STATE_RELEASED);
            vTaskDelay(3000 / portTICK_PERIOD_MS);

            esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PLAY, ESP_AVRC_PT_CMD_STATE_PRESSED);
            vTaskDelay(100 / portTICK_PERIOD_MS);
            esp_avrc_ct_send_passthrough_cmd(0, ESP_AVRC_PT_CMD_PLAY, ESP_AVRC_PT_CMD_STATE_RELEASED);
            vTaskDelay(1000 / portTICK_PERIOD_MS);
        }
}

static void vMainNvsInit(void)
{
    // initialize nvs
    esp_err_t err = nvs_flash_init();

    // check for error
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK(err);
}

static void bt_gap_callback(esp_bt_gap_cb_event_t event, esp_bt_gap_cb_param_t *param);
static void bt_avrc_callback(esp_avrc_ct_cb_event_t event, esp_avrc_ct_cb_param_t * params);

static void vMainBluedroidInit(void)
{
    // release memory for bluetooth low energy
    esp_bt_controller_mem_release(ESP_BT_MODE_BLE);

    // initialize peripherial
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_bt_controller_init(&bt_cfg));
    ESP_ERROR_CHECK(esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT));

    // initialize bluedroid
    esp_bluedroid_config_t bluedroid_cfg = BT_BLUEDROID_INIT_CONFIG_DEFAULT();
#if 0
    bluedroid_cfg.ssp_en = false;
#endif
    ESP_ERROR_CHECK(esp_bluedroid_init_with_cfg(&bluedroid_cfg));
    ESP_ERROR_CHECK(esp_bluedroid_enable());

    // setup seurity features
    esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
    esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;
    esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));

    // setup BT GAP pin
    esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_FIXED;
    esp_bt_pin_code_t pin_code;
    pin_code[0] = '1';
    pin_code[1] = '2';
    pin_code[2] = '3';
    pin_code[3] = '4';
    esp_bt_gap_set_pin(pin_type, 4, pin_code);

    // finish GAP setup
    esp_bt_gap_set_device_name("JetJack");
    esp_bt_gap_register_callback(bt_gap_callback);

    // init a2dp
    esp_avrc_ct_init();
    esp_avrc_ct_register_callback(bt_avrc_callback);
    esp_a2d_sink_init();

    // setup device connectable
    esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
}

#define TAG "bt_gap_callback"
static void bt_gap_callback(esp_bt_gap_cb_event_t event, esp_bt_gap_cb_param_t *param)
{
    switch (event)
    {
        case ESP_BT_GAP_PIN_REQ_EVT:
        {
            esp_bt_pin_code_t pin_code;
            pin_code[0] = '1';
            pin_code[1] = '2';
            pin_code[2] = '3';
            pin_code[3] = '4';
            esp_bt_gap_pin_reply(param->pin_req.bda, true, 4, pin_code);
            break;
        }
        case ESP_BT_GAP_CFM_REQ_EVT:
            ESP_LOGI(TAG, "Accepting ssp");
            esp_bt_gap_ssp_confirm_reply(param->cfm_req.bda, true);
            break;
        default:
            break;
    }
}

#undef TAG
#define TAG "bt_avrc_callback"
static void bt_avrc_callback(esp_avrc_ct_cb_event_t event, esp_avrc_ct_cb_param_t * params)
{
    switch (event)
    {
        case ESP_AVRC_CT_CONNECTION_STATE_EVT:
            if (params->conn_stat.connected)
            {
                ESP_LOGI(TAG, "Connected");
                xSemaphoreGive(xSemaphore);
            }
            else
                ESP_LOGI(TAG, "Disconnected");
            break;
        default:
            ESP_LOGW(TAG, "Unhandled event: %d", event);
            break;
    }
}
```

Please note that the a2dp is NOT USED.

Please do not forget to configure esp-idf menuconfig correctly:

- enable bluetooth (menuconfig->component config->Bluetooth->Bluetooth; see sdkconfig BT_ENABLED)
- setup host to bluedroid dual mode (menuconfig->component config->Bluetooth->Host->Bluedroid; see sdkconfig BT_BLUEDROID_ENABLED in choice BT_HOST)
- make sure controller is enabled (menuconfig->component config->Bluetooth->Controller->Enabled; see sdkconfig BT_CONTROLLER_ENABLED in choice BT_CONTROLLER)
- enable a2dp (menuconfig->component config->Bluetooth->Bluedroid options->A2DP; see sdkconfig BT_A2DP_ENABLE)