pschatzmann/ESP32-A2DP #430: Feature request - SPDIF extend with ringbuf and nonstop play

mmar22, 2023-06-22:
Hi i use your A2DP - SPDIF and every data break breaks too spdif stream and decoders have required relocking usw.
Perfect will bee when your class have variant with ringbuff autofilled with silence if no data.
?

pschatzmann, 2023-06-22:
I do not consider this as a valid bug/issue: After all you should be able to implement this in your sketch with just a few lines of code!

closed 2023-06-22 by pschatzmann

mmar22, 2023-06-23:
Maybe for you , but i cant handle few lines... Try show me how replace 
`        data = (uint8_t *)xRingbufferReceive(s_ringbuf_i2s, &item_size, (portTickType)0); //portMAX_DELAY`
with silence if data not arrive and back . Now i use this , that else NULL send silence ok, but crash when a2dp audio arrive

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x401262a4  PS      : 0x00060c30  A0      : 0x8012611c  A1      : 0x3fff5340  
A2      : 0x00000000  A3      : 0x3ffd289c  A4      : 0x3ffd2898  A5      : 0x3f4f217c
A6      : 0x3ffd2e9c  A7      : 0x00000004  A8      : 0x80095272  A9      : 0x3fff52e0  
A10     : 0x00000000  A11     : 0x3fff5348  A12     : 0x00000000  A13     : 0x3fff3e58
A14     : 0x3fff3e04  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40092444  LEND    : 0x4009245a  LCOUNT  : 0x00000000

Backtrace: 0x401262a1:0x3fff5340 0x40126119:0x3fff5370

  #0  0x401262a1:0x3fff5340 in spdif_write at lib/spdif/spdif.c:143
  #1  0x40126119:0x3fff5370 in bt_i2s_task_handler at lib/spdif/bt_app_core.c:158

pschatzmann, 2023-06-23:
I would try something as follows:
- Just start with [this example](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/basic-a2dp-spdif/basic-a2dp-spdif.ino)
- In the loop check the audio_state and if stopped call writeSilence() on the spdif object

mmar22, 2023-06-23:
Isnt so simple. 1. my code is based from example.
2.writeSIlence dont exist and ofcourse i can create one , but this need be called precise in samplerate speed usw.
SinkQueued work for I2S not for SPDIF .... agghhh this is your lib not my

pschatzmann, 2023-06-23:
Just check the documentation: writeSilence does exist: https://pschatzmann.github.io/arduino-audio-tools/classaudio__tools_1_1_s_p_d_i_f_output.html
I don't think you need to bother with the sample rate....

mmar22, 2023-06-23:
I dont see here https://github.com/pschatzmann/arduino-audio-tools/blob/main/src/AudioTools/AudioSPDIF.h

And my issue isnt audible , but spdif stop act and unlock

pschatzmann, 2023-06-23:
I think you overlooked that this is a subclass of AudioStream!

pschatzmann, 2023-06-23:
I don't quite understand your issue with spdif stop act and unlock: could you please explain what you mean exactly

mmar22, 2023-06-23:
My app require after start continuos spdif . Now every BT issue breaks it . If you connect this into for example AVReceiver your display about connected media blink as Manhatan

pschatzmann, 2023-06-23:
Did you try my suggested proposal using writeSilence()? 
I dont see the need for making things more complicated...

mmar22, 2023-06-23:
Now i cant, as you write i use few lines and write own ringbuf , then your tools lib is removed and only a2dp used.
I try this 
`static void bt_i2s_task_handler(void *arg)
{
    uint8_t *data = NULL;

    static uint8_t silence[4] = {0,0,0,0};

    size_t item_size = 0;
#ifndef CONFIG_EXAMPLE_A2DP_SINK_OUTPUT_SPDIF
    size_t bytes_written = 0;
#endif

    for (;;) {
        data = (uint8_t *)xRingbufferReceive(s_ringbuf_i2s, &item_size, (portTickType)0); //portMAX_DELAY 
        if (item_size != 0){
#ifdef CONFIG_EXAMPLE_A2DP_SINK_OUTPUT_INTERNAL_DAC
	    uint16_t *dac = (uint16_t *)data;

	    for (int i = 0; i < item_size / sizeof(uint16_t); i++) {
		dac[i] += 32768; // convert to unsigned 16bit audio data
	    }
#endif
#ifdef CONFIG_EXAMPLE_A2DP_SINK_OUTPUT_SPDIF
	    spdif_write(data, item_size);
#else
            i2s_write(0, data, item_size, &bytes_written, portMAX_DELAY);
#endif
            vRingbufferReturnItem(s_ringbuf_i2s,(void *)data);
        }
        else // timeout no sample place silence loop but how exit
        {
            spdif_write(silence, 4);
        }
    }
}`
silence works perfect , but when data arrive in ring app crash as on first today message

pschatzmann, 2023-06-23:
Well, that's not what I recommended you to do!

- Just start with [this example](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/basic-a2dp-spdif/basic-a2dp-spdif.ino)
- In the loop check the audio_state and if stopped just call the writeSilence() method on the spdif object

mmar22, 2023-06-23:
Ahh i have app with tausend lines and more threads , your test is simple and maybe works but unusable in real life

mmar22, 2023-06-23:
FYI solved problem was byggy rtos need change to this
`        data = (uint8_t *)xRingbufferReceive(s_ringbuf_i2s, &item_size, (portTickType)0); //portMAX_DELAY otazka co robi 0
        if (data != NULL && item_size != 0){
`

mmar22, 2023-06-24:
I have one clearing question is your callback based as ISR?
`// Write data to SPDIF in callback
void read_data_stream(const uint8_t *data, uint32_t length) `

pschatzmann, 2023-06-25:
No: check the IDF documentation for esp_a2d_sink_register_data_callback

mmar22, 2023-07-27:
I handle now silence and spdif plays ok. I add switching 44/48 based on a2dp , but now i check change pooling in loop. Exist some callback , that inform a2dpsink change samplerate? Thx for help