pschatzmann/ESP32-A2DP #433: *help needed* Change default I2S pins

Sausebraus, 2023-07-04:
Hi,

I want to use an external DAC for my audio sink and there are these pins in the wiki: 

    bck_io_num = 26
    ws_io_num = 25
    data_out_num = 22

but in my current design these GPIO pins are already in use, how can i change them to my own GPIO pins? Is this even possible?

thank you in advance for helping me

pschatzmann, 2023-07-04:
No, that's predefined by the hardware and you can't change the internal DAC pins, so the only thing that you can do is to reassign the other functionality which has more flexibility.

closed 2023-07-04 by pschatzmann

mrhempman69, 2023-09-30:
Just to confirm, the internal DAC pins must be used when connecting to an external DAC?

Reading this document, it isn't entirely obvious to me why that is the case? https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#:~:text=It%20needs%20at%20least%20a%20CLK%20pin%20for,clock%2C%20which%20is%20not%20needed%20between%20PDM%20devices%29.

It's not like the external DAC relies on an analog voltage...? 

Yet it's true that when I try to set BCK and WS off the ESP32's internal DAC pins, there is no sound.

Are these pins (25 26) special in some other way besides the DAC? Or does I2S simply require an analog signal?

pschatzmann, 2023-09-30:
Please don't post your nonsense to closed issues!