pschatzmann/ESP32-A2DP #445: Analog Input (Head Phone Jack) to Bluetooth

Metaln00b, 2023-08-11:
Hi,

Is it possible to stream/transmit an analogue audio signal directly to a bluetooth speaker?

Example: Using analog.read(), writing to a 2 channel array, a2dp_source.write_data(data);

My plan was to use an old walkman and stream the audio with an ESP32 to a bluetooth only speaker.

Thanks

joba-1, 2023-08-11:
no, not directly.

a2dp is a digital protocol, so you always need an adc (analog to digital converter) and the adc pin of an esp is too slow for the task.
You need one that can sample 2 channels simultaneously 44000 times per second into 16 bits or more, ideally output as i2s stream.

A pcm1808 board connected to an ESP might fit your bill.

closed 2023-08-11 by Metaln00b

IRA1777, 2024-01-19:
I've got an external ADC, and the question remains. How can I use it ? The "source" examples are not covering this.