pschatzmann/ESP32-A2DP #450: when there is no sound, dac outputs 1.7V

tesla-cat, 2023-08-14:
when there is no sound from the bluetooth, the dac outputs 1.7V, how can i make it output 0V ?

pschatzmann, 2023-08-14:
This is neither a bug, nor I am able to make any sense out of your question. It does not make any sense to ask me to guess what exactly your use case / scenario is that you try to implement.

closed 2023-08-14 by pschatzmann

tesla-cat, 2023-08-15:
**I mean, when i upload the code below to ESP32 board, the DAC output (pin 25) is around 1.7 Volts, instead of 0 Volt**

> You can also send the output directly to the internal DAC of the ESP32 by providing the corresponding i2s_config:

```cpp
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void setup() {
    static const i2s_config_t i2s_config = {
        .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),
        .sample_rate = 44100, // corrected by info from bluetooth
        .bits_per_sample = (i2s_bits_per_sample_t) 16, /* the DAC module will only take the 8bits from MSB */
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
        .communication_format = (i2s_comm_format_t)I2S_COMM_FORMAT_STAND_MSB,
        .intr_alloc_flags = 0, // default interrupt priority
        .dma_buf_count = 8,
        .dma_buf_len = 64,
        .use_apll = false
    };

    a2dp_sink.set_i2s_config(i2s_config);
    a2dp_sink.start("MyMusic");

}

void loop() {
}
```

pschatzmann, 2023-08-15:
Using an analog output the signal is oscillating between 0 and 3.1V. 
So your measured voltage is about the neutral (0) position. 

So I would expect that you will need to switch of the I2S interface to bring it down to 0
By default this is managed by the connection status. But there are many possibilities to enhance the functionality to use the audio status instead.