pschatzmann/ESP32-A2DP #462: ESP32-A2DP

silelis, 2023-08-25:
Is there any chance to adjusty component to ESP-IDF 5.x??
In this API i2s.h is in depreciated diary so there is compilation error:
 1) 
In file included from ../../../main/components/ESP32-A2DP-main/src/BluetoothA2DPCommon.cpp:16:
../../../main/components/ESP32-A2DP-main/src/BluetoothA2DPCommon.h:47:10: fatal error: driver/i2s.h: No such file or directory
   47 | #include "driver/i2s.h"
      |          ^~~~~~~~~~~~~~

2) In file included from ../../../main/exception_example_main.cpp:12:
../../../main/components/ESP32-A2DP-main/src/BluetoothA2DPSink.h:288:5: error: 'xQueueHandle' does not name a type; did you mean 'QueueHandle_t'?
  288 |     xQueueHandle app_task_queue = nullptr;
      |     ^~~~~~~~~~~~

pschatzmann, 2023-08-26:
Feel free to submit a pull request that makes sure that both idf 4 and idf 5 are supported.

e.g. adding the following in config.h

#  if ESP_ARDUINO_VERSION_MAJOR >= 5
using xQueueHandle = QueueHandle_t;
#  endif

might resolve one of your problems. 

The old I2S api should still be supported...

pschatzmann, 2023-08-26:
The latest commit allows you to deactivate the I2S API: So you can use e.g. the AudioTools as output library.

closed 2023-08-26 by pschatzmann

silelis, 2023-08-26:
How to do this??

pschatzmann, 2023-08-26:
With the help of compile_definitions: 
- A2DP: A2DP_I2S_SUPPORT=0
- AudioTools: USE_I2S_NEW

silelis, 2023-08-26:
Sorry that bother You  but... I do not get it.

I have copied new release.
I see the difference but under 5.1 API when  I build it as it is I still have error:
../../../main/components/ESP32-A2DP-main/src/BluetoothA2DPCommon.h:55:12: fatal error: driver/i2s.h: No such file or directory
   55 | #  include "driver/i2s.h"
      |            ^~~~~~~~~~~~~~

to use new API approached i should place: "#define USE_I2S_NEW" in config.h or ....?

pschatzmann, 2023-08-26:
I guess you need to learn how to use compile_definitions  in cmake first.

silelis, 2023-08-26:
Probably. Cmake and ESP is new to me... ESP adventure is about 3 weeks. Earlier I played with Microchip PICs and CCS compiler and STM32.

Every think worked in 4.4 API but in 5.x stopped.

I use Your component in non Arduino IDE (environment) but in Visual GDB in Visual Studio.

I still in the same error.

pschatzmann, 2023-08-26:
You will need to find out yourself if Visual GDB supports Preprocessor definitions for compiling a target's sources. If not, I told you how to proceed.

silelis, 2023-08-26:
Is it ok If I just ask You to write where I should paste what phrase to:
- support i2s legacy drivers
- new approached?

I understand that should learn a lot but it will help me to understand.

pschatzmann, 2023-08-26:
I suggest you stick with idf4 until you have the proper know how

silelis, 2023-08-26:
That will be taught because I have to rewrite all my code. but thanks for Your time

pschatzmann, 2023-08-26:
Why would you need to do that ? After all the difference is not that big

silelis, 2023-08-26:
I use some peripherals that are different in 5.x and 4.4.

As I wrote I am new in ESP and hole ESP IDF. It quite different approached that in STM32 Cube IDE.

Those are my 1st steps.

pschatzmann, 2023-08-26:
So if you still want to use this library you would need to correct the compile errors or do the i2s output yourself after changing #  define A2DP_I2S_SUPPORT true to false in the config.h

silelis, 2023-08-26:
Rewrite all code in:
#if A2DP_I2S_SUPPORT

#endif

it seems to be easier than write all BT

pschatzmann, 2023-08-26:
Here is an example using the AudioTools: https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/basic-a2dp-i2s/basic-a2dp-i2s.ino

pschatzmann, 2023-08-26:
The reason for the i2s compile error is that 5.x requires driver in the idf_component_register

silelis, 2023-08-26:
So if the i2s.h is in:
C:\Espressif\frameworks\esp-idf-v5.0.3\components\driver\deprecated\driver
what should I add and where?
As You noticed I have no knowladge about cmake. I use that what receive from creators.

silelis, 2023-08-26:
> Here is an example using the AudioTools: https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-basic-api/basic-a2dp-i2s/basic-a2dp-i2s.ino
> 
> Of cause you can do the same directly with the ESP32 API that I sent you above.
> 
> However you should not get any compile errors. So I suggest that you try to find out first why this is. Maybe it is just a missing entry in the sdkconfig.defaults
> 
> Did you use the new I2S API anywere in your code already ?

No I have not use it. I test ESP-A2DP component as stand alone app. And the error is always when I add this component and activate bluetooth in menuconfig.
Now the error looks like bellow:

In file included from D:/!!!__GitHUB_repositories/MustangRadio_develepementVersion/02_Firmware/01_ESP32/VisualGDB_BT_cpp/main/components/ESP32-A2DP-main/src/BluetoothA2DPSource.h:22,
                 from D:/!!!__GitHUB_repositories/MustangRadio_develepementVersion/02_Firmware/01_ESP32/VisualGDB_BT_cpp/main/components/ESP32-A2DP-main/src/BluetoothA2DPSource.cpp:16:
D:/!!!__GitHUB_repositories/MustangRadio_develepementVersion/02_Firmware/01_ESP32/VisualGDB_BT_cpp/main/components/ESP32-A2DP-main/src/BluetoothA2DPCommon.h:47:10: fatal error: driver/i2s.h: No such file or directory
   47 | #include "driver/i2s.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
[0/1] Re-running CMake...-- Building ESP-IDF components for target esp32
-- Project sdkconfig file D:/!!!__GitHUB_repositories/MustangRadio_develepementVersion/02_Firmware/01_ESP32/VisualGDB_BT_cpp/sdkconfig
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.3/components/soc/esp32/ld/esp32.peripherals.ld
-- App "bootloader" version: v5.0.3-dirty
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.3/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.3/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.3/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.3/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.3/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.3/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf-v5.0.3/components/bootloader C:/Espressif/frameworks/esp-idf-v5.0.3/components/bootloader_support C:/Espressif/frameworks/esp-idf-v5.0.3/components/efuse C:/Espressif/frameworks/esp-idf-v5.0.3/components/esp_app_format C:/Espressif/frameworks/esp-idf-v5.0.3/components/esp_common C:/Espressif/frameworks/esp-idf-v5.0.3/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v5.0.3/components/esp_rom C:/Espressif/frameworks/esp-idf-v5.0.3/components/esp_system C:/Espressif/frameworks/esp-idf-v5.0.3/components/esptool_py C:/Espressif/frameworks/esp-idf-v5.0.3/components/freertos C:/Espressif/frameworks/esp-idf-v5.0.3/components/hal C:/Espressif/frameworks/esp-idf-v5.0.3/components/log C:/Espressif/frameworks/esp-idf-v5.0.3/components/bootloader/subproject/main C:/Espressif/frameworks/esp-idf-v5.0.3/components/bootloader/subproject/components/micro-ecc C:/Espressif/frameworks/esp-idf-v5.0.3/components/newlib C:/Espressif/frameworks/esp-idf-v5.0.3/components/partition_table C:/Espressif/frameworks/esp-idf-v5.0.3/components/soc C:/Espressif/frameworks/esp-idf-v5.0.3/components/spi_flash C:/Espressif/frameworks/esp-idf-v5.0.3/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: D:/!!!__GitHUB_repositories/MustangRadio_develepementVersion/02_Firmware/01_ESP32/VisualGDB_BT_cpp/build/bootloader
[1/1] cmd.exe /C "cd /D D:\!!!__GitHUB_repositories\Mustan...01_ESP32/VisualGDB_BT_cpp/build/bootloader/bootloader.bin"Bootloader binary size 0x6740 bytes. 0x8c0 bytes (8%) free.
ninja: build stopped: subcommand failed.
HINT: Please make sure that the header name is correct. Also please check if you've specified all component dependencies with 'idf_component_register(REQUIRES ...)'. If the component is not present then it should be added by the IDF Component Manager. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the header file has been removed, renamed or relocated - refer to the migration guide for more information.
ninja failed with exit code 1, output of the command is in the D:\!!!__GitHUB_repositories\MustangRadio_develepementVersion\02_Firmware\01_ESP32\VisualGDB_BT_cpp\build\log\idf_py_stderr_output_23276 and D:\!!!__GitHUB_repositories\MustangRadio_develepementVersion\02_Firmware\01_ESP32\VisualGDB_BT_cpp\build\log\idf_py_stdout_output_23276
PS D:\!!!__GitHUB_repositories\MustangRadio_develepementVersion\02_Firmware\01_ESP32\VisualGDB_BT_cpp>

Even HINT says that something is REQUIRED in 'idf_component_register(REQUIRES ...)' but I do not know how to handle this. :/

pschatzmann, 2023-08-27:
just look at my latest commits!

silelis, 2023-08-27:
Yes I know.
Based on information that I feed the issue, are You able to write fix?

pschatzmann, 2023-08-27:
I told you what the issue is, how to correct it and that the correction is in the latest commits.
Did you look at them ?

silelis, 2023-08-27:
Yes there had been other issues. Mostly because some files in 5.x do not contain data from 4.4.x
I have fixed theme and it is compileing but .... I have not checked yet if it is working on device (no fatal register errors).

At least tommorow I will upload it to ESP32 and push repo to Your git.
I am not proud of some fixex (i.e.:

#if __has_include("esp_idf_version.h")	//Silelis fix
#include "esp_idf_version.h"	//Silelis fix
#  if ESP_IDF_VERSION_MAJOR >= 5	//Silelis fix
#ifndef ESP_IDF_5	//Silelis fix
#define ESP_IDF_5	//Silelis fix
#endif	//Silelis fix
#define I2S_MCLK_MULTIPLE_DEFAULT	0		//Silelis fix - by default API 5.x do not have this define
#define portTickType				TickType_t        //Silelis fix
#define eTaskStateGet                 eTaskGetState
#define portTickType                  TickType_t
#define xTaskHandle                   TaskHandle_t
#define xQueueHandle                  QueueHandle_t
#define xSemaphoreHandle              SemaphoreHandle_t
#define xQueueSetHandle               QueueSetHandle_t
#define xQueueSetMemberHandle         QueueSetMemberHandle_t
#define xTimeOutType                  TimeOut_t
#define xMemoryRegion                 MemoryRegion_t
#define xTaskParameters               TaskParameters_t
#define xTaskStatusType               TaskStatus_t
#define xTimerHandle                  TimerHandle_t
#define xCoRoutineHandle              CoRoutineHandle_t
#define pdTASK_HOOK_CODE              TaskHookFunction_t
#define portTICK_RATE_MS              portTICK_PERIOD_MS
#define pcTaskGetTaskName             pcTaskGetName
#define pcTimerGetTimerName           pcTimerGetName
#define pcQueueGetQueueName           pcQueueGetName
#define vTaskGetTaskInfo              vTaskGetInfo
#define xTaskGetIdleRunTimeCounter    ulTaskGetIdleRunTimeCounter
#  endif
#endif

but all of theme are marked by:  //Silelis fix so after push You will able to locate them easylly and make in more elegant way.

pschatzmann, 2023-08-27:
I suggest that you double check what you are doing. Most of these types are not even used in my code, so I guess you are doing something completely wrong. The only ones that I found was the ones that I was already taking care of.

silelis, 2023-08-27:
I know. That's why I said that 1st I have to check it on chip

pschatzmann, 2023-08-27:
The other thing I don't understand is why you would need to do another #include "esp_idf_version.h" //Silelis fix