pschatzmann/ESP32-A2DP #463: Watchdog issue

tutoduino, 2023-08-26:
Hi,

I face a watchdog issue in the example **bt_music_sender** with an ESP32-WROOM streaming to JBL GO 2 Bluetooth speaker.
I can solve it adding a 10ms delay in the callback before returning, but I am not sure it is a good idea...
I see another error, which I do not think is related to: Invalid AVRC event: 6
Here are the debug traces :

14:18:36.975 -> [ 10971][D][BluetoothA2DPSource.cpp:735] bt_app_av_state_connected(): [BT_AV] bt_app_av_state_connected evt 3
14:18:36.975 -> [ 10980][D][BluetoothA2DPSource.cpp:795] bt_app_av_media_proc(): [BT_AV] bt_app_av_media_pro109vt][D
14:18:36.975 -> [[ l09to][I]hA2uetoSthrcD.cpurce.cpp b1t] pp_appcav_m_dia)proc(CT [Bb_Aa] a2dpcmcdtacbt rtvtuccess[ l1y.
98] 1099B][D]eBoottoAthA2DPSDPrcurcpp:301].bc_p:p_t0]k_handaerp)work_d_API]att_h()_taskBT_APIlert_sig _wo, 0dispa[ h e08nt ]x0,uetramhl2n Source. p10132]D][_aue_avthA_hdPSou: [.cpAV8 b] bpp_pv_sr_hdt_cst):e[R, e] b0x_app_ 11_c7t[_]cBleetoo5
A2DP[our01.][p:[00l ptocethAuseP_Statce.allb25k] b: [pp_wor prispsschse: [tatAPc] lbaaps_
14:18:37.042 -> ork110spa[c] Blunt othA 2arSo lee.c2
:619] process_user_state_callbacks(): [BT_AV] process_user_state_callbacks ESP_A2D_AUDIO_STATE_EVT: Started
14:18:37.042 -> [ 11040][D][BluetoothA2DPSource.cpp:735] bt_app_av_state_connected(): [BT_AV] bt_app_av_state_connected evt 1
14:18:37.042 -> [ 11049][D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): [BT_API] bt_app_task_handler, sig 0x1, 0x0
14:18:37.042 -> [ 11059][D][BluetoothA2DPSource.cpp:899] bt_av_hdl_avrc_ct_evt(): [RCCT] bt_av_hdl_avrc_ct_evt evt 0
14:18:37.075 -> [ 11068][I][BluetoothA2DPSource.cpp:905] bt_av_hdl_avrc_ct_evt(): [RCCT] AVRC conn_state evt: state 1, [70:99:1c:ad:a2:1e]
14:18:37.075 -> [ 11078][D][BluetoothA2DPSource.cpp:301] bt_app_task_handler(): [BT_API] bt_app_task_handler, sig 0x1, 0x5
14:18:37.109 -> [ 11088][D][BluetoothA2DPSource.cpp:899] bt_av_hdl_avrc_ct_evt(): [RCCT] bt_av_hdl_avrc_ct_evt evt 5
14:18:37.109 -> [ 11097][I][BluetoothA2DPSource.cpp:936] bt_av_hdl_avrc_ct_evt(): [RCCT] AVRC remote feature 11104][D][BluetoothA2DPSource.cpp:856] bt_app_rc_ct_cb(): [RCCT] bt_app_rc_ct_cb evt 6
14:18:37.109 -> [ 11112][E][BluetoothA2DPSource.cpp:867] bt_app_rc_ct_cb(): [RCCT] Invalid AVRC event: 6
14:18:42.220 -> E (32627) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
14:18:42.220 -> E (32627) task_wdt:  - IDLE (CPU 0)
14:18:42.220 -> E (32627) task_wdt: Tasks currently running:
14:18:42.220 -> E (32627) task_wdt: CPU 0: BTC_TASK
14:18:42.220 -> E (32627) task_wdt: CPU 1: IDLE
14:18:42.220 -> E (32627) task_wdt: Aborting.
14:18:42.220 -> 
14:18:42.220 -> abort() was called at PC 0x400da6a9 on core 0

My project is to build a web radio, with stream collected by ESP32 via Wifi and sent to JBL GO 2 speaker via Bluetooth.
Thanks for your help.

pschatzmann, 2023-08-26:
Adding a delay is the only reliable way to resolve watchdog errors.

The actual updated version is issuing warnings instead of errors for unhandled events. I also extended the BluetoothA2DPSource logic quite a bit.

closed 2023-08-26 by pschatzmann

tutoduino, 2023-08-26:
Thanks. I discover that new ESP32C3 does not support anymore A2DP, that's a pity!
It is not possible anymore to stream music to Bluetooth speakers on such microcontroller?

Patapom, 2024-08-02:
Hi, sorry to reopen this issue but I'm using the same sketch, **bt_music_sender** with an ESP32-WROOM, and I keep having some stuttering because of the mandatory delay( 1 ) call.
If I remove it, the sound is crystal clear but then the watchdog even occurs after 5 seconds and my device reboots...

Is there any solution to that problem? Maybe requesting less samples would avoid starving the other threads, is this configurable?

Thank you for your time and for your amazing work!

pschatzmann, 2024-08-02:
Did you try to increase the delay to give the a2dp task more time ?

Patapom, 2024-08-03:
Hi, I was talking of the delay( 1 ) call _inside_ the sampling loop callback, where we feed samples to the A2DP. if I increase the delay there then the stuttering gets worse.
If I increase/decrease the delay() in the main loop() function, nothing changes... :/