pschatzmann/ESP32-A2DP #466: Compilation error with bt_music_sender example

d3mac123, 2023-08-29:
Hi,

I am trying to learn how to play a sound from my m5stack ATOM Matrix to my Bluetooth headphones. It is basically a countdown (my initial idea was to use the Talkie library but I couldn't make it work).

Assuming this is possible, I tried to use the bt_music_sender example but I am getting the following error (on Arduino IDE 2.2). Any ideas on what I am doing wrong?

`/Users/alexsouza/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::bt_app_gap_callback(esp_bt_gap_cb_event_t, esp_bt_gap_cb_param_t*)':
/Users/alexsouza/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPSource.cpp:546:8: error: 'ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT' was not declared in this scope
   case ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT:
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/alexsouza/Documents/Arduino/libraries/ESP32-A2DP/src/BluetoothA2DPSource.cpp:546:8: note: suggested alternative: 'ESP_BT_GAP_QOS_CMPL_EVT'
   case ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT:
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        ESP_BT_GAP_QOS_CMPL_EVT

exit status 1

Compilation error: exit status 1`

pschatzmann, 2023-08-29:
Please don't ignore that I am asking for Software Verisions

https://github.com/pschatzmann/ESP32-A2DP/wiki/Reporting-Bugs
https://github.com/pschatzmann/ESP32-A2DP#support

closed 2023-08-30 by pschatzmann

d3mac123, 2023-08-31:
Phil, thank you so much for your input. Just started playing with ESP/Arduino boards, this was very valuable. I really appreciated that.

I will try to address the memory issues I am having now to, then, see how to implement the Simple TTS library into the project. 

Again, thanks a lot!