pschatzmann/ESP32-A2DP #470: Compile Error()

mongonta0716, 2023-09-01:
Hi, 

I am getting the following error when building with arduino-esp32. Probably, esp-idf does not give this error. The cause is this commit.

793aac4fec7d49958a50d1cdb88548835d573034

There is a definition in esp-idf but not in esp_gap_bt_api.h of arduino-esp32.

How should it be handled?

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/bt/host/bluedroid/api/include/api/esp_gap_bt_api.h#L209-L227

```.pio/libdeps/m5stack-core-esp32/ESP32-A2DP/src/BluetoothA2DPSource.cpp: In member function 'virtual void BluetoothA2DPSource::bt_app_gap_callback(esp_bt_gap_cb_event_t, esp_bt_gap_cb_param_t*)':
.pio/libdeps/m5stack-core-esp32/ESP32-A2DP/src/BluetoothA2DPSource.cpp:547:8: error: 'ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT' was not declared in this scope
   case ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT:
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/m5stack-core-esp32/ESP32-A2DP/src/BluetoothA2DPSource.cpp:547:8: note: suggested alternative: 'ESP_BT_GAP_QOS_CMPL_EVT'
   case ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT:
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        ESP_BT_GAP_QOS_CMPL_EVT
Compiling .pio/build/m5stack-core-esp32/libecb/YAMLDuino/ArduinoYaml.cpp.o
Compiling .pio/build/m5stack-core-esp32/libecb/YAMLDuino/cJSON/cJSON.c.o
*** [.pio/build/m5stack-core-esp32/lib523/ESP32-A2DP/BluetoothA2DPSource.cpp.o] Error 1
```

I have been using ESP32-A2DP for a long time. Thank you.

pschatzmann, 2023-09-01:
https://github.com/pschatzmann/ESP32-A2DP#support

closed 2023-09-01 by pschatzmann

mongonta0716, 2023-09-01:
arduino-esp32 is using 2.0.8 (platform:espressif@6.2.0).

Well, if I set the version of esp32-a2dp to fixed (v1.7.3) in the tag, there is no problem, so I will go with that solution.

pschatzmann, 2023-09-01:
The actual version is 2.0.11. I suggest that you upgrade...

mongonta0716, 2023-09-01:
I built it with arduino-esp32 2.0.11 (espressif@6.4.0) as you said and it went through. Sorry for taking your time.